evcc-io/evcc #12739: Option zur Akkukalibrierung im PV-Modus bei Schwellwert

floqqi, 2024-03-07:
Tesla empfiehlt seine LFP-Akkus regelmäßig auf 100% aufzuladen, um das BMS zu kalibrieren. Ich nutze nahezu ausschließlich den PV only-Modus in evcc. Nun kann es vorkommen, dass die Ladung aufgrund mangelndem Überschuss in den letzten Prozentbereichen abbricht. Wäre es denkbar eine Option alá "Akkukalibrierung erzwingen" einzubauen? Umgesetzt beispielsweise als wählbarer Schwellwert in Prozent, ab wann die Akkukalibrierung im PV-Modus trotz fehlenden Überschuss erzwungen werden kann.

andig, 2024-03-07:
Du tust so, als ob PV *immer* bei 98% abbrechen würde. Es kann ja auch voll werden und schon ist das Problem gelöst.

closed 2024-03-07 by andig

floqqi, 2024-03-08:
> Du tust so, als ob PV _immer_ bei 98% abbrechen würde. Es kann ja auch voll werden und schon ist das Problem gelöst.

Klar, kann es das. Passiert auch oft. Aber generell finde ich es nicht wünschenswert einen Ladevorgang ab einem gewissen Schwellwert abzubrechen, nur weil kein Überschuss mehr da ist, denn die BMS-Kalibrierung ist wichtig.

Du tust so, als wäre das kein realistisches Szenario, was aber doch häufiger auftritt, wenn die Wallbox 3-phasig angeschlossen ist, weil 4,1 kW halt ab einem gewissen Sonnenstand ab Nachmittag nicht mehr runterkommt, aber noch locker zu BMS-Kalibrierung genügen würden.

andig, 2024-03-08:
Einfach auf "Sofort" schalten und das Problem ist gelöst. Akkukalibirierung ist keine Funktion die wir umsetzen wollen (oder die normalerweise auch nur entfernt notwendig wäre).

premultiply, 2024-03-08:
...oder Min+PV nutzen.

floqqi, 2024-03-08:
> Einfach auf "Sofort" schalten und das Problem ist gelöst. Akkukalibirierung ist keine Funktion die wir umsetzen wollen (oder die normalerweise auch nur entfernt notwendig wäre).

Das ist das was ich meinte. Es ist ein manueller Eingriff nötig, ich muss ständig im Blick haben, wie viel Überschuss ich habe bzw. wann nicht mehr genügt geliefert wird und den Ladestand im Blick haben. Das alles könnte man mit einem Schwellwert automatisieren. Sind nur meine 2 cents, aber ich erachte das als grundsätzlich gute Funktion (nicht nur für Tesla). 

> ...oder Min+PV nutzen.

Auch hier, wie oben bereits beschrieben, manueller Eingriff nötig. Ich bin ein Fall von Automate all the things :-) 

Gegenvorschlag: Man muss das Feature ja nicht "Akkukalibrierung" nennen, sondern man nennt es in die Richtung "Restladung erzwingen ab {wählbar} %"? Gleiches Kind beim anderen Namen, aber IMHO eine sinnvolle Funktion, dass ich z. B. sagen kann, bitte immer vollmachen, wenn du mind. 95% erreicht hast, auch wenn der Überschuss nicht mehr reicht (was bei 3-phasig schnell passieren kann).

panzerdev, 2024-03-08:
@floqqi Bis auf Tesla LFP Autos und paar von den mini Akku Autos will normal doch fast niemand bis 100% laden im PV Modus. Wenn ich meine Autos auf 100% lade dann nur wenn ich weiter weg will und dann auch nur mit dem Planer zur Abfahrtszeit hin.

Dein Featurerequest kannst dir extern mit einer Automation per api selber basteln.