evcc-io/evcc #12740: Hyundai Bluelink: EVCC uses invalid SOC of 0% from Bluelink

Kishi85, 2024-03-07:
### Describe the bug

I'll try this again, this time with logs (but they just tell the same story I'm describing here so nothing new):

If the BMS on a Hyundai (Kona EV 2020 in my case) is sleeping or otherwise busy internally then there is a possible condition where on requesting a status update the Battery SoC will be reported at 0% by BlueLink (even in the App, which is obviously very very rarely correct as it should never happen during "normal" use), once the BMS is ready a few seconds later (30s are usually more than enough if refreshing manually with the app) the next status update request to BlueLink returns the correct SoC (being >0%).

With EVCC if a wrongly reported 0% is cached for the usual 15 minutes the minimum charge option is active during that time trying to charge the car up to the defined safe SoC at full speed ignoring PV-Mode even though it might be already (far) above that limit -> I've had to disable this for now as it drains my batteries or even worse pulls from grid unnecessarily.

What should happen is that EVCC ignores the wrongly reported 0% SoC (at least if the last cached SoC is >0% and therefore valid) and just keep working with known the last known good SoC from the previous status request until the next scheduled cache update.

### Steps to reproduce

1. Connect Bluelink Car
2. Let Car charge using EVCC
3. If BMS is to slow to react BlueLink will report in status update that SoC is (wrongly) 0%
4. If minimum charge limit is set then car will fast charge until the next scheduled status cache update receiving the real SoC
5. On the next BlueLink status update the real (correct) SoC will be reported and EVCC works correctly again.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local
  port: 7070

log: info
levels:
  cache: error
  lp-1: debug
  hyundai: debug
  bluelink: debug

# unique installation id
plant: <redacted>

interval: 10s # control cycle interval (default: 10s)

meters:
- type: template
  template: solaredge-hybrid 
  id: 11  
  host: <redacted>
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 11  
  host: <redacted>
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: solaredge-hybrid 
  id: 11  
  host: <redacted>
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery1
  capacity: 9.2
- type: template
  template: solaredge-hybrid 
  id: 12  
  host: <redacted>
  port: 1502  
[... 181 lines cut ...]
Mar 07 11:59:47 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 11:59:47 vehicle soc: 1%
Mar 07 11:59:58 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 11:59:58 vehicle soc: 1%
Mar 07 12:00:07 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:07 vehicle soc: 1%
Mar 07 12:00:17 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:17 vehicle soc: 1%
Mar 07 12:00:27 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:27 vehicle soc: 1%
Mar 07 12:00:36 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:36 vehicle soc: 1%
Mar 07 12:00:47 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:47 vehicle soc: 1%
Mar 07 12:00:57 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:00:57 vehicle soc: 1%
Mar 07 12:01:06 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:06 vehicle soc: 1%
Mar 07 12:01:17 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:17 vehicle soc: 1%
Mar 07 12:01:27 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:27 vehicle soc: 1%
Mar 07 12:01:37 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:37 vehicle soc: 1%
Mar 07 12:01:47 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:47 vehicle soc: 1%
Mar 07 12:01:57 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:01:57 vehicle soc: 1%
Mar 07 12:02:07 evcc evcc[1455402]: [lp-1  ] INFO 2024/03/07 12:02:07 stop charging <-
Mar 07 12:02:13 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 12:02:13 vehicle soc: 58%
```

### What type of operating system are you running?

Linux

### Version

0.124.7

andig, 2024-03-07:
Du meinst diese Zeile?

> Mar 07 11:52:59 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 11:52:59 vehicle soc: 0%

Bitte `hyundai: trace` damit wir sehen was da vom API kommt.

Kishi85, 2024-03-07:
> Du meinst diese Zeile?
> 
> > Mar 07 11:52:59 evcc evcc[1455402]: [lp-1  ] DEBUG 2024/03/07 11:52:59 vehicle soc: 0%
> 
> Bitte `hyundai: trace` damit wir sehen was da vom API kommt.

After setting that and force starting the charge with minpv and stopping it (off) after 10 seconds I've gotten lucky and it triggered the state first time on stop (EVCC now shows 0% for charge state even though the App shows 66% after refreshing).
Here's the (bad) API result (luckily I've managed to trigger it within the API Limit I've got left for today):
```
Mar 07 16:48:04 evcc evcc[1461890]: [hyundai] TRACE 2024/03/07 16:48:04 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/<redacted>/status/latest
Mar 07 16:48:05 evcc evcc[1461890]: [hyundai] TRACE 2024/03/07 16:48:05 {"retCode":"S","resCode":"0000","resMsg":{"vehicleStatusInfo":{"vehicleLocation":{"coord":{<redacted>},"head":<redacted>,"speed":{"value":0,"unit":0},"accuracy":{"hdop":10,"pdop":10},"time":"20240307154332"},"vehicleStatus":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"0CH","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":false,"batteryStatus":66,"batteryPlugin":2,"remainTime2":{"etc1":{"value":47,"unit":1},"etc2":{"value":1660,"unit":1},"etc3":{"value":370,"unit":1},"atc":{"value":245,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":221,"unit":1},"totalAvailableRange":{"value":221,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[1,2,3,4],"time":{"time":"0730","timeSection":0}},"reservChargeSet":true,"reservFatcSet":{"defrost":true,"airTemp":{"value":"16H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":2},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[1,2,3,4,5],"time":{"time":"0500","timeSection":1}},"reservChargeSet":false,"reservFatcSet":{"defrost":true,"airTemp":{"value":"16H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":9,"time":{"time":"1200","timeSection":0}},"end":{"day":9,"time":{"time":"1200","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"dte":{"rangeByFuel":{"evModeRange":{"value":271,"unit":1},"totalAvailableRange":{"value":271,"unit":1}},"type":2},"plugType":0},{"targetSOClevel":100,"dte":{"rangeByFuel":{"evModeRange":{"value":345,"unit":1},"totalAvailableRange":{"value":345,"unit":1}},"type":2},"plugType":1}]}},"ign3":true,"hoodOpen":false,"transCond":true,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":80,"batState":0},"time":"20240307164004"},"odometer":{"value":<redacted>,"unit":1}}},"msgId":"38f26420-ab79-475e-a30b-6fd85e4e69ac"}
Mar 07 16:48:05 evcc evcc[1461890]: [hyundai] TRACE 2024/03/07 16:48:05 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/<redacted>/status
Mar 07 16:48:16 evcc evcc[1461890]: [hyundai] TRACE 2024/03/07 16:48:16 {"retCode":"S","resCode":"0000","resMsg":{"airCtrlOn":false,"engine":false,"doorLock":true,"doorOpen":{"frontLeft":0,"frontRight":0,"backLeft":0,"backRight":0},"trunkOpen":false,"airTemp":{"value":"00H","unit":0,"hvacTempType":1},"defrost":false,"acc":false,"evStatus":{"batteryCharge":true,"batteryStatus":0,"batteryPlugin":2,"remainTime2":{"etc1":{"value":26,"unit":1},"etc2":{"value":1660,"unit":1},"etc3":{"value":370,"unit":1},"atc":{"value":245,"unit":1}},"drvDistance":[{"rangeByFuel":{"evModeRange":{"value":221,"unit":1},"totalAvailableRange":{"value":221,"unit":1}},"type":2}],"reservChargeInfos":{"reservChargeInfo":{"reservChargeInfoDetail":{"reservInfo":{"day":[1,2,3,4],"time":{"time":"0730","timeSection":0}},"reservChargeSet":true,"reservFatcSet":{"defrost":true,"airTemp":{"value":"16H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"offpeakPowerInfo":{"offPeakPowerTime1":{"starttime":{"time":"1200","timeSection":0},"endtime":{"time":"1200","timeSection":0}},"offPeakPowerFlag":2},"reserveChargeInfo2":{"reservChargeInfoDetail":{"reservInfo":{"day":[1,2,3,4,5],"time":{"time":"0500","timeSection":1}},"reservChargeSet":false,"reservFatcSet":{"defrost":true,"airTemp":{"value":"16H","unit":0,"hvacTempType":0},"airCtrl":0,"heating1":0}}},"reservFlag":0,"ect":{"start":{"day":0,"time":{"time":"0000","timeSection":0}},"end":{"day":0,"time":{"time":"0000","timeSection":0}}},"targetSOClist":[{"targetSOClevel":80,"plugType":0},{"targetSOClevel":100,"plugType":1}]}},"ign3":true,"hoodOpen":false,"transCond":false,"steerWheelHeat":0,"sideBackWindowHeat":0,"tirePressureLamp":{"tirePressureLampAll":0,"tirePressureLampFL":0,"tirePressureLampFR":0,"tirePressureLampRL":0,"tirePressureLampRR":0},"battery":{"batSoc":80,"batState":0},"time":"20240307164815"},"msgId":"e80acd4a-2a95-4f25-bb88-769fd567e96a"}
```

The call to ```https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/<redacted>/status``` returns ```... "evStatus":{"batteryCharge":true,"batteryStatus":0,"batteryPlugin":2, ...``` in this case whilst the immediate the real battery status is 66 at this time (as shown on the previous status/latest a second earlier) and upon refreshing status again (e.g. with the App).

Shouldn't there be a sanity check that filters out invalid API response values (like SoC value jumps from 66% to 0% which is physically highly improbable) in the estimator or somewhere along those lines? Such cases theoretically might occur on other APIs as well.

pkuehnel, 2024-09-17:
It looks like the issues is not rare. It always happens when you need to enter a PIN in the Bluelink app - not sure when this is triggered. The main tab of the car displays range and time until charge limit is reached, there is no PIN needed, but when opening the status tab in the app you need to enter a PIN, not always but after some time. Also, when you enter the status tab in the app you can see the SOC of 0% in the background, after entering the PIN, the soc is displayed correctly and EVCC also is able to display the correct SOC after the next API call. Not sure what this PIN actually is, but it seems to be something like a second authorization factor. If you have two phones with the app, both have the same pin, so the pin is somehow related to the account. You can reset the pin via app, If you do that you get a force logout on that specific device immediatly. On the other devices it takes some time until you get a forced logout there as well.

pkuehnel, 2024-11-18:
Nope. No solution. I will extend my software https://github.com/pkuehnel/TeslaSolarCharger in the coming month to support Hyundai and OCPP to fix this.