evcc-io/evcc #12741: evcc Batterie wird nicht geladen trotz Überschuss

phen0, 2024-03-07:
### Describe the bug

Trotz Überschuss wird die Batterie nicht geladen. Zumindest ist die Anzeige bei Null. Ich meine aber das trotzdem geladen wird.

### Steps to reproduce

Siehe Screenshot.
[image]

### Configuration details

```yaml
network:
  schema: http
  port: 7070

log: info
levels:
  cache: error
  site: info
  lp-1: info
  warp: info

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 10.0.x.x
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 10.0.x.x
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 10.0.x.x
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- name: wallbox1
  type: template
  template: tinkerforge-warp
  host: 10.0.x.x
  port: 1883
  topic: warp2/wallbox
  timeout: 30s
  energymanager: wem/26KN

vehicles:
- name: corsa-e
  type: template
  template: flobz
  title: 'PSA Corsa-e'
  url: http://10.0.x.x:5000
  vin: xxx
  capacity: 50

loadpoints:
- title: WARP2
  charger: wallbox1
  mode: pv
  enable:
    threshold: -1400
  disable:
    delay: 3m
  guardDuration: 5m

tariffs:
  grid:
    type: fixed
  feedin:
    type: fixed

influx:
  url: http://10.0.x.x:8086
  token: xxx
  org: casa

mqtt:
  broker: 10.0.x.x:1883

site:
  title: PV-EVCC
  meters:
    grid: grid1
    pv: pv2
    battery: battery3
  residualPower: 100
```

### Log details

```text
[lp-1  ] ERROR 2024/03/07 10:16:23 charge meter: outdated
[lp-1  ] ERROR 2024/03/07 10:16:23 charge meter: outdated
[lp-1  ] ERROR 2024/03/07 10:16:23 charge total import: outdated
[lp-1  ] ERROR 2024/03/07 10:16:23 charger: outdated
[lp-1  ] INFO 2024/03/07 12:38:23 car disconnected
[lp-1  ] INFO 2024/03/07 12:38:23 vehicle updated: PSA Corsa-e -> unknown
[lp-1  ] INFO 2024/03/07 13:04:53 car connected
[lp-1  ] INFO 2024/03/07 13:05:53 vehicle updated: unknown -> PSA Corsa-e
[lp-1  ] INFO 2024/03/07 13:08:23 start charging ->
[lp-1  ] ERROR 2024/03/07 13:31:53 charge meter: outdated
[lp-1  ] ERROR 2024/03/07 13:31:53 charge rater: charge meter error outdated
[lp-1  ] ERROR 2024/03/07 13:31:53 charge total import: outdated
[lp-1  ] ERROR 2024/03/07 13:31:53 charger: outdated
[main  ] INFO 2024/03/07 13:36:19 evcc 0.124.7
[main  ] INFO 2024/03/07 13:36:19 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/07 13:36:19 starting ui and api at :7070
[db    ] INFO 2024/03/07 13:36:19 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/07 13:36:19 connecting evcc-1224418702 at tcp://10.0.55.85:1883
[site  ] INFO 2024/03/07 13:36:20 site config:
[site  ] INFO 2024/03/07 13:36:20   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/07 13:36:20     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/07 13:36:20     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/03/07 13:36:20     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/07 13:36:20   vehicles:
[site  ] INFO 2024/03/07 13:36:20     vehicle 1: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/03/07 13:36:20 loadpoint 1:
[lp-1  ] INFO 2024/03/07 13:36:20   mode:        pv
[lp-1  ] INFO 2024/03/07 13:36:20   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/07 13:36:20   meters:      charge ✓
[lp-1  ] INFO 2024/03/07 13:36:20     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/03/07 13:36:20 car connected
[lp-1  ] INFO 2024/03/07 13:36:20 start charging ->
[lp-1  ] INFO 2024/03/07 13:36:20 vehicle updated: unknown -> PSA Corsa-e
[lp-1  ] INFO 2024/03/07 13:36:51 stop charging <-
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.7

andig, 2024-03-07:
> Trotz Überschuss wird die Batterie nicht geladen. Zumindest ist die Anzeige bei Null. Ich meine aber das trotzdem geladen wird.

Was ist hier das konkrete Problem?

phen0, 2024-03-07:
Hi, obwohl die Batterie nicht voll ist steht bei Ladung 0% und es wird eingespeist. Ich frage mich nur warum er nichts mehr anzeigt und einspeist?

andig, 2024-03-07:
Keine Ahnung- die Ladung steuert Dein WR. Mach doch mal bitte

    evcc meter

und schau welche Messwerte da stimmen oder nicht.

phen0, 2024-03-07:
Das macht der WR, aber evcc zeigt nichts an.
Hier das evcc meter
```/app # evcc meter
[main  ] INFO 2024/03/07 15:16:52 evcc 0.124.7
[main  ] INFO 2024/03/07 15:16:52 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/07 15:16:52 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/07 15:16:52 connecting evcc-1721300768 at tcp://10.0.55.85:1883
grid1
-----
Power:  -1178W
Energy: 8009.5kWh

pv2
---
Power:  1877W
Energy: 16267.3kWh

battery3
--------
Power:  0W
Energy: 3608.7kWh
Soc:    95%
```

Und der aktuelle Screenshot
[image]

andig, 2024-03-07:
> Das macht der WR, aber evcc zeigt nichts an.

*Was* zeigt evcc nicht an? Bitte kein Suchspiel sondern konkret.

> und schau welche Messwerte da stimmen oder nicht.

Da fehlt die Antwort. Stimmen die Messwerte von `evcc meter`?

alexonbandit, 2024-03-07:
Hi, kann es sein, das evcc keinen exklusiven Zugriff auf die Register deines WR hat? Zum Beispiel Home Assistant?

closed 2024-03-08 by andig