evcc-io/evcc #12745: battery mode: modbus: exception '3' (illegal data value), function '16'

HolgerLexow, 2024-03-07:
### Describe the bug

after implementation battery SMA sunny Island I get this bug and as well bug
"modbus: exception '3' (illegal data value), function '16'"

### Steps to reproduce

1. UI configuration - Grid config changes are not possible
2.add device config sma-si-modbus with IP and modes address 3 check - all is ok (green)
then restart - and issue is happened see above

### Configuration details

```yaml
# open evcc at http://evcc.local:7070

network:
  schema: http
  port: 7070
  host: evcc # .local suffix announces the hostname on MDNS
#  host: localhost # .local suffix announces the hostname on MDNS

log: trace
#levels:
#  cache: error

interval: 60s # control cycle interval

sponsortoken: xxx

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: shm2  
  name: grid1
  
- name: my_pv
  type: template
  template: sma-inverter-modbus
  usage: pv
  # Modbus TCP
  modbus: tcpip
  id: 3
  host: SunnyIsland6.0H12 # Hostname
  port: 502 # Port 

chargers:
- type: template
  template: easee 
  user: xxx  
  password: xxx 
  charger: xxx  
  name: wallbox3
- type: template
  template: easee 
  user: xxx  
  password: xxx  
  charger: xxx  
  name: wallbox4

vehicles:
- type: template
  template: vw 
  title: Passat GTE  
  user: xxx  
  password: xxx  
  vin: xxx  
  capacity: 13  
  phases: 1  
[... 87 lines cut ...]
2024/03/07 15:17:57	stdout	[sma   ] TRACE 2024/03/07 15:17:57 recv 192.168.178.11: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
2024/03/07 15:17:57	stdout	[semp  ] TRACE 2024/03/07 15:17:57 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/evcc-io/evcc ModelName:EVCC SEMP Server 0.124.7 UDN:uuid:7f1fa681-dc8b-11ee-86ae-001132fd8a66 PresentationURL:http://192.168.178.49:7070 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.178.49:7070 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0} ServiceList:[]}}
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}} {Identification:{DeviceID:F-28081973-df08cdcf74dc-00 DeviceName:Garage DeviceType:EVCharger DeviceSerial:001132fd8a66-1 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:11040 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}} {Identification:{DeviceID:F-28081973-df08cdcf74dc-00 DeviceName:Garage DeviceType:EVCharger DeviceSerial:001132fd8a66-1 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:11040 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}} {Identification:{DeviceID:F-28081973-df08cdcf74dc-00 DeviceName:Garage DeviceType:EVCharger DeviceSerial:001132fd8a66-1 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:11040 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}} {Identification:{DeviceID:F-28081973-df08cdcf74dc-00 DeviceName:Garage DeviceType:EVCharger DeviceSerial:001132fd8a66-1 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:11040 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}} {Identification:{DeviceID:F-28081973-df08cdcf74dc-00 DeviceName:Garage DeviceType:EVCharger DeviceSerial:001132fd8a66-1 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:11040 MinOnTime:0 MinOffTime:0} Capabilities:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
2024/03/07 15:17:56	stdout	[sma   ] TRACE 2024/03/07 15:17:56 recv 192.168.178.11: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
2024/03/07 15:17:56	stdout	[semp  ] TRACE 2024/03/07 15:17:56 send: {Xmlns:http://www.sma.de/communication/schema/SEMP/v1 DeviceInfo:[{Identification:{DeviceID:F-28081973-df08cdcf74db-00 DeviceName:Carport DeviceType:EVCharger DeviceSerial:001132fd8a66-0 DeviceVendor:github.com/evcc-io/evcc} Characteristics:{MinPowerConsumption:1380 MaxPowerConsumption:3680 [... cut ...] ties:{CurrentPowerMethod:Measurement AbsoluteTimestamps:false InterruptionsAllowed:true OptionalEnergy:true}}] DeviceStatus:[{DeviceID:F-28081973-df08cdcf74db-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}} {DeviceID:F-28081973-df08cdcf74dc-00 EMSignalsAccepted:false Status:Off PowerInfo:{AveragePower:0 Timestamp:0 AveragingInterval:60}}] PlanningRequest:[]}
....
```

### What type of operating system are you running?

Linux

### Version

0124.7

andig, 2024-03-07:
@premultiply ähnlich zu https://github.com/evcc-io/evcc/issues/12230?

HolgerLexow, 2024-03-07:
ggf ist die Lösung wie folgt
# open evcc at http://evcc.local:7070

network:
  schema: http
  port: 7070
  host: evcc # .local suffix announces the hostname on MDNS
#  host: localhost # .local suffix announces the hostname on MDNS

log: fatal
levels:
  cache: error

interval: 60s # control cycle interval

sponsortoken: xxx

meters:
- name: grid1
  type: template
  template: sma-home-manager 
  usage: grid  
  host: shm2  
  
- name: SunnyIsland6
  type: template
  template: sma-si-modbus
  usage: battery
  # Modbus TCP
  modbus: tcpip
  id: 3
  host: SunnyIsland6.0H12 # Hostname
  port: ccc # Port 
  
- name: pv-WG
  type: template
  template: sma-inverter-modbus #modbus
  #template: sma-inverter-speedwire #speedwire
  usage: pv
  # Modbus TCP
  modbus: tcpip #modbus
  id: 10 #modbus
  host: SUNNYTRIPOWER6WLAN # Hostname
  port: ccc # Port  #modbus

- name: pv-Haus
  type: template
  template: sma-inverter-modbus #modbus
  #template: sma-inverter-speedwire #speedwire
  usage: pv
  # Modbus TCP
  modbus: tcpip #modbus
  id: 7 #modbus
  host: STP12000TL-20-113 # Hostname
  port: ccc # Port  #modbus

- name: pv-Garage
  type: template
  template: sma-inverter-modbus #modbus
  #template: sma-inverter-speedwire #speedwire
  usage: pv   
  # Modbus TCP
  modbus: tcpip #modbus
  id: 6 #modbus
  host: STP5000TL-20-235 # Hostname
  port: ccc # Port #modbus 
    
- name: pv-Pool
  type: template
  template: sma-inverter-modbus #modbus
  #template: sma-inverter-speedwire #speedwire
  usage: pv  
  # Modbus TCP
  modbus: tcpip #modbus
  id: 4 #modbus
  host: STP5000TL-20-694 # Hostname
  port: ccc # Port  #modbus 

chargers:
- type: template
  template: easee 
  user: xxx  
  password: xxx  
  charger: xxx  
  name: Wallbox-Carport
- type: template
  template: easee 
  user: xxx  
  password: xxx  
  charger: xxx  
  name: Wallbox-Garage

vehicles:
- type: template
  template: vw 
  title: Passat GTE  
  user: xxx  
  password: xxx  
  vin: xxx  
  capacity: 13  
  phases: 1  
  cache: 15m  
  mode: pv  
  name: VW-Passat
- name: Mercedes
  type: mercedes
  capacity: 30  
  phases: 3
  title: Mercedes GLE 350de
  vin: xxx
 

loadpoints:
- title: Carport
  charger: Wallbox-Carport
  vehicle: VW-Passat
  mode: pv

  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 1000 # maximum import power (W)
    delay: 15m
- title: Garage
  charger: Wallbox-Garage
  vehicle: Mercedes
  mode: pv

  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 1000 # maximum import power (W)
    delay: 15m

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    battery: SunnyIsland6
    pv:
      - pv-Garage
      - pv-WG
      - pv-Haus
      - pv-Pool

hems:
  type: sma
  allowcontrol: false

tariffs:
  currency: EUR # (default EUR)
  grid:
  # either
    # static grid price
    type: tibber #fixed #tibber
   # price: 0.2874 # [currency]/kWh
    token: "xxx"

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1187 # [currency]/kWh

premultiply, 2024-03-08:
Ich blicke hier nicht durch. Log zeigt den Fehler nicht, sinnlos "anonymisierte" Configs die keinen Zusammenhang erkennbar lassen. Mache daher erstmal zu. Ggf. weitere dienliche Infos ergänzen.

closed 2024-03-08 by premultiply