evcc-io/evcc #12746: Anbindung von Solarprognose.de / Ladeplanung anhand von PV-Ertrags-Prognose

kafetzke, 2024-03-07:
Moin zusammen,

ich hoffe ich habe im Vorfeld dieses Thema nicht in einem vorhandenen Ticket übersehen - falls ja, mea culpa.

**Die Idee**
Die API der Website https://www.solarprognose.de/ bietet eine stundenbasierte Prognose über den erwarteten PV-Ertrag, und zwar abhängig von Standort, installierter PV-Leistung und Ausrichtung der Anlage. Die API selbst benötigt einen Anmelde-Token, anhand dieses Tockens werden Accountweise die o.g. Kriterien herangezogen. Bis 10kWp ist der Account kostenfrei, darüber hinaus für kleines Geld zu haben.
Die API selbst ist hier beschrieben: https://www.solarprognose.de/web/de/solarprediction/page/api

**Usecase**
Aktuell schaue ich manuell wann wohl die (sonnentechnisch) beste Zeit zum Laden wäre und starte die Ladung dann von Hand. 
Diese stündlichen Daten könnte man in EVCC aber auch prima für eine automatisierte Ladeplanung verwenden: "Ziel: Ladeziel bis xx:xx" -> "Prüfe, wann ist die meiste Sonne zu erwarten und plane die Ladung entsprechend". Also vergleichbar mit der Anbindung für dynamische Strompreise.

**Das Ziel**
Liegt auf der Hand - möglichst viel dann laden, wenn gerade Sonne vorhanden ist :-)

**Kosten / Anforderungen**
Der EVCC-Anwender müsste sich hierfür selbstständig einen Account auf https://www.solarprognose.de/ erstellen und dort seine persönliche PV-Anlage hinterlegen. Diese Daten werden dann über den Token (der dann in der Konfiguration von EVCC hinterlegt werden muss) für die Prognose verwendet.

Vielleicht findet mein Vorschlag ja Anklang in der Community und beim Entwicklungssteam :-)

andig, 2024-03-07:
Es gibt bereits einen verwandten PR, siehe https://github.com/evcc-io/evcc/pull/9454. Weitere Prognosesysteme lassen sich dann immer noch einbinden ;)

mdkeil, 2024-03-08:
hat sich glaube leider eh erledigt.

> With a heavy heart, I must unfortunately discontinue this project. Shutdown on 31.03.2024. Many thanks to all users and supporters. The registration is already disabled.

Quelle: https://www.solarprognose.de/web/de

mucki12, 2024-03-08:
Zum Glück gibt es ja zig andere Dienste.

closed 2024-03-08 by andig

lmfgpeace, 2025-01-12:
Ich lasse das mal wieder aufleben hier :)

Nicht nur für das Laden wäre es sinnvoll, sondern gerade für das Thema intelligentes „Netzladen“ des Speichers.

Damit der Speicher

A) nur dann aus dem Netz geladen wird, wenn der Strom günstig ist und
B) nur dann, wenn perspektivisch morgen (oder zum eingestellten Zeitpunkt) keine Sonne scheinen wird.

Man könnte also einen Modus beim Netzladen hinzufügen der da lautet „intelligentes/prognostiziertes Laden“ - dann natürlich nur so viel wie vmtl. notwendig sein wird.

Ergibt ja nämlich keinen Sinn den Speicher z.B. von 3-5 Uhr Nachts zu laden (weil der Strom günstig ist) aber am nächsten Tag vom Dach kostenloser Strom kommt.

premultiply, 2025-01-13:
> Ergibt ja nämlich keinen Sinn den Speicher z.B. von 3-5 Uhr Nachts zu laden (weil der Strom günstig ist) aber am nächsten Tag vom Dach kostenloser Strom kommt.

Das kann man so pauschal nicht sagen, da das wesentlich vom (unbekannten) Verbrauch, der Speicherkapazität, des SoC und der Speicherleistung abhängt.

In meinem Fall wäre der Speicher selbst bei der skizzierten Zwangsvollladung aus dem Netz wieder völlig leer bevor auch nur die Sonne aufgeht. Die WP zieht alles weg.

Das ist somit nur ein "Luxusproblem" für die warmen Sommermonate wo ohnehin oft Energie im Überfluss vorhanden ist.

Leider nicht so einfach.

kafetzke, 2025-01-13:
Moin,

das Folgende ist als **Anregung** für Lösungswege gedacht, nicht als "Best-practise" :-)

Mein Szenario:
- 2 PV-Anlagen (6,75kWp mit 40° Dachneigung und 0°-Süd-Ausrichtung + 4,5kWp mit 15° Aufständerung und ebenfalls 0°-Süd-Ausrichtung)
- 16,5kWh Heimspeicher
- Wärmepumpe
- Wallbox (tägliche Ladung ca. 25kWh)
- Stromtarif: Octopus Energy mit zwei Zeitbereichen (00:00-05:00 = günstig, 05:00-24:00 teurer)

Mein Lösungsweg (mit Hilfe von ioBroker)
- Mein ioBroker ruft täglich für beide PV-Anlagen getrennt den zu erwartenden Ertrag des Folgetages ab. Dafür nutze ich das ioBroker-Plugin für pvforecast (https://github.com/iobroker-community-adapters/ioBroker.pvforecast/blob/main/docs/de/README.md). Dieses berücksichtigt zusätzlich auch den Standort der Anlage und damit auch die Wetterprognose. Die errechneten Daten sind übrigens zumindest als Näherungswert durchaus zu gebrauchen.
- Regel in ioBroker: 
1. Wenn Wärmepumpe im Winterbetrieb, lade Hausspeicher immer bis mindestens 75% -> mehr als 25% zusätzliche PV-Ladung erreicht der Speicher durch die Wärmepumpe selbst an perfekten Wintertagen eh nicht. Für die restlichen 25%: Differenz Summe der PV-Prognose zu besagten 25% Akku-Kapazität. 
2. Wenn Wärmepumpe im Sommerbetrieb, lade 100% der Kapazität - 80% des erwarteten Gesamt-Ertrags des Nachts in den Speicher. Das kommt bei mir ganz gut hin.
- Steuerung EVCC durch ioBroker: über die EVCC-API (https://docs.evcc.io/docs/integrations/rest-api) lässt sich der Datenpunkt "battery grid charge limit" setzen. Soll der Speicher nicht mehr laden, wird dieser Wert auf 5ct gesetzt. Dann lädt da gar nix mehr :-)

Die Ladung meiner Fahrzeuge berücksichtige ich hierbei übrigens bewusst nicht. Ich nutze hierfür die Zeitpläne, sperre während der Ladung die Batterie und lade bei Sonne via PV-Überschuss und den Rest nachts in der günstigeren Zeit.

Ggf. ließe sich so ein Lösungsweg auch direkt in EVCC umsetzen. Allerdings sollte das Entwicklungsteam sich an dieser Stelle aber auch mit dem Gedanken befassen, ob EVCC damit nicht über sein eigentliches Ziel hinausschießt und aus einer (sehr) intelligenten Ladesteuerung so langsam zu einem Energie-Managementsystem mutiert/weiterentwickelt.

profex1337, 2025-02-18:
Wäre auch für eine Implementierung.

Meine Anlage sieht ähnlich aus.

andig, 2025-02-18:
> https://github.com/evcc-io/evcc/issues/12746#issuecomment-1985206312

?

typxxi, 2025-03-21:
Das Angebot ist top, die Leistungen passen in unserer Lage recht genau und damit meine ich + / -5% von Mitternacht (Prognose) bis IST Ertrag zum Sonnenuntergang.

Das ist für uns im vergangenen Jahr die beste Prognose gewesen, wobei ich mir hier nur auf die Energieerzeugung pro Tag beziehe und insgesamt auf einen Zeitraum von etwas über 1 Jahr.

Daher wäre es Klasse, diese Alternative zu bekommen, da sie 
1. auch gratis war 
2. mehrere Strings und damit mehrere Ausrichtungen und Neigungen wie auch mppt abbilden kann
3. die Beachtung von Schatten modelliert und in der professionellen Version skyclear bietet

Ich habe seit unserer 2020 beginnenden HA Nutzung 
* den HA Service getestet genutzt gehabt, bis der abgelöst wude durch
* den bayrischen forecast solar und
* dann aus Australien, mittlerweile einem norwegischen Konzern gehörend SOLCAST
* evcc mit meteo und
* solarprognose

Letzteres hatte bezogen auf den  Tagesertrag über 1 Jahr gesehen die geringste Abweichung mit Jahren zwischen 3 und 5%. Ich schaute überhaupt nicht auf die stündliche Verteilung, was jetzt aber mehr und mehr ein Thema wird, weil die Wärmepumpe kommt und die Konkurrenz Auto & Wärmepumpe öfter zu einer dominanten Frage werden wird, was man wie priorisieren kann. 
Noch ist die WP auf dem Weg, aber spätestens im Herbst ist es so weit.

FelixK-creator, 2025-04-29:
Bitte wieder aufleben lassen! Bin auch Fan von solarprognose.de und würde gerne die Daten in evcc nutzen wollen zur Steuerung des Auto- und Batterieladens.

andig, 2025-04-30:
Gerne PR. Und natürlich kannst du das heute schon als `custom` Tarif nutzen.

andig, 2025-10-15:
> Gerne PR

Einfach Datei hier editieren!