evcc-io/evcc #12757: Dynamic charging power per phase

zigad, 2024-03-07:
**Is your feature request related to a problem? Please describe.**
I have 3 EVs at home. BMW i3 that has 3 phase charging, VW e-Golf and VW e-up that both have 2 phase charging.
Now I can only change amps per phase. So if I say my limit is 16A/phase, my VW cars will both charge at max speed at 2x 16A that means 7,1 kW and my i3 will also charge at max speed since it uses 3 phases and not 2 and will also apply 16A/phase setting meaning 3x16A = 11kW. Same goes with other settings. 

**Describe the solution you'd like**
I would love to have an option to say I want to draw around xy ampers on all phases. If I select for example 16A that means that EVCC will allow 16A on 1phase car, 8A/phase on 2 phase car and 5A or 6A on 3phase car. That means that all of my 3 vehicles will charge with example 7kWh or around that number. As I can see EVCC already knows that my VW is charging with 2 phases and i3 with 3 phases so I thing there is an option to add a new settings for max charging power no matter if i have a car that has 1, 2 or 3 phase charging.

**Describe alternatives you've considered**
I tried to set the charging limits on cars but cant even get close to the wanted result. If i could do this on EVCC i would have much better control.

**Additional context**
This should probably be an "advanced" feature or documented very well.

GrimmiMeloni, 2024-03-08:
see https://github.com/evcc-io/evcc/pull/8427

closed 2024-03-08 by andig

zigad, 2024-03-08:
Amazing, thank you @GrimmiMeloni!