evcc-io/evcc #12758: SMA + AVM DECT Steckdose Heizstab

SNeber89, 2024-03-07:
**Is your feature request related to a problem? Please describe.**
Ich habe einen SMA Sunny Homemanager, ein SMA Sunny Tripower, einen SMA Sunny Boy Storage, eine BYD Batterie, eine SMA und eine Easee Wallbox und eine AVM Dect Steckdose welche von SMA bei Überschuss angesteuert wird, an der Steckdose hängt ein 2kW Heizstab für Warmwasser

**Describe the solution you'd like**
Kann die AVM Steckdose auch über EVCC angesteuert werden? Aktuell läuft das über SMA, doch leider geht die Steckdose bei Überschuss vor die Wallboxen an welche von EVCC gesteuert werden. Die Wallboxen sollen aber bei Überschuss vor der Steckdose kommen.

**Describe alternatives you've considered**
Ich habe bei SMA ein Ticket eröffnet ob die AVM Dect Steckdose erst nach einer Verzögerung von z.B. "5 Minuten Überschuss vorhanden" angehen kann. 

**Additional context**
Add any other context or screenshots about the feature request here.

VolkerK62, 2024-03-08:
Ist dann eher eine Diskussion, als ein Issue.
Ja, das geht ohne Probleme.
Die Steckdose als Charger konfigurieren https://docs.evcc.io/docs/devices/chargers#avm-fritzdect
einen entsprechenden Loadpoint anlegen 
und `priority` nutzen https://docs.evcc.io/docs/reference/configuration/loadpoints#priority
Wenn dein Auto mehr als 2kW braucht um die Ladung zu starten, wird der Heizstab u.U. zwar vorher eingeschaltet, aber wenn incl. der Heizstableistung genug Überschuss vorhanden ist, schaltet der sich aus und das Auto ein.

closed 2024-03-08 by naltatis

SNeber89, 2024-03-08:
Es hat geklappt, vielen Dank für die gute und schnelle Antwort. Ein Hoch auf EVCC!