evcc-io/evcc #12763: Battery Mode schaltet bei Günstiges Netzladen hin und her

xerion3800, 2024-03-08:
### Describe the bug

Ich hatte gerade ein komisches Verhalten. PV Überschuss deutlich vorhanden und Tibber ist für ein LP unter die Preisgrenze gefallen. Da ich mir eine Überwachung des Battery Mode eingerichtet habe wann evcc die Batterie sperrt (Sungrow Speicher) bekam ich gerade ständig die Nachricht, das zwischen "hold" und "normal" geschaltet wird.

### Steps to reproduce

1. PV-Überschuss vorhanden
2. Andere LPs die auch am Laden sind über PV-Überschuss
3. In einem LP "Günstiges Netzladen" aktivieren und Grenze so einstellen das die unterhalb des aktuellen Preise liegt
4.  "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." unter Batterieeinstellungen aktivieren
5.
...

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.178.191
  port: 7070

log: debug

levels:
  # go: trace
  site: debug
  lp-1: debug
  lp-2: debug
  lp-3: debug
  lp-4: debug
  # mqtt: trace
  # modbus: trace

  
interval: 10s # control cycle interval

plant: ****

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: ***

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 
  - name: sungrow
    type: template
    template: sungrow-hybrid
    usage: pv     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.141 # Hostname
    port: 502 # Port 

  - name: plenticore
    type: template
    template: kostal-plenticore
    usage: pv      
    
    # Modbus TCP
    modbus: tcpip
[... 1474 lines cut ...]

2024-03-08T11:16:58.412387215Z [site  ] DEBUG 2024/03/08 12:16:58 grid meter: 5169W

2024-03-08T11:16:58.414381802Z [site  ] DEBUG 2024/03/08 12:16:58 giving loadpoint priority for additional: 12618W

2024-03-08T11:16:58.414411612Z [site  ] DEBUG 2024/03/08 12:16:58 site power: -7149W

2024-03-08T11:16:58.414556974Z [lp-1  ] DEBUG 2024/03/08 12:16:58 charge currents: [0 0 0]A

2024-03-08T11:16:58.414766406Z [lp-1  ] DEBUG 2024/03/08 12:16:58 charge total import: 4960.934kWh

2024-03-08T11:16:58.415338273Z [lp-1  ] DEBUG 2024/03/08 12:16:58 charger status: A

2024-03-08T11:16:58.424665307Z [site  ] DEBUG 2024/03/08 12:16:58 set battery mode: normal

2024-03-08T11:17:01.533418761Z [site  ] DEBUG 2024/03/08 12:17:01 set battery discharge control: false
```

### What type of operating system are you running?

Docker container

### Version

https://github.com/evcc-io/evcc/commit/b2921078

andig, 2024-03-08:
Der Battery Mode hängt am Ladestatus der zwischen A und C springt. Aber wo kommt das her?

mucki12, 2024-03-08:
Kann das Verhalten bestätigen (0.124.7). Im Modus Schnell wird die Batteriesperre aufrecht erhalten und im Modus PV (mit aktivem smartcostactive) wird dauerhaft zwischen hold<->normal gewechselt.

andig, 2024-03-08:
Der Fehler entsteht durch https://github.com/evcc-io/evcc/pull/12135 bei >1 Ladepunkt weil der Battery Mode jetzt je LP evaluiert wird. Tatsächlich müssen aber alle Ladepunkte gleichzeitig betrachtet werden.

/cc @naltatis

mucki12, 2024-03-08:
Tatsächlich - auf diese Idee bin ich gar nicht gekommen. Habe jetzt mal die Preisgrenze auf alle LP angewendet und dann ist das Verhalten wie gewünscht (dauerhaft hold). 

Wenn ich ein Stimmrecht hätte würde ich mir tatsächlich wünschen, dass die Batteriesperre sofort greift wenn lediglich ein LP auf smartcostactive=true steht. Bin da aber nicht so tief im Thema wie das für die Zukunft angedacht ist. Was passiert zB später mal wenn ein LP auf smartcostactive=true steht und die Funktion smartes Laden des Hausspeichers kommen sollte? Schlägt letzter dann die erste Einstellung?

closed 2024-03-08 by andig