evcc-io/evcc #12765: Probleme beim Überschußladen mit SmartEQ (Bj.2021)

michaestraub, 2024-03-08:
### Describe the bug

Hi,
vorne weg ich habe 2 Easee Wallboxen, beim Mercedes C300e funktioniert das Überschuss laden problemlos.

Beim Smart habe ich schon von Anfang an das Problem, daß nach dem Phasenumschalten das Laden nicht fortgesetzt wird. Da bin ich leider nie weiter gekommen.
Nun funktioniert das Überschußladen beim Smart leider nicht mehr.

Mit 0.124.6 und 7 getestet, gleiches verhalten

### Steps to reproduce

- Überschuß vorhanden, Warte auf Fahrzeug, Laden beginnt nicht.
- Ich schalte von AUS auf PV, nun wird der Überschuß erkannt, und der Contdown zählt runter von 60 Sekunden
- Nun beginnt  der Ladevorgang kurz mit 11KW (Warum auch immer) fällt dann ab und meldet zu wenig Überschuß.
- Kurz danach schaltet der Ladevorgang wieder ab mit der Meldung. Ladebereit Warte auf Fahrzeug.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: 10.0.0.39 # .local suffix announces the hostname on MDNS
  port: 7070

#uri: 0.0.0.0:7070

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price:  # [currency]/kWh

  feedin:
    type: fixed
    price:  # [currency]/kWh

log: trace
levels:
  cache: error

# unique installation id
plant: 

interval: 10s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: grid1
  type: template
  template: openems 
  usage: grid  
  host: 10.0.0.30  
  password: user  
  
- name: pv2
  type: template
  template: openems 
  usage: pv  
  host: 10.0.0.30  
  password: user  
  
- name: battery3
  type: template
[... 298 lines cut ...]
Mar 08 11:59:53 evcc evcc[902]: [site  ] DEBUG 2024/03/08 11:59:53 battery soc: 100%
Mar 08 11:59:53 evcc evcc[902]: [site  ] DEBUG 2024/03/08 11:59:53 battery power: -45W
Mar 08 11:59:53 evcc evcc[902]: [http  ] TRACE 2024/03/08 11:59:53 GET http://10.0.0.30/rest/channel/_sum/GridActivePower
Mar 08 11:59:53 evcc evcc[902]: [http  ] TRACE 2024/03/08 11:59:53 {"address":"_sum/GridActivePower","type":"INTEGER","accessMode":"RO","text":"Grid exchange power. Negative values for sell-to-grid; positive for buy-from-grid","unit":"W","value":-3549}
Mar 08 11:59:53 evcc evcc[902]: [site  ] DEBUG 2024/03/08 11:59:53 grid meter: -3549W
Mar 08 11:59:53 evcc evcc[902]: [site  ] DEBUG 2024/03/08 11:59:53 site power: -3594W
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 charge currents: [0.008 0.008 0.008]A
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 charge total import: 1095.125kWh
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 charger status: B
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 pv charge current: 15.6A = 0A + 15.6A (-3594W @ 1p)
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 site power -3594W <= 0W enable threshold
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 pv enable in 0s
Mar 08 11:59:53 evcc evcc[902]: [lp-1  ] DEBUG 2024/03/08 11:59:53 pv enable timer elapsed
Mar 08 11:59:53 evcc evcc[902]: [easee ] TRACE 2024/03/08 11:59:53 POST https://api.easee.com/api/chargers/EHKPQDYG/settings
Mar 08 11:59:54 evcc evcc[902]: [easee ] TRACE 2024/03/08 11:59:54 {"dynamicChargerCurrent":6}
Mar 08 11:59:54 evcc evcc[902]: --
```

### What type of operating system are you running?

Linux

### Version

0.124.6 and 0.124.7

rothdani, 2024-03-08:
Hallo,

Der Smart EQ for two bzw. for four ist eine absolute Ladezicke, ich hatte mit meinem auch anfangs massive Probleme. Phasenumschaltung ist grundsätzlich so eine Sache wo ich gar nicht weis, ob er das verträgt. Bei OpenWB war bei den Smarts und den Twingos (baugleich) explizit der Hinweis, dass diese für Phasenumschaltung nicht geeignet sind.
Ich lade meinen for two mit PV Überschuss nur 3-Phasig. Was bei mir geholfe hat, dass er auch konstant läd war die Mindeststromstärke hoch zu setzen. Mit 6A kommt mein Smart nicht klar.
Ich habe bei meinem Smart im evcc.yaml noch "mincurrent: 8" unter vehicles ergänzt, damit funktioniert das Überschussladen ohne Unterbrechungen (außer du hast natürlich zu wenig PV Strom).

michaestraub, 2024-03-08:
Danke Dir, super Hinweis, ich werde testen :-)  Dankeschön