evcc-io/evcc #12773: Easee Phasenumschaltung und Logic Error

Cytron1980, 2024-03-08:
### Describe the bug

Dies ist mein erstes Issue. Sollte also etwas nicht richtig angegeben oder falsch formatiert sein, so lasst es mich bitte wissen.
zum Problem:
Modus PV: Easee Home lädt bei einstecken des ID4 und Freischalten mittels RFID mit voller Leistung (dies wird über die Fehlermeldung "logic error: disabled but charging" angezeigt. Nach einer kurzen Zeit (ich denke mal, bis vom Server bei Easee eine Rückmeldung kommt) geht EVCC in den Standby. Sobald Überschuss da ist, wird angefangen zu Laden. Allerdings nicht wie in der Config und der WB selber, angegeben einphasig sondern direkt mit 4,7kW. Dadurch ist dann wieder ein Mangel an Überschuss, was die Ladung nach Zeit X unterbricht, nur um dann wieder festzuzstellen, dass wieder Überschuss vorhanden ist, wodurch das Prozedere von vorne beginnt. Es scheint also ein Problem mit der Umschaltung der Phasen zu geben, was durch die Meldung "ignoring inconsistent phases: 1p < 3p observed active" angezeigt wird.

Einen Log kann ich gerade nicht einfügen, da gerade nach einem Neustart von EVCC alles so läuft wie es soll, werde ich aber 

[image]

### Steps to reproduce

1.Starte Laden

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 40e7a34a253eb356e07322451451f552e79457aa1fd68cf282472088ac0f1c4b

interval: 10s # control cycle interval

sponsortoken: XXXXXXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.2.201  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: easee 
  user: XXXteme.com  
  password: XXX  
  charger: XXX 
[... 639 lines cut ...]
[lp-1  ] DEBUG 2024/03/08 16:08:59 pv enable in 1m0s
[site  ] DEBUG 2024/03/08 16:09:05 ----
[lp-1  ] DEBUG 2024/03/08 16:09:05 charge power: 0W
[site  ] DEBUG 2024/03/08 16:09:05 pv power: 1934W
[site  ] DEBUG 2024/03/08 16:09:05 battery soc: 98%
[site  ] DEBUG 2024/03/08 16:09:05 battery power: -1617W
[site  ] DEBUG 2024/03/08 16:09:05 grid meter: -16W
[site  ] DEBUG 2024/03/08 16:09:05 site power: -1633W
[lp-1  ] DEBUG 2024/03/08 16:09:05 charge currents: [0.009 0.011 0.011]A
[lp-1  ] DEBUG 2024/03/08 16:09:05 charge total import: 2337.757kWh
[lp-1  ] DEBUG 2024/03/08 16:09:05 charger status: B
[lp-1  ] DEBUG 2024/03/08 16:09:05 pv charge current: 7.1A = 0A + 7.1A (-1633W @ 1p)
[lp-1  ] DEBUG 2024/03/08 16:09:05 site power -1633W <= 0W enable threshold
[lp-1  ] DEBUG 2024/03/08 16:09:05 pv enable in 53s
[lp-1  ] DEBUG 2024/03/08 16:09:05 pv enable timer remaining: 53s
[lp-1  ] DEBUG 2024/03/08 16:09:09 charge total import: 2337.757kWh
```

### What type of operating system are you running?

Windows

### Version

0.124.6

GrimmiMeloni, 2024-03-08:
@Cytron1980 für die Analyse benötige ich von Dir ein Log mit Easee auf Trace.

```
log: debug
levels:
  cache: error
  easee: trace
```

Cytron1980, 2024-03-08:
Werde ich beim nächsten Überschuss erstellen.

Cytron1980, 2024-03-10:
Hier mal der Log von gerade eben. Nach hause gekommen, Auto angeschlossen, Klimatisierung erkannt (warum weiß ich nicht im Auto ist nichts aktiv) und er hat die Meldung ausgespuckt, dass keine Phasenumschaltung erfolgen konnte...

[lp-1  ] DEBUG 2024/03/10 09:47:06 charge power: 10462W
[site  ] DEBUG 2024/03/10 09:47:06 pv power: 1169W
[site  ] DEBUG 2024/03/10 09:47:06 battery soc: 33%
[site  ] DEBUG 2024/03/10 09:47:06 battery power: 4997W
[site  ] DEBUG 2024/03/10 09:47:06 grid meter: 19W
[site  ] DEBUG 2024/03/10 09:47:06 site power: 5016W
[lp-1  ] DEBUG 2024/03/10 09:47:06 charge currents: [15.2 15.3 15.2]A
[lp-1  ] DEBUG 2024/03/10 09:47:06 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/10 09:47:06 charge total import: 2365.780kWh
[lp-1  ] DEBUG 2024/03/10 09:47:06 charger status: C
[lp-1  ] DEBUG 2024/03/10 09:47:06 soc estimated: 24.00% (vehicle: 24.00%)
[lp-1  ] DEBUG 2024/03/10 09:47:06 vehicle soc: 24%
[lp-1  ] DEBUG 2024/03/10 09:47:06 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/10 09:47:06 vehicle range: 61km
[lp-1  ] WARN 2024/03/10 09:47:06 ignoring inconsistent phases: 1p < 3p observed active
[lp-1  ] DEBUG 2024/03/10 09:47:06 available power 5446W > 4140W min 3p threshold
[lp-1  ] DEBUG 2024/03/10 09:47:06 start phase scale3p timer
[lp-1  ] DEBUG 2024/03/10 09:47:06 phase scale3p in 1m0s
[lp-1  ] DEBUG 2024/03/10 09:47:06 pv charge current: 0A = 6A + -21.8A (5016W @ 1p)
[lp-1  ] DEBUG 2024/03/10 09:47:06 projected site power 5016W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/10 09:47:06 pv disable in 0s
[lp-1  ] DEBUG 2024/03/10 09:47:06 pv disable timer elapsed
[lp-1  ] DEBUG 2024/03/10 09:47:06 climater active
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) IN_CURRENT_T3 6.0279998779296875
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) IN_CURRENT_T4 6.01200008392334
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) IN_CURRENT_T5 6.02400016784668
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) TOTAL_POWER 4.176000118255615
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 6.0279998779296875
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L2 6.01200008392334
[easee ] TRACE 2024/03/10 09:47:07 ProductUpdate EHNRT639: (2024-03-10 08:47:07 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L3 6.02400016784668
[site  ] DEBUG 2024/03/10 09:47:16 ----
[lp-1  ] DEBUG 2024/03/10 09:47:16 charge power: 4176W
[site  ] DEBUG 2024/03/10 09:47:16 pv power: 1133W
[site  ] DEBUG 2024/03/10 09:47:16 battery soc: 33%
[site  ] DEBUG 2024/03/10 09:47:16 battery power: 5002W
[site  ] DEBUG 2024/03/10 09:47:16 grid meter: -3W
[site  ] DEBUG 2024/03/10 09:47:16 site power: 4999W
[lp-1  ] DEBUG 2024/03/10 09:47:16 charge currents: [6.03 6.01 6.02]A
[lp-1  ] DEBUG 2024/03/10 09:47:16 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/10 09:47:16 charge total import: 2365.780kWh
[lp-1  ] DEBUG 2024/03/10 09:47:16 charger status: C
[lp-1  ] DEBUG 2024/03/10 09:47:16 soc estimated: 24.00% (vehicle: 24.00%)
[lp-1  ] DEBUG 2024/03/10 09:47:16 vehicle soc: 24%
[lp-1  ] DEBUG 2024/03/10 09:47:16 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/10 09:47:16 vehicle range: 61km
[lp-1  ] WARN 2024/03/10 09:47:16 ignoring inconsistent phases: 1p < 3p observed active
[lp-1  ] DEBUG 2024/03/10 09:47:16 phase timer inactive
[lp-1  ] DEBUG 2024/03/10 09:47:16 pv charge current: 0A = 6A + -21.7A (4999W @ 1p)
[lp-1  ] DEBUG 2024/03/10 09:47:16 projected site power 4999W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/10 09:47:16 pv disable in 0s
[lp-1  ] DEBUG 2024/03/10 09:47:16 pv disable timer elapsed
[lp-1  ] DEBUG 2024/03/10 09:47:16 climater active
[site  ] DEBUG 2024/03/10 09:47:26 ----
[lp-1  ] DEBUG 2024/03/10 09:47:26 charge power: 4176W
[site  ] DEBUG 2024/03/10 09:47:26 pv power: 1151W
[site  ] DEBUG 2024/03/10 09:47:26 battery soc: 32%
[site  ] DEBUG 2024/03/10 09:47:26 battery power: 4998W
[site  ] DEBUG 2024/03/10 09:47:26 grid meter: 33W
[site  ] DEBUG 2024/03/10 09:47:26 site power: 5031W
[lp-1  ] DEBUG 2024/03/10 09:47:26 charge currents: [6.03 6.01 6.02]A
[lp-1  ] DEBUG 2024/03/10 09:47:26 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/10 09:47:26 charge total import: 2365.780kWh
[lp-1  ] DEBUG 2024/03/10 09:47:26 charger status: C
[lp-1  ] DEBUG 2024/03/10 09:47:26 soc estimated: 24.00% (vehicle: 24.00%)
[lp-1  ] DEBUG 2024/03/10 09:47:26 vehicle soc: 24%
[lp-1  ] DEBUG 2024/03/10 09:47:26 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/10 09:47:26 vehicle range: 61km
[lp-1  ] WARN 2024/03/10 09:47:26 ignoring inconsistent phases: 1p < 3p observed active
[... 320 lines cut ...]
[site  ] DEBUG 2024/03/10 09:49:56 battery power: 5000W
[site  ] DEBUG 2024/03/10 09:49:56 grid meter: 49W
[site  ] DEBUG 2024/03/10 09:49:56 site power: 5049W
[lp-1  ] DEBUG 2024/03/10 09:49:56 charge currents: [6.03 6.01 6.02]A
[lp-1  ] DEBUG 2024/03/10 09:49:56 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/10 09:49:56 charge total import: 2365.780kWh
[lp-1  ] DEBUG 2024/03/10 09:49:56 charger status: C
[lp-1  ] DEBUG 2024/03/10 09:49:56 soc estimated: 24.00% (vehicle: 24.00%)
[lp-1  ] DEBUG 2024/03/10 09:49:56 vehicle soc: 24%
[lp-1  ] DEBUG 2024/03/10 09:49:56 vehicle soc limit: 80%
[... cut ...]
[easee ] TRACE 2024/03/10 09:49:59 ProductUpdate EHNRT639: (2024-03-10 08:49:59 +0000 UTC) TEMP_INPUT_T5 15
[easee ] TRACE 2024/03/10 09:49:59 ProductUpdate EHNRT639: (2024-03-10 08:49:59 +0000 UTC) TEMP_OUTPUT_N 15
[easee ] TRACE 2024/03/10 09:49:59 ProductUpdate EHNRT639: (2024-03-10 08:49:59 +0000 UTC) TEMP_OUTPUT_L1 15
[easee ] TRACE 2024/03/10 09:49:59 ProductUpdate EHNRT639: (2024-03-10 08:49:59 +0000 UTC) TEMP_OUTPUT_L3 15
[easee ] TRACE 2024/03/10 09:49:59 ProductUpdate EHNRT639: (2024-03-10 08:49:59 +0000 UTC) TEMP_OUTPUT_L2 15

GrimmiMeloni, 2024-03-10:
Der Climater ist hier erstmal nur Nebelkerze und verhindert lediglich das Abschalten.

@andig hier gibt es ein Problem mit der Logik im Loadpoint.
Wenn die phases als inconsistent erkannt werden, resetten wir `measuredPhases` auf 0.
Das führt dann aber automatisch dazu dass activePhases in 

https://github.com/evcc-io/evcc/blob/12f3fbf173de8921a184719c3e90b161f9b2d048/core/loadpoint.go#L1053

immer 1 ist, und somit ein Schalten der Phasen verhindert wird. Aus der Situation kommt man aber nicht mehr raus, weil der Loadpoint auf 1 beharrt, und der Charger korrekt immer wieder 3 melden wird.

Mir erscheint es generell nicht richtig, das Loadpoint.phases bei der Ermittlung von "activePhases" herangezogen wird. Eigentlich kennt doch nur der Charger die Wahrheit...?

andig, 2024-03-11:
> Wenn die phases als inconsistent erkannt werden, resetten wir measuredPhases auf 0.

Hilf mir mal bitte- wo passiert das? Vllt. sollten wir an der Stelle für einen 1p3p Charger zusätzlich auch wieder auf "auto" umschalten?

> Mir erscheint es generell nicht richtig, das Loadpoint.phases bei der Ermittlung von "activePhases" herangezogen wird. 

Im Zweifel ist das halt der letzte bekannte Stand. Ich frag mich ja eher, wie wir hier überhaupt in die inkonsistente Situation kommen?

GrimmiMeloni, 2024-03-11:
> > Wenn die phases als inconsistent erkannt werden, resetten wir measuredPhases auf 0.
> 
> Hilf mir mal bitte- wo passiert das? 

[Hier](https://github.com/evcc-io/evcc/blob/e6f0d61c5b623ae571b6142b9014a9f7ffa19bc2/core/loadpoint_phases.go#L34). 

> Vllt. sollten wir an der Stelle für einen 1p3p Charger zusätzlich auch wieder auf "auto" umschalten?

Ich glaube das ist das, was die 0 ausdrücken soll? (Analog zur Konfiguration?)

> > Mir erscheint es generell nicht richtig, das Loadpoint.phases bei der Ermittlung von "activePhases" herangezogen wird.
> 
> Im Zweifel ist das halt der letzte bekannte Stand. 

Ja, das ist auch OK. Das ist Problem ist, das die Logik in [activePhases()](https://github.com/evcc-io/evcc/blob/e6f0d61c5b623ae571b6142b9014a9f7ffa19bc2/core/loadpoint_phases.go#L61) mit der 0 nicht gut umgehen kann.

> Ich frag mich ja eher, wie wir hier überhaupt in die inkonsistente Situation kommen?

Das konnte ich aus dem Log nicht erkennen, weil es erst beginnt wenn dieser Zustand schon erreicht ist. @Cytron1980 gibt es die Chance auf ein vollständiges Log, oder ist das schon gelöscht?

Cytron1980, 2024-03-11:
Das Log vom besagten Tag wurde schon gelöscht. Da der Fehler sich aber reproduzieren lässt, werde ich sobald er wieder auftritt ( Auto ist gerade halbwegs voll) den Log hier einstellen...

andig, 2024-03-13:
Ich mache mal zu bis es ein Logfile gibt.

closed 2024-03-13 by andig

Tonno87, 2024-03-17:
Ich hab das gleiche Problem, ziemlich nervig... Muss man das loglevel auf Trace setzen? Denn debug liefert nicht wirklich was brauchbares.

andig, 2024-03-17:
Das würde helfen. Nur für die Easee.

Cytron1980, 2024-03-17:
Der Fehler mit der Phasenumschaltung ist bei mir nicht mehr vorgekommen. Nur der Logic Error kommt bei jedem Start nach wie vor. Log werde ich nachher noch erstellen.

Cytron1980, 2024-03-21:
Hier ein neues Log von Heute:
habe den priority soc auf 75 gesetzt und Ladung hat 3 Phasig mit 4,7kW angefangen zu laden ohne die Phasen auf einphasig zu steuern (bzw. hat die Easee dies nicht gemacht). Dann erkannt, dass der Überschuss zu gering ist (PV-Leistung war bei ca 2-3kW) um die Ladung nach 3 Minuten zu beenden und dann wieder einen Überschuss erkannt. Somit dann das ganze Prozedere von vorne.

[lp-1  ] DEBUG 2024/03/21 11:07:46 charge power: 0W
[site  ] DEBUG 2024/03/21 11:07:46 pv power: 2814W
[site  ] DEBUG 2024/03/21 11:07:46 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:07:46 battery power: -2623W
[site  ] DEBUG 2024/03/21 11:07:46 grid meter: 34W
[site  ] DEBUG 2024/03/21 11:07:46 battery has priority at soc 76% (< 95%)
[site  ] DEBUG 2024/03/21 11:07:46 site power: 34W
[lp-1  ] DEBUG 2024/03/21 11:07:46 charge currents: [0.009 0.01 0.01]A
[lp-1  ] DEBUG 2024/03/21 11:07:46 charge total import: 2454.610kWh
[lp-1  ] DEBUG 2024/03/21 11:07:46 charger status: B
[lp-1  ] DEBUG 2024/03/21 11:07:46 pv charge current: 0A = 0A + -0.148A (34W @ 1p)
[site  ] DEBUG 2024/03/21 11:07:51 set priority soc: 75
[site  ] DEBUG 2024/03/21 11:07:56 ----
[lp-1  ] DEBUG 2024/03/21 11:07:56 charge power: 0W
[site  ] DEBUG 2024/03/21 11:07:56 pv power: 2759W
[site  ] DEBUG 2024/03/21 11:07:56 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:07:56 battery power: -2579W
[site  ] DEBUG 2024/03/21 11:07:56 grid meter: 33W
[site  ] DEBUG 2024/03/21 11:07:56 site power: -2546W
[lp-1  ] DEBUG 2024/03/21 11:07:56 charge currents: [0.009 0.01 0.01]A
[lp-1  ] DEBUG 2024/03/21 11:07:56 charge total import: 2454.610kWh
[lp-1  ] DEBUG 2024/03/21 11:07:56 charger status: B
[lp-1  ] DEBUG 2024/03/21 11:07:56 pv charge current: 11.1A = 0A + 11.1A (-2546W @ 1p)
[lp-1  ] DEBUG 2024/03/21 11:07:56 site power -2546W <= 0W enable threshold
[lp-1  ] DEBUG 2024/03/21 11:07:56 pv enable timer start: 1m0s
[lp-1  ] DEBUG 2024/03/21 11:07:56 pv enable in 1m0s
[site  ] DEBUG 2024/03/21 11:08:06 ----
[lp-1  ] DEBUG 2024/03/21 11:08:06 charge power: 0W
[site  ] DEBUG 2024/03/21 11:08:06 pv power: 2769W
[site  ] DEBUG 2024/03/21 11:08:06 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:08:06 battery power: -2555W
[site  ] DEBUG 2024/03/21 11:08:06 grid meter: 21W
[site  ] DEBUG 2024/03/21 11:08:06 site power: -2534W
[lp-1  ] DEBUG 2024/03/21 11:08:06 charge currents: [0.009 0.01 0.01]A
[lp-1  ] DEBUG 2024/03/21 11:08:06 charge total import: 2454.610kWh
[lp-1  ] DEBUG 2024/03/21 11:08:06 charger status: B
[lp-1  ] DEBUG 2024/03/21 11:08:06 pv charge current: 11A = 0A + 11A (-2534W @ 1p)
[lp-1  ] DEBUG 2024/03/21 11:08:06 site power -2534W <= 0W enable threshold
[lp-1  ] DEBUG 2024/03/21 11:08:06 pv enable in 50s
[lp-1  ] DEBUG 2024/03/21 11:08:06 pv enable timer remaining: 50s
[site  ] DEBUG 2024/03/21 11:08:16 ----
[lp-1  ] DEBUG 2024/03/21 11:08:16 charge power: 0W
[site  ] DEBUG 2024/03/21 11:08:16 pv power: 2657W
[site  ] DEBUG 2024/03/21 11:08:16 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:08:16 battery power: -2459W
[site  ] DEBUG 2024/03/21 11:08:16 grid meter: 36W
[site  ] DEBUG 2024/03/21 11:08:16 site power: -2423W
[lp-1  ] DEBUG 2024/03/21 11:08:16 charge currents: [0.009 0.01 0.01]A
[lp-1  ] DEBUG 2024/03/21 11:08:16 charge total import: 2454.610kWh
[lp-1  ] DEBUG 2024/03/21 11:08:16 charger status: B
[lp-1  ] DEBUG 2024/03/21 11:08:16 pv charge current: 10.5A = 0A + 10.5A (-2423W @ 1p)
[lp-1  ] DEBUG 2024/03/21 11:08:16 site power -2423W <= 0W enable threshold
[lp-1  ] DEBUG 2024/03/21 11:08:16 pv enable in 40s
[lp-1  ] DEBUG 2024/03/21 11:08:16 pv enable timer remaining: 40s
[easee ] TRACE 2024/03/21 11:08:18 ProductUpdate EHNRT639: (2024-03-21 10:00:00 +0000 UTC) LIFETIME_HOURS 6946
[easee ] TRACE 2024/03/21 11:08:18 ProductUpdate EHNRT639: (2024-03-21 10:00:00 +0000 UTC) LIFETIME_ENERGY 2454.6103577777785[easee ] TRACE 2024/03/21 11:08:19 ProductUpdate EHNRT639: (2024-03-21 10:00:00 +0000 UTC) ENERGY_PER_HOUR 0
[site  ] DEBUG 2024/03/21 11:08:26 ----
[lp-1  ] DEBUG 2024/03/21 11:08:26 charge power: 0W
[site  ] DEBUG 2024/03/21 11:08:26 pv power: 2570W
[site  ] DEBUG 2024/03/21 11:08:26 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:08:26 battery power: -2365W
[site  ] DEBUG 2024/03/21 11:08:26 grid meter: 1W
[site  ] DEBUG 2024/03/21 11:08:26 site power: -2364W
[lp-1  ] DEBUG 2024/03/21 11:08:26 charge currents: [0.009 0.01 0.01]A
[lp-1  ] DEBUG 2024/03/21 11:08:26 charge total import: 2454.610kWh
[lp-1  ] DEBUG 2024/03/21 11:08:26 charger status: B
[lp-1  ] DEBUG 2024/03/21 11:08:26 pv charge current: 10.3A = 0A + 10.3A (-2364W @ 1p)
[... 618 lines cut ...]
[lp-1  ] DEBUG 2024/03/21 11:12:36 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/21 11:12:36 vehicle range: 191km
[lp-1  ] DEBUG 2024/03/21 11:12:36 pv charge current: 0A = 6A + -6.91A (1590W @ 1p)
[lp-1  ] DEBUG 2024/03/21 11:12:36 projected site power 1590W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/21 11:12:36 pv disable in 1m34s
[lp-1  ] DEBUG 2024/03/21 11:12:36 pv disable timer remaining: 1m34s
[site  ] DEBUG 2024/03/21 11:12:46 ----
[lp-1  ] DEBUG 2024/03/21 11:12:46 charge power: 4162W
[site  ] DEBUG 2024/03/21 11:12:46 pv power: 2824W
[site  ] DEBUG 2024/03/21 11:12:46 battery soc: 76%
[site  ] DEBUG 2024/03/21 11:12:46 battery power: 1649W
[site  ] DEBUG 2024/03/21 11:12:46 grid meter: -11W
[site  ] DEBUG 2024/03/21 11:12:46 site power: 1638W
[lp-1  ] DEBUG 2024/03/21 11:12:46 charge currents: [5.91 5.99 6.04]A
[... cut ...]
[lp-1  ] DEBUG 2024/03/21 11:12:46 soc estimated: 48.00% (vehicle: 48.00%)
[lp-1  ] DEBUG 2024/03/21 11:12:46 vehicle soc: 48%
[lp-1  ] DEBUG 2024/03/21 11:12:46 vehicle soc limit: 80%
[lp-1  ] DEBUG 2024/03/21 11:12:46 vehicle range: 191km
[lp-1  ] DEBUG 2024/03/21 11:12:46 pv charge current: 0A = 6A + -7.12A (1638W @ 1p)
[lp-1  ] DEBUG 2024/03/21 11:12:46 projected site power 1638W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/21 11:12:46 pv disable in 1m24s
[lp-1  ] DEBUG 2024/03/21 11:12:46 pv disable timer remaining: 1m24s

Tonno87, 2024-03-21:
Vielleicht packst du das besser in eine Textdatei.... So ist das schon mühselig.

Ich konnte bisher kein Log erstellen weil ich das HA Add-on benutze. Das erstellt kein Log und ich muss dann immer alles von Hand rauskopieren. 
Am Wochenende stelle ich das mal um auf eine VM.

Jedenfalls hatte ich das jetzt auch wieder: man hat Überschuss von 3kw und er startet die Ladung mit 3p und beeendet sie dann wieder wegen zu wenig Überschuss.
Deshalb hab ich aktuell den Lademodus auf 1p festgesetzt und lade dann halt direkt einphasig.

Cytron1980, 2024-03-21:
Hier das Log als Textdatei
[log.txt](https://github.com/evcc-io/evcc/files/14693913/log.txt)

Cytron1980, 2024-03-21:
> Vielleicht packst du das besser in eine Textdatei.... So ist das schon mühselig.
>
> Ich konnte bisher kein Log erstellen weil ich das HA Add-on benutze. Das erstellt kein Log und ich muss dann immer [...]

Danke für den Tip, habe meinen Eintrag editiert.
Habe es jetzt auch mal auf 1-Phasig gestellt. Dieses funktioniert allerdings genauso wenig, da ja die Easee anscheinend ein Problem bei der Umschaltung des Phasenmodus von 3 auf 1-Phasig hat.

Tonno87, 2024-03-21:
In der Easee App kannst du das ebenfalls vorgeben bei den Einstellungen des Laderoboters:

[image]

Cytron1980, 2024-03-21:
> In der Easee App kannst du das ebenfalls vorgeben bei den Einstellungen des Laderoboters:
> 
> ![Screenshot_20240321-122609.png](https://private-user-images.githubusercontent.com/59800400/315268908-056a2b92-6e8a-4921-8f80-9312ddd6ba71.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwMjA2ODUsIm5iZiI6MTcxMTAyMDM4NSwicGF0aCI6Ii81OTgwMDQwMC8zMTUyNjg5MDgtMDU2YTJiOTItNmU4YS00OTIxLThmODAtOTMxMmRkZDZiYTcxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIxVDExMjYyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4NDQ1YmYyNjI2MGUzYjk5YTQ0YmM4ZjFjMGQ3MWI1NzQ3NjJhODI5OGI4Y2JkM2E2ZTI5YjdhZTE3MzNlNjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.yIVIMitki8CSC4c2vBVfIxSXQdHiPVovPwk3H4N1COc)

So hab ich es jetzt auch erstmal gelöst. Danke für den Tip. Habe nach beginn der 1-Phasigen Ladung jetzt direkt mal wieder auf Automatisch gestellt. Mal sehen, was EVCC dann macht, wenn auf 3-Phasig umgestellt werden soll.

Cytron1980, 2024-03-21:
Neuer Log: 
nach dem Umschalten in der Easee Einstellung von Automatisch auf Einphasig ,dem Starten des PV-Modus in EVCC und dem erneuten aktivieren der automatischen Phasensteuerung in der Easee selber, scheint die Funktion wieder zu klappen. 

Um ca 13:08 wurde automatisch von 3 auf 1-Phasig umgestellt.
[log nach umschaltung.txt](https://github.com/evcc-io/evcc/files/14695051/log.nach.umschaltung.txt)

GrimmiMeloni, 2024-03-21:
@Cytron1980 zwecks Klarheit. Ich brauche zur Analyse wenn der Fehler aufgetreten ist ein vollständiges(!) Log. Also, ab Startup von evcc. Ausschnitte helfen nicht, weil dann nicht klar ist, in welchem Zustand die Wallbox ist wenn die Fehler auftreten.

Cytron1980, 2024-03-21:
Werde morgen zu Beginn der Überschuss-Periode EVCC neu starten und nach Auftreten des Fehlers hier einen kompletten Log einstellen.

Cytron1980, 2024-03-22:
Kurze Rückmeldung. Gerade hat es ohne Probleme funktioniert. Ladung einphasig begonnen, obwohl sowohl Easee als auch EVCC auf automatische Phasenumschaltung steht. Warte die nächste Zeit ab und werde sobald der Fehler wieder auftritt hier das komplette Log hoch laden.

Cytron1980, 2024-03-25:
Hallo, Gerade nach Hause gekommen und nach dem einstecken und starten direkt den Fehler provoziert. Danach EVCC neu gestertet und PV-Modus wieder aktiviert. Sofort wieder auf 3 Phasig gegangen. Easee erstmal auf Ein-Phasig gestellt.

[Log25_03_24.txt](https://github.com/evcc-io/evcc/files/14745628/Log25_03_24.txt)

Cytron1980, 2024-03-27:
> Ich mache mal zu bis es ein Logfile gibt.

Sollte der Fall wieder aufgemacht werden? Aktuelle Logfile habe ich vorgestern eingestellt.
Danke

andig, 2024-03-28:
@GrimmiMeloni willst du da rein schauen?

GrimmiMeloni, 2024-03-28:
Nach dem Neustart:
Loadpoint macht korrekten Scale down.
Easee bestätigt das auch und geht für alle 3 Phasen von 40,40,40 auf 16,0,0 Amps.

```
[lp-1  ] DEBUG 2024/03/25 15:46:24 phase scale1p in 0s
[easee ] TRACE 2024/03/25 15:46:24 GET https://api.easee.com/api/sites/660895/circuits/624692/settings
[easee ] TRACE 2024/03/25 15:46:25 {"allowOfflineMaxCircuitCurrent":false,"equalizerId":null,"masterChargerId":"EHNRT639","useDynamicMaster":false,"dynamicCircuitCurrentP1":40.0,"dynamicCircuitCurrentP2":40.0,"dynamicCircuitCurrentP3":40.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":16,"offlineMaxCircuitCurrentP2":16,"offlineMaxCircuitCurrentP3":16}
[easee ] TRACE 2024/03/25 15:46:25 POST https://api.easee.com/api/sites/660895/circuits/624692/settings
[easee ] TRACE 2024/03/25 15:46:25 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":0,"dynamicCircuitCurrentP3":0}
...
[easee ] TRACE 2024/03/25 15:46:26 ProductUpdate EHNRT639: (2024-03-25 14:46:26 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 16
[easee ] TRACE 2024/03/25 15:46:26 ProductUpdate EHNRT639: (2024-03-25 14:46:26 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 0
[easee ] TRACE 2024/03/25 15:46:26 ProductUpdate EHNRT639: (2024-03-25 14:46:26 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 0
```

Bisher alles gut. Kurz drauf schaltet aber die Easee ohne das evcc dazu Kommandos sendet wieder auf 40A hoch. Den Fehler im Error String `no schedule to override, but restoring dynamic current anyway` habe ich so erstmal noch nie gesehen. 

```
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:42 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 40
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) ERROR_STRING no schedule to override, but restoring dynamic current anyway
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 40
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:42 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 40
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 40
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:42 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 40
[easee ] TRACE 2024/03/25 15:46:41 ProductUpdate EHNRT639: (2024-03-25 14:46:41 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 40
[easee ] TRACE 2024/03/25 15:46:42 CommandResponse EHNRT639: {SerialNumber:EHNRT639 ID:22 Timestamp:2024-03-25 14:46:42.10673 +0000 UTC DeliveredAt:2024-03-25 14:46:42.078 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638469748021067305}
[easee ] TRACE 2024/03/25 15:46:42 CommandResponse EHNRT639: {SerialNumber:EHNRT639 ID:63 Timestamp:2024-03-25 14:46:43.268448 +0000 UTC DeliveredAt:2024-03-25 14:46:43.13 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638469748032684480}
[easee ] TRACE 2024/03/25 15:46:42 CommandResponse EHNRT639: {SerialNumber:EHNRT639 ID:48 Timestamp:2024-03-25 14:46:43.31007 +0000 UTC DeliveredAt:2024-03-25 14:46:43.204 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638469748033100705}
[easee ] TRACE 2024/03/25 15:46:42 CommandResponse EHNRT639: {SerialNumber:EHNRT639 ID:22 Timestamp:2024-03-25 14:46:43.308352 +0000 UTC DeliveredAt:2024-03-25 14:46:43.289 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638469748033083524}
[easee ] TRACE 2024/03/25 15:46:42 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2024/03/25 15:46:42 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2024/03/25 15:46:42 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) ERROR_STRING no schedule to override, but restoring dynamic current anyway
[easee ] TRACE 2024/03/25 15:46:43 CommandResponse EHNRT639: {SerialNumber:EHNRT639 ID:11 Timestamp:2024-03-25 14:46:44.247579 +0000 UTC DeliveredAt:2024-03-25 14:46:44.093 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638469748042475791}
[easee ] TRACE 2024/03/25 15:46:43 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 40
[easee ] TRACE 2024/03/25 15:46:43 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 40
[easee ] TRACE 2024/03/25 15:46:43 ProductUpdate EHNRT639: (2024-03-25 14:46:43 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 40
```

Auch wichtig - die Command Ticks die hier in diesem Block als Bestätigung vom Charger kommen (`638469748021067305`, `638469748032684480`, `638469748033100705`,  `638469748033083524`), haben wir sonst nicht im Log. 

D.h. nach meinem Verständnis, irgendetwas anderes regelt hier parallel zu evcc. 

@Cytron1980 falls Du noch irgendetwas anderes "smartes" in Deinem Netz hast, daß mit Easee spricht wird dies wohl der Grund sein. Alternativ, hast Du irgendwelche Ladepläne oder ähnliches bei Easee direkt definiert, die die Easee Cloud dazu veranlassen könnten deinen Charger wieder auf 40A zurückzusetzen? Ansonsten würde ich bei Easee mal anfragen, was genau dieser Fehlertext andeutet? 

Nach jetzigem Stand ist das hier aber kein evcc Problem.

Cytron1980, 2024-03-29:
@GrimmiMeloni Frohe Ostern und danke erst einmal für deine Bemühungen. Den "Verdacht" anderer Smarter Devices, die Zugriff auf die Wallbox haben, hatte ich auch schon und danach die Easee Instanz auf meinem Iobroker gestoppt. Daran liegt es wohl nicht. Desweiteren habe ich einen Stromvertrag bei Tibber, wo die Wallbox zwar eingebunden aber das "Smart-Charging" deaktiviert ist und der Zeitplan der Wallbox steht auf "Aus" steht. Die Wallbox bei Tibber aus dem System zu nehmen ist ohne Hilfe des Supports leider nicht möglich. Werde dieses aber mal beim Support anreizen und das Verhalten dann beobachten. soll ich die Command Ticks zu Easee senden oder den Error String? Womit können die wohl am Meisten was anfangen?
Sollte es neue Erkenntnisse geben, so melde ich mich nochmal.

VG

Tonno87, 2024-03-29:
Du kannst in der easee cloud den Betreiber entfernen oder wechseln:

[image]

Bei mir ist Tibber auch seit Monaten als Betreiber eingetragen, die Probleme sind aber erst seit kurzem vorhanden.
Ich hab bei mir noch den Equalizer im System, aber der begrenzt ja über andere Parameter und sollte evcc eigentlich nicht stören.

Cytron1980, 2024-03-29:
@Tonno87 Danke für den Tipp. Ist durch das Löschen des technischen Betreibers auch tatsächlich die Anbindung an Tibber aufgehoben?
Wenn dies allerdings bei dir nicht zur Behebung des Fehlers geführt hat, sehe ich ert einmal keine Veranlassung dazu dieses zu Ändern. So wie ich das sehe, ist dieser Eintrag auch nur ein Text und hat keinerlei Auswirkung auf die Regelung des Laderoboters, oder?

VG

Tonno87, 2024-03-29:
Über die Angabe des Betreibers wird diesem der Zugriff gestattet die Wallbox zu steuern, man kann natürlich nicht ausschließen daß trotz der Pausierung bei Tibber eine Beeinflussung stattfinden kann. Immerhin kann sich da ja auch etwas in der Cloud geändert haben. 
Wenn du den Eintrag rausnimmst hat tibber keinen Zugriff mehr, in der Kachel steht das dann auch so drin.

Ich vermute mal es handelt sich technisch gesehen um einen ocpp Zugang, daher kann man da auch nur einen Betreiber auswählen.
Evcc greift ja direkt auf die API zu.

Cytron1980, 2024-03-29:
Stimmt, wirklich "Smart" ist das Laden von Tibber nicht. Aber wenn Tibber immer 3-Phasig laden muss/will, könnte das ja doch schon zu einer Umschaltung führen, oder? ich werde in jedem Fall mal den Support sowohl von Tibber las auch Easee bemühen. Mal sehen, was die dazu sagen.

Cytron1980, 2024-04-07:
Kurzes Update: Nach dem Löschen von Tibber als Betreiber der Easee funktioniert alles wie gewollt. Danke für den Support hier. 
Viele Grüße