evcc-io/evcc #12776: UI: 124.7 Netzbezug grau und nicht wählbar

rofl0815, 2024-03-08:
### Describe the bug

Auf der UI ist der Netzbezug zu einem bestimmten Preis nicht mehr auswählbar. Wir haben das genutzt, wenn zB bei Tibber der Strom besonders günstig war, um schnell zu laden. Das scheint seit v124.4 nicht mehr zu klappen.
[image]

### Steps to reproduce

1.install 124.4 nightly
2. install 124.7
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
 #levels:
  #site: debug
  #lp-1: debug
  #lp-2: debug
  #cache: error
  #db: error

sponsortoken: 

# unique installation id
# plant:

interval: 15s # control cycle interval

meters:
  - name: gridsma
    type: template
    template: sma-home-manager
    usage: grid
    host: ... # IP-Adresse oder Hostname 

chargers:
- type: template
  template: twc3 
  host: ...  
  name: wallbox4

vehicles:
- name: Fuego
  type: template
  template: tesla 
  title: Fuego  
  accessToken: ...
  refreshToken: ...
  vin: ...
  capacity: 55
  control: true
- name: Data
  type: template
  template: tesla
  title: Data  
  accessToken: ...
  refreshToken:...
  vin:  ...
  capacity: 55
  control: true

loadpoints:
- title: ...
[... 156 lines cut ...]
[site  ] DEBUG 2024/03/08 16:44:15 ----
[lp-1  ] DEBUG 2024/03/08 16:44:16 charge power: 0W
[site  ] DEBUG 2024/03/08 16:44:16 grid meter: -450W
[site  ] DEBUG 2024/03/08 16:44:16 grid powers: [-260 -117 -73]W
[site  ] DEBUG 2024/03/08 16:44:16 grid currents: [-1.4 -1.51 -1.38]A
[site  ] DEBUG 2024/03/08 16:44:16 pv power: 750W
[site  ] DEBUG 2024/03/08 16:44:16 site power: -150W
[lp-1  ] DEBUG 2024/03/08 16:44:16 vehicle odometer: 23931km
[lp-1  ] DEBUG 2024/03/08 16:44:16 charge voltages: [0 4.1 0]V
[lp-1  ] DEBUG 2024/03/08 16:44:16 charge currents: [0 0.2 0.2]A
[lp-1  ] DEBUG 2024/03/08 16:44:16 charger status: B
[lp-1  ] DEBUG 2024/03/08 16:44:16 vehicle status: B
[lp-1  ] DEBUG 2024/03/08 16:44:16 vehicle soc: 100%
[lp-1  ] DEBUG 2024/03/08 16:44:16 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/03/08 16:44:16 vehicle range: 401km
[lp-1  ] DEBUG 2024/03/08 16:44:16 pv charge current: 0.217A = 0A + 0.217A (-150W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.7

andig, 2024-03-08:
Die Einstellung ist jetzt je Ladepunkt.

@naltatis hier wieder die Frage, das bei nur einem LP auch an der alten Stelle anzuzeigen...

rofl0815, 2024-03-08:
Ähh…sorry, was heisst „ Die Einstellung ist jetzt je Ladepunkt.“ ?