evcc-io/evcc #12779: Nach Update auf 124.7 evcc nicht mehr erreichbar

iphikles3000, 2024-03-08:
### Describe the bug

Hallo,
beim Update auf evcc 124.7 erhalte ich die Fehlermeldung 

"your evcc configuration is not compatible with the new version." sowie den Prompt "Do you want to keep your old (working) evcc version?"

Installiere ich das Update dennoch, ist evcc nicht erreichbar. Installiere ich ältere Vorversionen (124.4;5;), funktioniert die Oberfläche wie gehabt.

Allein beim Upgrade auf 124.7 habe ich das Problem.

Besten Dank für eure Tipps  und viele Grüße
Patrick

### Steps to reproduce

1.Update to 124.7
2. Install anyway
3. Try to reach the GUI via browser
...

### Configuration details

```yaml
network:
  port: 7070

log: warn

interval: 6s

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCJ
meters:
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.119
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.119
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.0.119
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery3

chargers:

- type: template
  template: elliconnect
  ski: A3D0 58F2 A79F 92FF
  ip: 192.168.0.105
  name: wallbox1

- type: template
  template: elliconnect
  ski: D787 FE38 09C9 30F6 0174835C
  ip: 192.168.0.103
[... 140 lines cut ...]
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 read 7 NodeManagementUseCaseData
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement call 66236 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 result 8 66236 ResultData 0
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to DeviceClassification reply 66240 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement result 66243 4 ResultData 0
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to DeviceDiagnosis reply 66246 6 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement reply 66247 7 NodeManagementUseCaseData
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement read 66248 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 reply 9 66248 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 result 10 66248 ResultData 0
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement read 66249 NodeManagementDestinationListData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 reply 11 66249 NodeManagementDestinationListData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 result 12 66249 ResultData 0
[eebus ] DEBUG 2024/03/08 18:03:15 Recv: d:_i:47859_Elli-Wallbox-2102A150C4: to NodeManagement read 66251 NodeManagementUseCaseData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 reply 13 66251 NodeManagementUseCaseData
[eebus ] DEBUG 2024/03/08 18:03:15 Send: d:_i:47859_Elli-Wallbox-2102A150C4 result 14 66251 ResultData 0
```

### What type of operating system are you running?

Linux

### Version

0.124.7

andig, 2024-03-08:
Tronity fixed in nightly release.

closed 2024-03-08 by andig

iphikles3000, 2024-03-08:
Sorry, war leider zu doof und habe die offenen Issues dazu übersehen... Tausend dank für die schnelle Reaktion!

Peter1984Git, 2024-03-09:
Ich habe das Problem auch noch beim Update-Versuch auf die 0.124.9. 

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  cache: error

# unique installation id
plant: asdf

interval: 30s # control cycle interval

sponsortoken: asdf

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.178.39
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.178.39
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.178.39
  port: 502
  name: battery3
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.178.38
  name: pv4
- type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.178.40
  name: pvost

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.178.110
  port: 502
  modbus: tcpip
  name: wallbox6
vehicles:
- name: ev5
  type: template
  template: id
  title: ID.4 GTX
  user: asdf
  password: asdf
  vin: asdf
  capacity: 77

loadpoints:
- title: Carport
  charger: wallbox6
#mode: now = volle Leistung minpv = minimale Leistung pv = nur Ueberschuss off = aus
  mode: pv
#phases: 0
#mincurrent: 6
#maxcurrent: 16
  enable:
   threshold: 0
   delay: 2m
  disable:
   threshold: 200
   delay: 5m

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    - pvost
    battery:
    - battery3
  residualPower: 100

mqtt:
  broker: 192.168.178.82:1883
  topic: evcc

andig, 2024-03-10:
@Peter1984Git dann bitte neues Issue oder Config korrigieren. Vmtl. Ists bei Dir eine richtiger Hinweis.