evcc-io/evcc #12787: Keba P30 - Phasenumschaltung scheint seit 0.124.7 nicht mehr zu funktionieren

stefa75, 2024-03-08:
### Describe the bug

Die Umschaltung von 1p auf 3p klappt seit dem Update nicht mehr. Weder automatisch noch durch manellen Umschalten in der GUI.

### Steps to reproduce

Trotz mehrmaligem Neustart von Wallbox sowie HA-Addon schaltet EVCC die Phasen nicht um.
Nach manuellen Umschalten der Phasen (mein Schütz) wird mit 3p geladen. Aber EVCC scheint den "Befehl" an die WB nicht zu geben umzuschalten.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# sponsor token, valid until 2026-11-30 (renew for free then) 
sponsortoken: XXXX
log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
  - name: grid
    type: custom
    power: #Power # default value, optionally override # Leistung (W)
      source: mqtt
      topic: Haus/netzleistung
      timeout: 60s
      scale: -1
  - name: pv # Leistung PV
    type: custom
    power:
      source: mqtt
      topic: Haus/pvleistung
      timeout: 60s

chargers:
  - name: shellyplug
    type: template
    template: shelly
    host: 192.168.10.88 # IP-Adresse oder Hostname
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (optional)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    channel: 0 # optional
    standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional) 
  - name: keba
    type: template
    template: keba-modbus
    
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.10.181 # Hostname
    port: 502 # Port 
    
site:
  title: Wohnhaus
  meters:
    grid: grid
    pv:
    - pv
  residualPower: 200

vehicles:
  - name: byd
    type: template
    template: offline
    title: BYD Atto 3 # Wird in der BenutzeroberflÃ¤che angezeigt (optional)
    capacity: 60.5 # AkkukapazitÃ¤t in kWh (optional) 
  - name: pickman
    type: template
    template: offline
    title: Pickman # Wird in der BenutzeroberflÃ¤che angezeigt (optional)
    capacity: 10 # AkkukapazitÃ¤t in kWh (optional) 

loadpoints:
  - title: Wallbox # display name for UI
    charger: keba # Wallbox Referenz
    vehicle: byd
    mode: now # charge mode (off, now, minpv, pv)
    phases: 0
  - title: Steckdose-Stadel # display name for UI
    charger: shellyplug # Wallbox Referenz
    vehicle: pickman
    mode: now # charge mode (off, now, minpv, pv)
    
# mqtt message broker
mqtt:
  broker: haus.homeassistant:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt
  password: ullif197
  
# tarife:
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price with price zones)
    type: fixed
    price: 0.2102 # EUR/kWh
    zones:
      - days: Mo-Fr
        hours: 8-20
        price: 0.2381 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
#    type: fixed
#    price: 0.0963 # [currency]/kWh
    type: custom
    price: 
      source: mqtt
      topic: Haus/feedin-tarif
      timeout: 25h
```

### Log details

```text
[site  ] DEBUG 2024/03/08 19:34:29 ----
[lp-1  ] DEBUG 2024/03/08 19:34:29 charge power: 3645W
[lp-2  ] DEBUG 2024/03/08 19:34:29 charge power: 0W
[site  ] DEBUG 2024/03/08 19:34:29 pv power: 0W
[site  ] DEBUG 2024/03/08 19:34:29 grid meter: 4233W
[site  ] DEBUG 2024/03/08 19:34:29 site power: 4433W
[lp-1  ] DEBUG 2024/03/08 19:34:30 charge currents: [16 0 0]A
[lp-1  ] DEBUG 2024/03/08 19:34:30 detected active phases: 1p
[lp-1  ] DEBUG 2024/03/08 19:34:30 charge total import: 1320.781kWh
[lp-1  ] DEBUG 2024/03/08 19:34:30 charger status: C
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.7

andig, 2024-03-08:
> Aber EVCC scheint den "Befehl" an die WB nicht zu geben umzuschalten.

Bitte trace log für Modbus für alte und neue Version. Da hat sich ncihts geändert...

closed 2024-03-08 by andig

stefa75, 2024-03-08:
Im Tracelog schaltet evcc auf 3p um jedoch macht die Keba das nicht. Von 3p auf 1p dürfte das Umschalten klappen aber nicht retour.
Ich werde das weiter beobachten. Evt. klappts ja mit PV Überschuss.
Hat vor ein paar Tagen noch perfekt geklappt. Und es war inzwischen nur ein evcc Update...

andig, 2024-03-08:
> Im Tracelog schaltet evcc auf 3p um jedoch macht die Keba das nicht.

Dann bitte Frage an den KEBA Support

> Und es war inzwischen nur ein evcc Update...

Da hat sich nicht geändert. Wenn das Trace Log identisch ist wäre das damit auch bewiesen...

andig, 2024-03-09:
Das hat sicher nichts damit zu tun. Der Eintrag wird ignoriert.

stefa75, 2024-03-09:
Dann weiss ich auch nicht. Ohne phases Eintrag geht's jedenfalls nicht. 
Aber egal. Jetzt klappts.

andig, 2024-03-09:
Wenn das so wäre wärs ein Bug. Dann bräuchte es Logfile sowie Ausgabe von /api/state mit und ohne die Einstellung.