evcc-io/evcc #12794: Unterstützung für SHELLY PLUS 0-10V DIMMER

OnurbHartmann, 2024-03-08:
**Is your feature request related to a problem? Please describe.**
Die Steuerung von ohmschen Lasten wie z.B. Heizstäbe ist derzeit nur über Schaltmodule wie den Shelly möglich. Das reine Ein/Ausschalten ist nicht optimal, hier könnte eine Steuerung des Stroms besser arbeiten.

**Describe the solution you'd like**
Der SHELLY PLUS 0-10V DIMMER in Verbindung mit z.B. Kemo M150 und Kemo M028N Leistungsregler kann hier eine Lösung sein.
Bitte diesen konfigurierbar als 'charger' hinzufügen.

**Describe alternatives you've considered**
keine

**Additional context**
Add any other context or screenshots about the feature request here.

- [SHELLY PLUS 0-10V DIMMER](https://www.shelly.com/de/products/shop/shelly-plus-0-10-v-dimmer)
- [Leistungsregler M028](https://www.kemo-electronic.de/de/Licht-Ton/Effekte/Module/M028-Leistungsregler-110-240-V-AC-2600-VA.php)
- [M150-DC-Puls-Converter](https://www.kemo-electronic.de/de/Wandler-Regler/Konverter/M150-DC-Puls-Converter.php)

premultiply, 2024-03-09:
Das ist schon sehr speziell und bei 3kVA mit Phasenanschnitt auch noch hinsichtlich der Netzverträglichkeit zumindest fragwürdig.
Ich gehe nicht davon aus, dass dies größere Verbreitung finden wird.

Hier ist zudem die Zuordnung von Prozentwert bzw. Spannung zu tatsächlicher Leistung am Ende nicht wirklich vorhersehbar und passt daher nicht zum Regelungsmodell von evcc.

Ggf. einfach per Custom-Charger implementieren?

closed 2024-03-09 by premultiply

OnurbHartmann, 2024-03-09:
Schade, aber oke. Wie kann ich eine Custom-Charger implementieren? VG