evcc-io/evcc #12799: MQTT topics und messages fehlen

Jan2412, 2024-03-09:
### Describe the bug

Hallo,

seit dem Update auf v.0.124.8 fehlen sämtliche MQTT Einträge - mit Ausnahme das status-Topics.

Ausgabe des MQTT Explorers:
Oben: v.0.124.7
Unten: v.0.124.8 

[image]

### Steps to reproduce

Update auf v.0.124.8  von v.0.124.7 kommend.

### Configuration details

```yaml
network:
  schema: http
#  host: 192.168.1.100
  port: 7070
interval: 5s # control cycle interval
telemetry: true
plant: 

log: error

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1  
  #host: 192.168.1.42
  #port: 502
  host: 192.168.1.100
  port: 5200
  usage: grid
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1  
  #host: 192.168.1.42
  #port: 502
  host: 192.168.1.100
  port: 5200
  usage: pv
  modbus: tcpip  
  name: pv1
- type: template 
  name: my_battery
  template: huawei-dongle-powersensor
  usage: battery
  modbus: tcpip
  id: 1
  #host: 192.168.1.42
  #port: 502
  host: 192.168.1.100
  port: 5200
- type: template
  name: pv2
  template: solarman
  usage: pv
  host: 192.168.1.27 # IP-Adresse oder Hostname
  user: admin # Standard-User ist admin (Optional)
  password: admin # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (Optional) 
chargers:
- name: go-e-v3
  type: go-e
  uri: http://192.168.1.48 # go-e ip address (local)

loadpoints:
- title: Garage
  charger: go-e-v3
  mode: pv
  mincurrent: 6
  maxcurrent: 16
  guardduration: 3m
  enable: # pv mode enable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 0 # Ein positiver Wert für Netzbezug, ein negativer Wert für Export. Bei 0 muss der Export die minimale Ladeleistung erreicht haben.
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 0 # Ein maximaler Netzbezug von 0W ist erlaubt
vehicles:
  - name: tesla_m3
    title: Model 3
    capacity: 60 
    icon: car
    mincurrent: 6
    type: template
    template: tesla
    accessToken:
    refreshToken: 
    vin: 
  - name: twingo
    type: template
    template: renault
    title: Twingo Urban Night # display name for UI
    capacity: 22 # kWh
    icon: car
    mincurrent: 8
    user: # user
    password:  # password
    vin: 
    alternativewakeup: true
  - name: Pool
    type: custom
    title: Wassertemperatur
    capacity: 1000
    icon: waterheater
    phases: 1
    soc:           # SoC = Wassertemperatur
      source: http
      uri: http://192.168.1.45/status/
      method: GET
      jq: .ext_temperature.[].tC
      timeout: 10s
    onIdentify:
      mode: off

site:
  title: Zuhause
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2
    battery: my_battery 
  residualPower: 200

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    # type: fixed
    # price: 0.2329 # [currency]/kWh
    type: awattar
    region: de # optional
    charges: 0.1277 # optional, additional charges per kWh
    tax: 0.19 # optional, additional tax (0.1 for 10%)
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0 # [currency]/kWh
  co2:
    type: grünstromindex
    zip: 51399 # PLZ mit führender Null mit in "" setzen

mqtt:
  broker: 192.168.1.100:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: foo
  user: mqtt-user
  password: mqtt-user

sponsortoken:
```

### Log details

```text
Falls dies relevant für den Bug sein sollte, liefere ich es nach (sofern im HomeAssistant Add-on möglich)
```

### What type of operating system are you running?

HomeAssistant Add-on

VolkerK62, 2024-03-09:
Bei mir auch. Was ich im MQTT Explorer beobachten kann, dass ganz kurz (keine Sekunde) das topic `site` reinkommt und sofort wieder gelöscht wird

xerion3800, 2024-03-09:
Kann das auch bestätigen habe heute auch auf das aktuelle Nightly hochgezogen (wegen https://github.com/evcc-io/evcc/issues/12763)
Nach dem keine Werte mehr. Ich habe nur eine herausgefunden. Wenn ich den Broker neu starte nachdem evvc hochgefahren ist, scheint die Anmeldung zu klappen. Wenn der Broker aber schon läuft und man einen evcc Neustart durchführt klappt es nicht.
Meine Umgebung ist aber ein Docker Container.

xerion3800, 2024-03-09:
> Bei mir auch. Was ich im MQTT Explorer beobachten kann, dass ganz kurz (keine Sekunde) das topic `site` reinkommt und sofort wieder gelöscht wird

ja kann ich auch bestätigen.

nuiler, 2024-03-09:
bei mir leider auch

Tonno87, 2024-03-09:
Hier auch... Hab leider heute auf das Backup verzichtet 😕

vistalba, 2024-03-09:
Bei mir ebenfalls... seit 0.124.8 

[image]

StevieC121176, 2024-03-09:
probiert mal die neues nightly `0.124.8 (b5758f6a)` Damit funktioniert es bei mir wieder.

premultiply, 2024-03-09:
Fixed by #12801

closed 2024-03-09 by premultiply