evcc-io/evcc #12806: "Min+PV" EVCC Mode, Elli Pro und Taycan wollen nicht mehr laden, nur noch "Schnell" Modus funktioniert

mmrazovic, 2024-03-09:
### Describe the bug

EVCC with Min+PV Mode starts and stops and starts and stops the whole time. If I set to EVCC mode "Schnell" (max. Power) Taycan can charge without any problems. In the past, before the current winter season, mode "Min+PV" run without any problems, therefore I can not say with which version exactly in the past everything worked. During winter season PV power was too low therefore during winter I always used "Schnell" mode. But now, sun is strong again I want to use PV Power. But with the current version of evcc it is not working.

Elli Pro charger has the current version (v3.8.13;v3.29.7).

Taycan can charge on lot of other public AC charges without problems.

For me 2 thing a kind of strange:
a) See the log, you see "charger status: B" and "charger status: C", but I had ALWAYS (please trust me) "Min+PV" selected
For me, it looks like if it is Charger Status B, it runs unstable, with charger Status C it runs stable.
Again, I had ALWAYS (please trust me) "Min+PV" selected!!

b) As you can see in the logs: WHY is there always a "wake-up timer: stop"? What kind of timer? And then "wake-up timer: start".
I do not understand why it is working that unstable within this mode, because "Min+PV" sets always the minimum ampere, therefore "stable" ampere power.

### Steps to reproduce

1. Set EVCC to Min+PV mode
2. Plug in to Taycan
3. Taycan charges with 4,2 kw
4. Taycan stops charging, show "error"
5. After a while it starts charging again
6. GOTO 3 (endless...)
...

### Configuration details

```yaml
network:
  schema: http
  host: pi3bplus.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  eebus: debug
  cache: error

# unique installation id
plant: xxxremovedxxx

interval: 30s ## control cycle interval

meters:
- type: template
  template: eastron
  id: 1
  host: modbusgw
  port: 502
  modbus: rs485tcpip
  name: grid1
- type: template
  template: kostal-piko-legacy
  usage: pv
  host: pvserver
  user: pvserver
  password: xxxremovedxxx
  name: pv2
- type: custom
  power:
    source: http
    uri: http://192.168.1.61/meter/0/power
    method: GET
    jq: .power
    timeout: 10s
  name: ShellyPlug1

chargers:
- type: template
[... 222 lines cut ...]
Mar 09 13:29:07 pi3bplus evcc[4683]: [site  ] DEBUG 2024/03/09 13:29:07 grid currents: [3.53 2.99 3.46]A
Mar 09 13:29:07 pi3bplus evcc[4683]: [site  ] DEBUG 2024/03/09 13:29:07 site power: 2302W
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 charge currents: [6.09 6.06 6.17]A
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 detected active phases: 3p
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 charger status: C
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 soc estimated: 54.50% (vehicle: 54.00%)
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 vehicle soc: 55%
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 vehicle range: 192km
Mar 09 13:29:07 pi3bplus evcc[4683]: [lp-1  ] DEBUG 2024/03/09 13:29:07 pv charge current: 2.66A = 6A + -3.34A (2302W @ 3p)
Mar 09 13:29:07 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:07 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3150 DeviceDiagnosisHeartbeatData
Mar 09 13:29:08 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:08 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3151 DeviceDiagnosisHeartbeatData
Mar 09 13:29:09 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:09 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3152 DeviceDiagnosisHeartbeatData
Mar 09 13:29:10 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:10 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3153 DeviceDiagnosisHeartbeatData
Mar 09 13:29:11 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:11 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3154 DeviceDiagnosisHeartbeatData
Mar 09 13:29:11 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:11 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3155 DeviceDiagnosisHeartbeatData
Mar 09 13:29:12 pi3bplus evcc[4683]: [eebus ] DEBUG 2024/03/09 13:29:12 Send: d:_i:47859_Elli-Wallbox-2040A0ZF1D notify 3156 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

### Version

0.124.8

andig, 2024-03-09:
Was sagt /api/state?

andig, 2024-03-09:
From the log the vehicle stops charging by itself.

/cc @DerAndereAndi

mmrazovic, 2024-03-09:
result of /api/state from right now:

`{
  "result": {
    "auth": {
      "vehicles": {}
    },
    "batteryConfigured": false,
    "batteryDischargeControl": false,
    "batteryMode": "unknown",
    "bufferSoc": 0,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "greenShareHome": 0,
    "greenShareLoadpoints": 0,
    "gridConfigured": true,
    "gridCurrents": [
      0.8031361699104309,
      0.35878852009773254,
      0.5974457859992981
    ],
    "gridEnergy": 7184.64208984375,
    "gridPower": 80.76851654052734,
    "homePower": 80.76851654052734,
    "loadpoints": [
      {
        "chargeCurrent": 0,
        "chargeCurrents": [
          0,
          0,
          0
        ],
        "chargeDuration": 0,
        "chargePower": 0,
        "chargedEnergy": 0,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": false,
        "chargerPhysicalPhases": null,
        "charging": false,
        "connected": false,
        "connectedDuration": 9223372036854776000,
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": "0001-01-01T00:00:00Z",
        "effectivePriority": 0,
        "enableThreshold": 0,
        "enabled": false,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 16,
        "minCurrent": 6,
        "mode": "minpv",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
        "phasesActive": 3,
        "phasesConfigured": 3,
        "phasesEnabled": 3,
        "planEnergy": 0,
        "planOverrun": false,
        "planProjectedStart": "0001-01-01T00:00:00Z",
        "planTime": "0001-01-01T00:00:00Z",
        "priority": 0,
        "pvAction": "inactive",
        "pvRemaining": 0,
        "sessionCo2PerKWh": null,
        "sessionEnergy": 0,
        "sessionPrice": null,
        "sessionPricePerKWh": null,
        "sessionSolarPercentage": 0,
        "smartCostActive": false,
        "title": "Elli Charger Pro Wallbox",
        "vehicleClimaterActive": null,
        "vehicleDetectionActive": false,
        "vehicleName": "ev3",
        "vehicleOdometer": 22994,
        "vehicleRange": 0,
        "vehicleSoc": 0,
        "vehicleTargetSoc": 0
      }
    ],
    "prioritySoc": 0,
    "pv": [
      {
        "power": 0
      }
    ],
    "pvConfigured": true,
    "pvEnergy": 0,
    "pvPower": 0,
    "residualPower": 0,
    "siteTitle": "Schönbuchweg 9",
    "smartCostType": "pricestatic",
    "sponsor": "mmrazovic",
    "sponsorTokenExpires": 0,
    "statistics": {
      "30d": {
        "avgCo2": 0,
        "avgPrice": 0.35450439489837793,
        "chargedKWh": 428.19995225615764,
        "solarPercentage": 1.8521407916024315
      },
      "365d": {
        "avgCo2": 0,
        "avgPrice": 0.33751005387440464,
        "chargedKWh": 3122.0034555644966,
        "solarPercentage": 8.948508759564563
      },
      "total": {
        "avgCo2": 0,
        "avgPrice": 0.33751005387440464,
        "chargedKWh": 3122.0034555644966,
        "solarPercentage": 8.948508759564563
      }
    },
    "tariffFeedIn": 0.11,
    "tariffGrid": 0.2829,
    "tariffPriceHome": 0.2829,
    "tariffPriceLoadpoints": 0.2829,
    "vehicles": {
      "ev3": {
        "title": "Taycan GTS ST",
        "icon": "car",
        "capacity": 84
      }
    },
    "version": "0.124.8"
  }
}`

mmrazovic, 2024-03-09:
> From the log the vehicle stops charging by itself.

in the first part of the log output you see the trace while it was so unstable the second part was a stable trace.

What  does the Charger Status show? When it was in Charger status B, it run unstable, with charger Status C it run stable.

DerAndereAndi, 2024-03-09:
Status B is reported when the charging status is reported as being Finished or Paused. To know exactly, an log with EEBUS log level trace is required.

But in the end either the charger stopped charging or the EV.

mmrazovic, 2024-03-09:
> Status B is reported when the charging status is reported as being Finished or Paused. To know exactly, an log with EEBUS log level trace is required.
> 
> But in the end either the charger stopped charging or the EV.

ok, tomorrow I will make a new trace and post it here.

I'm just wondering why Min+PV was running so good in the past and not today. And why this happens only in this mode and not in "Schnell" mode.

mmrazovic, 2024-03-10:
> ok, tomorrow I will make a new trace and post it here.
>
> I'm just wondering why Min+PV was running so good in the past and not today. And why this happens only in this mode and not in "Schnell" mode.

I just did a charging session, and it was so easy to reproduce: Within the log you will see 2 or 3 (? don't remember any more) three loading interruptions! Exactly like yesterday. 

@DerAndereAndi with trace for rebus

I would be glad for your comments, thanks so much!

mmrazovic, 2024-03-10:
attached the file syslog
[syslog.txt](https://github.com/evcc-io/evcc/files/14549411/syslog.txt)

DerAndereAndi, 2024-03-10:
At 09:02:04: The charging stops without evcc doing anything. Either the wallbox triggers a stop (whysoeever) or the EV stops charging.

At 09:03:06: The EV is reported as being disconnected from the charger.

In general: the software in your wallbox is buggy as buggy can be. So it is possible that this is just another one. If you have the 22kW OBC in your Taycan, that also not trouble free as lots are being replaced as it stops charging on AC for some. So what you could see is also an early stage of the OBC to stop working.

andig, 2024-03-10:
Sounds like vehicle/charger software issue. Keeping this open for reference, but moving to discussions. 
thank you @DerAndereAndi