evcc-io/evcc #12807: SMA Sunnyboy Storage - Entladung obwohl deaktiviert

Hooch76, 2024-03-09:
### Describe the bug

Option "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" ist aktiv, trotzdem wird Haus-Akku entladen

### Steps to reproduce

1. Option "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" aktivieren
2. EVCC in den Modus "Schnell"
3. Haus-Akku wird trotzdem entladen
...

### Configuration details

```yaml
meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.150
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.178.151
  password: xxyyzz
  name: pv2
- type: template
  template: sma-sbs-modbus
  usage: battery
  modbus: tcpip
  host: 192.168.178.152
  port: 502
  id: 3
  name: battery3

chargers:
- type: template
  template: smaevcharger
  host: 192.168.178.153
  user: abc
  password: xxyyzz
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: off
  phases: 0
  vehicle: bev
  soc:
    poll:
      mode: always
      interval: 10m
    estimate: false

site:
  title: xxx
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 300

vehicles:
  - name: bev
    type: template
    template: teslamate
    title: Tesla
    id: 1 # optional
    capacity: 75 # Akkukapazität in kWh (optional)
    icon: car
```

### Log details

```text
evcc[3415748]: [site  ] DEBUG 2024/03/09 13:18:37 set battery mode: hold
```

### What type of operating system are you running?

Linux

### Version

0.124.8

closed 2024-03-09 by premultiply

Hooch76, 2024-03-09:
Wie lange ist denn "mehrere Minuten"? Bei mir hat sich in über 10min und einem SoC von 100-84% des Hausakkus nichts getan. Danach habe ich den Versuch abgebrochen. War das noch zu kurz?

(P.S. "Closed" finde ich ziemlich sportlich)

andig, 2024-03-09:
@Hooch76 ohne aussagekräftiges Logfile eigentlich immer Close…

Hooch76, 2024-03-09:
Hier ist das Log. Gerade eben nochmal reproduziert,

```
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 charge currents: [-0 -0 -0]A
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 charger status: A
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] INFO 2024/03/09 15:20:37 car disconnected
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 set session soc limit: 0
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 set session energy limit: 0
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 vehicle soc: 80%
Mär 09 15:20:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:37 vehicle range: 399km
Mär 09 15:20:38 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:38 charger disable
Mär 09 15:20:38 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:20:38 wake-up timer: stop
Mär 09 15:20:38 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:20:38 set battery mode: normal
Mär 09 15:21:06 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:06 ----
Mär 09 15:21:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:07 charge power: 0W
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 pv power: 2166W
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 battery soc: 100%
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 battery power: 0W
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 grid meter: -1860W
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 grid powers: [-665 -652 -543]W
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 grid currents: [-2.96 -2.83 -2.79]A
Mär 09 15:21:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:07 site power: -1560W
Mär 09 15:21:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:07 charge currents: [-0 -0 -0]A
Mär 09 15:21:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:07 charger status: A
Mär 09 15:21:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:07 next soc poll remaining time: 9m30s
Mär 09 15:21:36 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:36 ----
Mär 09 15:21:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:37 charge power: 0W
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 pv power: 2157W
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 battery soc: 100%
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 battery power: 0W
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 grid meter: -1814W
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 grid powers: [-664 -660 -490]W
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 grid currents: [-2.95 -2.87 -2.64]A
Mär 09 15:21:37 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:21:37 site power: -1514W
Mär 09 15:21:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:37 charge currents: [-0 -0 -0]A
Mär 09 15:21:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:37 charger status: A
Mär 09 15:21:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:21:37 next soc poll remaining time: 9m0s
Mär 09 15:22:06 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:06 ----
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 charge power: 2814W
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 pv power: 2164W
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 battery soc: 100%
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 battery power: 0W
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 grid meter: 1037W
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 grid powers: [752 811 -526]W
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 grid currents: [3.37 3.62 -2.89]A
Mär 09 15:22:07 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:07 site power: 1337W
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 charge currents: [6.06 6.06 0.584]A
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 charger status: C
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] INFO 2024/03/09 15:22:07 car connected
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 pv timer elapse
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 pv timer inactive

Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] INFO 2024/03/09 15:22:07 start charging ->

Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 wake-up timer: stop
Mär 09 15:22:07 evccsrv evcc[3447842]: [telegram] DEBUG 2024/03/09 15:22:07 sending to ***
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 vehicle soc: 80%
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 vehicle range: 399km
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] WARN 2024/03/09 15:22:07 charger out of sync: expected disabled, got enabled
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 charger disable
Mär 09 15:22:07 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:22:07 wake-up timer: stop
Mär 09 15:22:36 evccsrv evcc[3447842]: [site  ] DEBUG 2024/03/09 15:22:36 ----
[... cut ...]
Mär 09 15:34:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:34:37 charge currents: [-0 -0 -0]A
Mär 09 15:34:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:34:37 charger status: B
Mär 09 15:34:37 evccsrv evcc[3447842]: [lp-1  ] INFO 2024/03/09 15:34:37 stop charging <-
Mär 09 15:34:37 evccsrv evcc[3447842]: [telegram] DEBUG 2024/03/09 15:34:37 sending to ***
Mär 09 15:34:37 evccsrv evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:34:37 vehicle soc: 81%
Mär 09 15:34:37 **evccsrv** evcc[3447842]: [lp-1  ] DEBUG 2024/03/09 15:34:37 vehicle range: 405km
```

Hooch76, 2024-03-09:
@premultiply Kannst Du das Issue bitte wieder öffnen, nachdem ich die Logs nachgeliefert habe?

premultiply, 2024-03-09:
Wie du siehst geht die Batterieleistung immer weiter runter. Funktioniert also.
Die Zeitkonstante dafür kann und muss man im WR konfigurieren. Unnötigerweise braucht man dazu einen SMA Gridguard-Code.
Musst dich daher an einen Installateur etc. wenden der einen ebensolchen hat.

Hier gibt es nichts weiter zu tun.

Hooch76, 2024-03-09:
> Hier gibt es nichts weiter zu tun.

Und wieso dauert es bei einem anderen SMA SBS (siehe 
https://github.com/evcc-io/evcc/discussions/12649#discussioncomment-8729494) nur 2min und bei mir entlädt es den gesamten Akku? Ist am Log oben nicht ersichtlich kann ich aber noch liefern. 
Bzw beobachtet im Diskussionsthread auch @kachilles1402 das gleiche Verhalten?

andig, 2024-03-09:
> Die Zeitkonstante dafür kann und muss man im WR konfigurieren.

@premultiply kann man das per Modbus direkt konfigurieren? Muss das Template vllt. noch deutlicher werden?

premultiply, 2024-03-09:
Nein, leider nicht.
Wenn die Default-Einstellungen verhunzt wurden muss man sie auch nur wieder über das Webinterface mit GGC-Anmeldung "enthunzen".

andig, 2024-03-09:
Im Template steht von den Anforderungen allerdings nichts :O

Hooch76, 2024-03-09:
> Wenn die Default-Einstellungen verhunzt wurden muss man sie auch nur wieder über das Webinterface mit GGC-Anmeldung "enthunzen".

Um welche (default) Einstellung geht's denn genau? Dann könnten alle betroffenen dies mal vergleichen. (Wenn ich das richtig verstanden habe, dann sollte man ja mit Installateur-Login diesen Wert auf dem Wechselrichter zumindest lesend sehen können).

Übrigens: ich habe den GGC nicht, somit konnte ich hier gar nichts "verhunzen". Dreiste Unterstellung übrigens :-( wenn hier 2 User davon berichten, dass es nicht geht und nur 1 User berichtet, dass es geht.

premultiply, 2024-03-09:
[image]

Ich weiß ja auch nur dass es bei meinem System mit den Standardeinstellungen wie skizziert funktioniert.
Zu anderen Konfigurationen kann ich nichts sagen. Im Zweifelsfall muss das eben SMA beantworten.
Vor allem dass diese Einstellungen per Gridguard geschützt sind halte ich für ziemlich sinnfrei.

kachilles1402, 2024-03-09:
Ich habe nochmal etwas recherchiert:
Laut der SBS3.7-6.0 Modbus Doku kann Register 44039 benutzt werden.
[image]
Für den SBS2.5 ist ganz klar ein Hinweis, dass dieses Register nicht Zyklisch beschrieben werden soll.
[image]

Alternativ habe ich das mal in NodeRed probiert:
http://hausbau-becker.blogspot.com/2021/06/steuerung-des-sbs-25-per-modbus.html

Hier wird über das Register 40799 gesteuert. Das funktioniert auch wunderbar für den 2.5er

Hooch76, 2024-03-09:
> Ich weiß ja auch nur dass es bei meinem System mit den Standardeinstellungen wie skizziert funktioniert.
> Zu anderen Konfigurationen kann ich nichts sagen. Im Zweifelsfall muss das eben SMA beantworten.

Hier mal der Screenshot von meinen Settings. "Externe Vorgabe 2" ist gar nicht vorhanden:

[image]

Gerätetyp: SBS3.7-10
Softwarepaket: 30.12.63.R
Firmware-Version des Betriebssystems: 3.29.0.R

premultiply, 2024-03-09:
So ein Firmwareupdate auf eine aktuelle Versionen sollte nicht schaden...
https://www.sma.de/service/downloads

Hooch76, 2024-03-09:
> So ein Firmwareupdate auf eine aktuelle Versionen sollte nicht schaden... https://www.sma.de/service/downloads

Habe ich heute nachmittag schon probiert. Eine Suche nach einer neuen Version bringt auch nichts aktuelleres:

[image]

Manuell zu aktualisieren traue ich mich nicht, da ich gelesen habe, dass die Firmware-Version vom SBS ggfs. auch mit dem verwendeten Speicher zusammenhängt (LG in meinem Fall). Ich gehe davon, dass SMA schon grob weiss, was sie tun und das seine Richtigkeit hat.

premultiply, 2024-03-09:
Die 4er-Versionen haben keine solchen Sonderlocken mehr die auch bei den 3er schon problemlos waren.
Ich hab auch einen LG Resu M-Speicher.

kachilles1402, 2024-03-10:
> Ich habe nochmal etwas recherchiert: Laut der SBS3.7-6.0 Modbus Doku kann Register 44039 benutzt werden. ![Bildschirmfoto 2024-03-09 um 20 47 02](https://private-user-images.githubusercontent.com/48121689/311463999-d194c106-ab59-4486-a887-eade9623f43d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwNTkwMjksIm5iZiI6MTcxMDA1ODcyOSwicGF0aCI6Ii80ODEyMTY4OS8zMTE0NjM5OTktZDE5NGMxMDYtYWI1OS00NDg2LWE4ODctZWFkZTk2MjNmNDNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDA4MTg0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ4YzFjNzAzMWY3ZDAwMWM0OTQ5N2NmZjFjZTFlY2ZlNmE2OTRmZjU1ZTMzNDcxZTJiZjY4YmU3MTRkNzZiMTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jhMi1mHVZ0zQLkO0V-kAb5pN1A-BJuQ2DWP8xIkRi2M) Für den SBS2.5 ist ganz klar ein Hinweis, dass dieses Register nicht Zyklisch beschrieben werden soll. ![Bildschirmfoto 2024-03-09 um 20 46 35](https://private-user-images.githubusercontent.com/48121689/311463970-f6c80bfe-36ad-40dd-9009-536270a46f76.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwNTkwMjksIm5iZiI6MTcxMDA1ODcyOSwicGF0aCI6Ii80ODEyMTY4OS8zMTE0NjM5NzAtZjZjODBiZmUtMzZhZC00MGRkLTkwMDktNTM2MjcwYTQ2Zjc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEwVDA4MTg0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3YmJlYjdmMzNlNDg1ZDc4Mjg3OTIxOTViOTE2YTZiNzg5ZWViYWFjZWRhZTVmYjUwYzkzYTQ5YWEyMTgwZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Uhw6wdYMwopxjSWmoljBXQAxyauMXQ3A8uJaMiQeQac)
> 
> Alternativ habe ich das mal in NodeRed probiert: http://hausbau-becker.blogspot.com/2021/06/steuerung-des-sbs-25-per-modbus.html
> 
> Hier wird über das Register 40799 gesteuert. Das funktioniert auch wunderbar für den 2.5er. Über 40799 wird die "Maximale Batterieentladeleistung" auf 0 festgelegt. Über 40151 wird die "Wirk- und Blindleistungsregelung über Kommunikation" mit 802 aktiv und 803 inaktiv geschaltet.
> 
> Dh 40799 wird einmalig auf 0 gesetzt und im Modus "Schnell" bzw. "geplante Ladung" wird 40151 aktiv (802) geschaltet. Wenn Modus "PV", dann 40151 auf inaktiv (803).
> 
> Das ganze reagiert auch sofort und ohne Verzögerung.

Ich habe das einmal ganz simpel in NodeRed nachgebaut.
Die evcc Modi kommen dabei per mqtt.
[image]
Aktuell wird der Lademodus der beiden Loadpoints geprüft und 40151 entsprechend geschrieben.
Das funktioniert soweit. Lässt sich das als Template einbauen?

Hooch76, 2024-03-10:
> > Dh 40799 wird einmalig auf 0 gesetzt und im Modus "Schnell" bzw. "geplante Ladung" wird 40151 aktiv (802) geschaltet. Wenn Modus "PV", dann 40151 auf inaktiv (803).
> > Das funktioniert soweit. Lässt sich das als Template einbauen?

Ich kann das auch mal mit dem SBS 3.7 probieren. Allerdings habe ich noch nie was mit modbus gemacht (Home Assistent), kann also etwas dauern. 
Du setzst die Werte jeweils nur 1x oder zyklisch? 
Wenn Du noch weitere Infos für das Setzen der Werte hast dann gerne posten, dann geht's vielleicht schneller mit dem Modbus-Einstieg. 
P.S. Wie sind die defaults der beiden Werte?

kachilles1402, 2024-03-10:
> > Dh 40799 wird einmalig auf 0 gesetzt und im Modus "Schnell" bzw. "geplante Ladung" wird 40151 aktiv (802) geschaltet. Wenn Modus "PV", dann 40151 auf inaktiv (803).
> > Das funktioniert soweit. Lässt sich das als Template einbauen?

Ich kann das auch mal mit dem SBS 3.7 probieren. Allerdings habe ich noch nie was mit modbus gemacht (Home Assistent), kann also etwas dauern. 
Du setzst die Werte jeweils nur 1x oder zyklisch? 
Wenn Du noch weitere Infos für das Setzen der Werte hast dann gerne posten, dann geht's vielleicht schneller mit dem Modbus-Einstieg. 
P.S. Wie sind die defaults der beiden Werte?

_Originally posted by @Hooch76 in https://github.com/evcc-io/evcc/issues/12807#issuecomment-1987145198_
            
            
Ich habe im Home Assistant eine nodered Instanz laufen und das Template von dem oben geteilten link importiert. 
Die Werte setze ich einmalig. Ich teste nachher mal, ob das einmalige setzen funktioniert - aber der Akku muss erst laden 😁
Default Werte bei dem 2.5er:
40799: 2650 (maximale Entladeleistung des SBS 2.5)
40151: 803

Hooch76, 2024-03-16:
> EDIT: einmal schreiben reicht. Läuft seit 1,5 Stunden problemlos durch

Kann ich so bestätigen. Wenn man statt dem `sma-sbs-modbus` das `sma-si-modbus` Template für den SBS 3.7 verwendet, dann klappt's.

andig, 2024-03-16:
@premultiply was heisst das jetzt für Template/Doku?