evcc-io/evcc #12814: Elvi EVBox: Issue with getConfiguration

jpwenzel, 2024-03-09:
### Describe the bug

When the `elvi` template is used (instead of the generic `ocpp` template), my EVBox charger won't complete connecting after startup/reboot of either evcc or the charger.

Looks like a timing issue already described in https://github.com/evcc-io/evcc/discussions/9012#discussioncomment-6547463

Changing the configuration to the generic `ocpp` template and setting `getconfiguration: false` solves the connection issue.

### Steps to reproduce

1. Start evcc, wait until waiting for connection
2. Power on EVBox
3. Communication will start but dropped when `getConfiguration` call happens

### Configuration details

```yaml
chargers:
  - name: evbox-ocpp
    type: template
    template: elvi
    stationId: EVB-P203xxxxx
    connector: 1
    idtag: '1'
    connecttimeout: 5m
    timeout: 2m
    meter: true
```

### Log details

```text
(will try to reproduce and provide logs)
```

### What type of operating system are you running?

Docker container

andig, 2024-03-09:
Seems you are suggesting to add

    getconfiguration: false

to the template? Could you submit a PR?

jpwenzel, 2024-03-09:
> Seems you are suggesting to add
> 
> ```
> getconfiguration: false
> ```
> 
> to the template? 

That sounds like a solution, yes.

> Could you submit a PR?

Sure thing!

https://github.com/evcc-io/evcc/pull/12823

closed 2024-03-10 by andig