evcc-io/evcc #12819: Leere Werte für gridPower/homePower in mqtt nach update auf Version 0.124.8

C64Axel, 2024-03-09:
### Describe the bug

Ich habe gerade auf die Version 0.124.8 upgedated.
Beim debug im NodeRed habe ich festgestellt, dass nach einem gültigen Wert gleich ein leerer Wert gesendet wird:

'''
9.3.2024, 17:08:34[node: debug 5](http://nodered:1880/admin/#)
evcc/site/homePower : msg : Object
object
topic: "evcc/site/homePower"
payload: 898
qos: 1
retain: false
_msgid: "8df7e935e5d855d3"
9.3.2024, 17:08:35[node: debug 5](http://nodered:1880/admin/#)
evcc/site/homePower : msg : Object
object
topic: "evcc/site/homePower"
payload: ""
qos: 1
retain: false
_msgid: "e1ad2e6e3606747d"
'''

Ein Rückgang auf 0.124.7 ergibt wieder richtige Werte ohne Leerwerte.

### Steps to reproduce

update auf 0.124.8
leerer Wert über MQTT noch einem korrekten Wert

### Configuration details

```yaml
# log: debug

sponsortoken: xxxxxxxxxxxxxxxxx
plant: xxxxxxxxxxxxxxxxx

site:
  title: Zu Hause
  meters:
    grid: Hauptzaehler
    pv: sun2000

loadpoints:
  - title: Carport
    charger: autoladen
    vehicle: Mazda CX60
    phases: 1
    soc:
      poll:
        mode: connected
        interval: 10m

chargers:
  - name: autoladen
    type: template
    template: tasmota
    host: xxxxxxxxxxxxxxxxx
    user: xxxxxxxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxx
    channel: 1
    standbypower: 5

meters:
  - name: Hauptzaehler
    type: custom
    power:
      source: mqtt
      topic: stromzaehler/tele/SENSOR
      jq: .SML.Power_curr
  - name: sun2000
    type: template
    template: huawei-dongle
    usage: pv
    modbus: tcpip
    id: 1
    host: xxxxxxxxxxxxxxxxx
    port: 502
  - name: solarlog
    type: template
    template: solarlog
    usage: pv
    host: xxxxxxxxxxxxxxxxx
    port: 502

vehicles:
  - name: Mazda_CX60
    type: template
    template: mazda2mqtt
    title: Mazda CX60
    vin: xxxxxxxxxxxxxxxxx
    phases: 1
    capacity: 17.8
    timeout: 720h

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.xxx
  feedin:
    type: fixed
    price: 0.08

mqtt:
  broker: mosquitto:1883
  topic: evcc
  user: xxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxxxx

influx:
  url: http://influxdb:8086
  database: evcc
  token: xxxxxxxxxxxxxxxxx
  org: Hausautomation

modbusproxy:
  - port: 5021
    uri: xxxxxxxxxxxxxxxxx:502
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Version

0.124.8

mucki12, 2024-03-09:
Und das hat hier mit nichts zu tun?
https://github.com/evcc-io/evcc/issues/12805

C64Axel, 2024-03-09:
Ja, scheint so. Hab die nicht gesehen weil die schon geschlossen waren.
Dann warte ich auf den nächsten Container.

closed 2024-03-09 by C64Axel