evcc-io/evcc #12820: Peugeot vehicle-API failed mit Internal Server Error

Kubiac, 2024-03-09:
### Describe the bug

Bei meinen Eltern funktioniert jetzt schon länger nicht mehr die SOC-Abfrage bei ihrem Peugeot 208-e. In der Peugeot-App funktionieren die genutzten Anmeldedaten und es wird der SOC angezeigt.

### Steps to reproduce

Wenn ich auf dem raspi `evcc vehicle` ausführe klappt es auch nicht und ich bekomme ich den unten stehenden Log-Output.

### Configuration details

```yaml
vehicles:
- type: template
  template: peugeot 
  title: Peugeot  
  icon: car  
  user: ***@gmail.com
  password: *****
  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16
  capacity: 50  
  name: ev4
```

### Log details

```text
[main  ] INFO 2024/03/09 16:29:20 evcc 0.124.8
[main  ] INFO 2024/03/09 16:29:20 using config file: /etc/evcc.yaml
[db    ] INFO 2024/03/09 16:29:20 using sqlite database: /home/mporsch/.evcc/evcc.db
[main  ] ERROR 2024/03/09 16:29:21 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'peugeot': login failed: oauth2: "server_error" "Internal Server Error"
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'peugeot': login failed: oauth2: "server_error" "Internal Server Error"
Capacity: 50.0kWh
Features: [Offline]
```

### What type of operating system are you running?

Linux

### Version

0.124.8

andig, 2024-03-09:
PSA hat das API geändert. Das neue ist uns nicht zugänglich.

closed 2024-03-09 by andig

fschaal, 2024-03-10:
But the PSA controller fixed this using Oauth-flow and extracting the code. Why cant we use that with evcc?

andig, 2024-03-10:
You can already use this via flobz. The process is too cumbersome for integrating in evcc.