evcc-io/evcc #12821: Elvi EVBox: Wallbox goes into error state after unplugging cable

jpwenzel, 2024-03-09:
### Describe the bug

After stopping the charging process on our Hyundai Kona in Min+PV mode (stopped by setting charge mode to "Off" in the UI), I unplug the cable from the car first, and then from the EVBox socket.

Recently (I assume after the upgrade to 0.124.7?) this leads to a malfunction status in the wallbox (red indicator ring).

As a result, the cable is not released, and in order to unlock it, the wallbox needs to be either rebooted through the smartphone app or by power cycling it in the breaker box.

### Steps to reproduce

1. Charged the car on Min+PV setting
2. Disabled charging (Set to Off in UI)
3. Unplug cable from vehicle
4. Tried to unplug cable from wallbox socket

### Configuration details

```yaml
chargers:
  - name: evbox-ocpp
    type: template
    template: ocpp
    stationid: EVB-P203xxxxx
    connector: 1
    idtag: '1'
    connecttimeout: 5m
    timeout: 2m
    metervalues: Current.Import.L1,Current.Import.L2,Current.Import.L3,Energy.Active.Import.Register,Power.Active.Import,Voltage.L1,Voltage.L2,Voltage.L3
    meterInterval: 4s
    getconfiguration: false
```

### Log details

```text
[site  ] DEBUG 2024/03/09 15:17:59 ----
[lp-1  ] DEBUG 2024/03/09 15:17:59 charge power: 3198W
[site  ] DEBUG 2024/03/09 15:18:00 pv power: 1399W
[site  ] DEBUG 2024/03/09 15:18:00 battery soc: 88%
[site  ] DEBUG 2024/03/09 15:18:00 battery power: 2440W
[site  ] DEBUG 2024/03/09 15:18:00 grid meter: -3W
[site  ] DEBUG 2024/03/09 15:18:00 site power: 2537W
[lp-1  ] DEBUG 2024/03/09 15:18:00 charge currents: [4.8 4.8 4.8]A
[lp-1  ] DEBUG 2024/03/09 15:18:00 detected active phases: 3p
[lp-1  ] DEBUG 2024/03/09 15:18:00 charge total import: 5014.225kWh
[lp-1  ] DEBUG 2024/03/09 15:18:00 charger status: C
[lp-1  ] DEBUG 2024/03/09 15:18:00 soc estimated: 66.75% (vehicle: 66.00%)
[lp-1  ] DEBUG 2024/03/09 15:18:00 vehicle soc: 67%
[lp-1  ] DEBUG 2024/03/09 15:18:00 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/03/09 15:18:00 vehicle range: 289km
[ocpp  ] TRACE 2024/03/09 15:18:00 sent JSON message to EVB-P203xxxxx: [2,"2366288140","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"transactionId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[ocpp  ] TRACE 2024/03/09 15:18:00 received JSON message from EVB-P203xxxxx: [2,"2d7f2c18-df13-466d-b78a-409d7d6e913b","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-03-09T14:18:00Z","sampledValue":[{"value":"4.8","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"4.8","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"4.8","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"5014229","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"3193","measurand":"Power.Active.Import","unit":"W"},{"value":"239","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"239","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"238","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/03/09 15:18:00 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c18-df13-466d-b78a-409d7d6e913b",{}]
[ocpp  ] TRACE 2024/03/09 15:18:00 received JSON message from EVB-P203xxxxx: [3,"2366288140",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/03/09 15:18:00 charger disable
[lp-1  ] DEBUG 2024/03/09 15:18:00 wake-up timer: stop
[ocpp  ] TRACE 2024/03/09 15:18:01 received JSON message from EVB-P203xxxxx: [2,"2d7f2c18-df13-466d-b78a-409da3ed663d","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Charging,NoError,No error.,1,Blue,1,{20,7984,10},{5014229,3193},{C,11882,5868,12096},0,0,340,2024-03-09T14:18:00Z,1,3,{239,239,238,48,48,48,926,921,937},250,2,0,139,0,320,60,5004"}]
[ocpp  ] TRACE 2024/03/09 15:18:01 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c18-df13-466d-b78a-409da3ed663d",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/03/09 15:18:01 received JSON message from EVB-P203xxxxx: [2,"2d7f2c19-df13-466d-b78a-409d86c08141","StatusNotification",{"connectorId":1,"status":"SuspendedEVSE","errorCode":"NoError","info":"C;340","timestamp":"2024-03-09T14:18:01Z"}]
[ocpp  ] TRACE 2024/03/09 15:18:01 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c19-df13-466d-b78a-409d86c08141",{}]
[ocpp  ] TRACE 2024/03/09 15:18:01 received JSON message from EVB-P203xxxxx: [2,"2d7f2c19-df13-466d-b78a-409d6012c041","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,SuspendedEVSE,NoError,No error.,0,Yellow,1,{20,7984,0},{5014230,3021},{C,11872,6156,61629},0,0,340,2024-03-09T14:18:01Z,1,3,{239,239,238,48,48,44,927,921,937},250,2,0,139,0,320,0,5004"}]
[ocpp  ] TRACE 2024/03/09 15:18:01 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c19-df13-466d-b78a-409d6012c041",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/03/09 15:18:05 sent JSON message to EVB-P203xxxxx: [2,"405696538","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/03/09 15:18:06 received JSON message from EVB-P203xxxxx: [3,"405696538",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/03/09 15:18:06 TriggerMessage MeterValues for EVB-P203xxxxx: Accepted
[ocpp  ] TRACE 2024/03/09 15:18:06 received JSON message from EVB-P203xxxxx: [2,"2d7f2c1d-df13-466d-b78a-409db719355e","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-03-09T14:18:05Z","sampledValue":[{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L2","unit":"A"},{"value":"0.0","context":"Trigger","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"5014230","context":"Trigger","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"0","context":"Trigger","measurand":"Power.Active.Import","unit":"W"},{"value":"239","context":"Trigger","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"239","context":"Trigger","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"238","context":"Trigger","measurand":"Voltage","phase":"L3","unit":"V"}]}]}]
[ocpp  ] TRACE 2024/03/09 15:18:06 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c1d-df13-466d-b78a-409db719355e",{}]
[ocpp  ] TRACE 2024/03/09 15:18:06 received JSON message from EVB-P203xxxxx: [2,"2d7f2c1e-df13-466d-b78a-409da43de55e","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,SuspendedEVSE,NoError,No error.,0,Yellow,1,{20,7984,0},{5014230,0},{C,11872,6156,61616},0,0,340,2024-03-09T14:18:05Z,1,3,{239,239,238,0,0,0,927,921,937},250,0,0,139,0,320,0,5004"}]
[ocpp  ] TRACE 2024/03/09 15:18:06 sent JSON message to EVB-P203xxxxx: [3,"2d7f2c1e-df13-466d-b78a-409da43de55e",{"status":"Accepted"}]
[site  ] DEBUG 2024/03/09 15:18:10 ----
[... 209 lines cut ...]
[ocpp  ] TRACE 2024/03/09 16:33:08 sent JSON message to EVB-P203xxxxx: [3,"2d7f3db4-df13-456d-b78a-409d35e4ffab",{"status":"Accepted"}]
[site  ] DEBUG 2024/03/09 16:33:10 ----
[lp-1  ] DEBUG 2024/03/09 16:33:10 charge power: 0W
[site  ] DEBUG 2024/03/09 16:33:11 pv power: 994W
[site  ] DEBUG 2024/03/09 16:33:11 battery soc: 96%
[site  ] DEBUG 2024/03/09 16:33:11 battery power: -188W
[site  ] DEBUG 2024/03/09 16:33:11 grid meter: 13W
[site  ] DEBUG 2024/03/09 16:33:11 site power: -75W
[lp-1  ] DEBUG 2024/03/09 16:33:11 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/09 16:33:11 charge total import: 5014.230kWh
[lp-1  ] DEBUG 2024/03/09 16:33:11 charger status: A
[lp-1  ] INFO 2024/03/09 16:33:11 car disconnected
[lp-1  ] INFO 2024/03/09 16:33:11 vehicle updated: Hyundai Kona -> unknown
[lp-1  ] DEBUG 2024/03/09 16:33:11 set session soc limit: 0
[lp-1  ] DEBUG 2024/03/09 16:33:11 set session energy limit: 0
[site  ] DEBUG 2024/03/09 16:33:11 ----
```

### What type of operating system are you running?

Docker container

### Version

0.124.7

andig, 2024-03-09:
> Recently (I assume after the upgrade to 0.124.7?) this leads to a malfunction status in the wallbox (red indicator ring).

Similar to get configuration, that sounds like another bug in the Elvi firmware, probably triggered by https://github.com/evcc-io/evcc/pull/12459. We could make that configurable (which would add yet another variable) but I'd rather prefer to have Elvi fix their boxes.

What does Elvi support say?

andig, 2024-03-10:
@jpwenzel could you also check if 0.124.6 works for you? Then we'd know that #12459 causes this.

jpwenzel, 2024-03-10:
> @jpwenzel could you also check if 0.124.6 works for you? Then we'd know that #12459 causes this.

Yes, I will downgrade and check.

jpwenzel, 2024-03-10:
Works as expected with version 0.124.6. I can even hear the socket unlock shortly after switching to "Off" mode.
No red ring on the wallbox.

Logs:
[evcc-0.124.6.log](https://github.com/evcc-io/evcc/files/14551406/evcc-0.124.6.log)

andig, 2024-03-10:
So this is triggered by https://github.com/evcc-io/evcc/pull/12459. Afaikt nothing forbids #12459. It's up to Elvi to fix their software or explain why this should be an error condition.

/cc @cheerfulalex sieht aus als müssten wir das zurück drehen :(

premultiply, 2024-03-10:
>  I can even hear the socket unlock shortly after switching to "Off" mode.

This should NOT happen as the session should not be closed in this case.
Session should only be terminated by unplugging of the car. The car releases its cable and unplugging it should stop the session and release the cable at the station side (if not fixed cable setup).

Yes, we know that this is the old OCPP behavior but the new behavior is better, more "correct" and streamlined with other Modbus implementations. The session is paused now but not ended.

andig, 2024-03-11:
@premultiply es nutzt ja nix- mindestens ABB und Elvi haben Probleme damit. Wir müssen es wohl optional machen :(

Closed in https://github.com/evcc-io/evcc/commit/1580857163d006d744319aaf79570175161ecb50

closed 2024-03-11 by andig

jpwenzel, 2024-03-12:
> What does Elvi support say?

Opened a ticket with them, waiting for reply:

> I am running my Elvi EVBox wallbox connected to the evcc OCPP implementation.
> During testing, we realized that there is in issue in the implementation of the setChargingProfile method.
> 
> In evcc, we’d like to use „zero charging profile“ to pause/end charging. That profile is sent to EVBox.
> 
> When the user now unplugs their car (on the vehicle side), the charger turns into an error state (red ring) and locks up. Cable is not released, Bluetooth connection is not possible. In order to recover, one has to power cycle by flipping the circuit breaker.
> 
> Please pass this on to your product development/engineering team and ask them for a fix.

andig, 2024-03-14:
As sad this is for the Elvi software it's actually good news. The 0A Ocpp change is not part of 0.124.10, so it seems we can re-apply it ;) /cc @cheerfulalex

jpwenzel, 2024-03-27:
> What does Elvi support say?

Well...

> EVCC has not been network integrated with our backend system server management platform EVBox Everon and therefore is not fully compatible with our API unless fully integrated. This is a process only available for EVBox partners, please contact your asset account owner at Fastplug Energy GmbH for further procedure on this matter.

andig, 2024-03-27:
I have no idea what that should mean. Ocpp is a standard, not something you need to get onboarded on…

jpwenzel, 2024-03-28:
I think the poor support guy had no idea what I was talking about.
I'll try to reach out to some more technical staff at EVBox through LinkedIn... HTH.

coderph0x, 2024-06-01:
@jpwenzel Did you hear back from them yet?

jpwenzel, 2024-06-01:
No, they are absolutely unresponsive. 

I am so frustrated right now that I am actually considering throwing the EVBox away and to tell anyone to not even consider one of their (not) "OCPP compliant" products.

coderph0x, 2025-09-08:
There is no need for that detour though. EVbox has worked flawlessly for months with evcc's ocpp connector now.