evcc-io/evcc #12825: easee 3p -> 1p Phasenumschaltung

kai-morich, 2024-03-09:
### Describe the bug

um 14:05 schaltet evcc auf 1 Phase runter

	[lp-1  ] DEBUG 2024/03/09 14:05:44 available power 3229W < 4140W min 3p threshold
	[lp-1  ] DEBUG 2024/03/09 14:05:44 phase scale1p in 0s
	[easee ] TRACE 2024/03/09 14:05:44 GET https://api.easee.com/api/sites/123456/circuits/123456/settings
	[easee ] TRACE 2024/03/09 14:05:44 {"allowOfflineMaxCircuitCurrent":false,"equalizerId":null,"masterChargerId":"EHCUMYZ7","dynamicCircuitCurrentP1":16.0,"dynamicCircuitCurrentP2":16.0,"dynamicCircuitCurrentP3":16.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":16,"offlineMaxCircuitCurrentP2":16,"offlineMaxCircuitCurrentP3":16}
	[easee ] TRACE 2024/03/09 14:05:44 POST https://api.easee.com/api/sites/123456/circuits/123456/settings
	[easee ] TRACE 2024/03/09 14:05:44 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":0,"dynamicCircuitCurrentP3":0}
	[lp-1  ] DEBUG 2024/03/09 14:05:44 switched phases: 1p

kurze Zeit später glaubt evcc, das genug Energie für 3 Phasen zu 16A vorhanden wäre, und schaltet wieder zurück auf 3 Phasen, aber ohne die übliche Wartezeit

	[lp-1  ] DEBUG 2024/03/09 14:06:18 available power 7068W > 4140W min 3p threshold
	[lp-1  ] DEBUG 2024/03/09 14:06:18 phase scale3p in 0s
	[easee ] TRACE 2024/03/09 14:06:18 GET https://api.easee.com/api/sites/123456/circuits/123456/settings
	[easee ] TRACE 2024/03/09 14:06:18 {"allowOfflineMaxCircuitCurrent":false,"equalizerId":null,"masterChargerId":"EHCUMYZ7","dynamicCircuitCurrentP1":16.0,"dynamicCircuitCurrentP2":0.0,"dynamicCircuitCurrentP3":0.0,"enableIdleCurrent":true,"maxCircuitCurrentP1":16.0,"maxCircuitCurrentP2":16.0,"maxCircuitCurrentP3":16.0,"offlineMaxCircuitCurrentP1":16,"offlineMaxCircuitCurrentP2":16,"offlineMaxCircuitCurrentP3":16}
	[easee ] TRACE 2024/03/09 14:06:18 POST https://api.easee.com/api/sites/123456/circuits/123456/settings
	[easee ] TRACE 2024/03/09 14:06:18 {"dynamicCircuitCurrentP1":16,"dynamicCircuitCurrentP2":16,"dynamicCircuitCurrentP3":16}
	[lp-1  ] DEBUG 2024/03/09 14:06:18 switched phases: 3p
Wahrscheinlich wegen Schaltschutz führt die Easee dies deutlich später aus. Warum nur 6A statt 16A ist mir rätselhaft, aber wahrscheinlich nur ein Folgefehler 

	[easee ] TRACE 2024/03/09 14:11:13 ProductUpdate ABCDEF: (2024-03-09 13:11:10 +0000 UTC) OUTPUT_CURRENT 6

Danach wiederholt sich das Spiel.

[image]

Wird ein zu häufiges Umschalten wie um 14:06 nicht eigentlich verhindert?

### Steps to reproduce

weniger Sonne

### Configuration details

```yaml
network:
  schema: http
  host: localhost
  port: 7070
log: debug
levels:
  easee: trace
sponsortoken: ***
plant: ***

meters:
- name: grid
  type: template
  template: vzlogger
  usage: grid
  host: ***
  port: 2071
  uuid: ***
- name: pv
  type: template
  template: solarlog 
  usage: pv  
  host: ***
  port: 2081

chargers:
- name: wallbox
  type: template
  template: easee
  user: ***
  password: ***
  charger: ABCDEF
[... 468 lines cut ...]
[lp-1  ] DEBUG 2024/03/09 14:15:51 charge currents: [0.515 0.003 0.347]A
[lp-1  ] DEBUG 2024/03/09 14:15:51 charge total import: 3393.995kWh
[lp-1  ] DEBUG 2024/03/09 14:15:51 charger status: B
[lp-1  ] DEBUG 2024/03/09 14:15:51 pv charge current: 4.23A = 0A + 4.23A (-2920W @ 3p)
[lp-1  ] DEBUG 2024/03/09 14:15:51 wake-up vehicle, attempts left: 4
[site  ] DEBUG 2024/03/09 14:15:52 ----
[lp-1  ] DEBUG 2024/03/09 14:15:52 charge power: 0W
[site  ] DEBUG 2024/03/09 14:16:22 pv power: 3102W
[site  ] DEBUG 2024/03/09 14:16:22 grid meter: -3213W
[site  ] DEBUG 2024/03/09 14:16:22 site power: -3213W
[lp-1  ] DEBUG 2024/03/09 14:16:22 charge currents: [0.515 0.003 0.347]A
[lp-1  ] DEBUG 2024/03/09 14:16:22 charge total import: 3393.995kWh
[lp-1  ] DEBUG 2024/03/09 14:16:22 charger status: B
[lp-1  ] DEBUG 2024/03/09 14:16:22 pv charge current: 4.66A = 0A + 4.66A (-3213W @ 3p)
[site  ] DEBUG 2024/03/09 14:16:22 ----
[lp-1  ] DEBUG 2024/03/09 14:16:22 charge power: 0W
```

### What type of operating system are you running?

Linux

### Version

0.124.8

andig, 2024-03-10:
Was ist hier die Fehlerbeschreibung?

kai-morich, 2024-03-10:
Es schaltet nicht von 3 Phasen auf 1 Phase runter, sondern wechselt periodisch zwischen 3 Phasen und Aus.
Das Problem tritt nicht immer, aber doch häufig auf.

andig, 2024-03-10:
Wenn für 1p nicht genug Leistung da ist- warum sollte er dann auf 1p gehen? "Aus" ist in diesem Fall richtig.

/cc @MarkusGH

kai-morich, 2024-03-10:
Es ist genügend Leistung für 1p vorhanden, wie im Grafana Chart zu sehen (linke Achse ist Watt, grün ist PV-Ertrag und blau Linie = hellgrüne Fläche die Wallbox)

Evcc ist wohl der Annahme das es gleich wieder auf 3p hochschalten soll, und die Easee ist dann erst mal ein paar Minuten beleidigt, weil die Umschaltung zu oft wäre.

    [lp-1  ] DEBUG 2024/03/09 14:06:18 available power 7068W > 4140W min 3p threshold

MarkusGH, 2024-03-10:
@andig: Ich glaube nicht, dass das mit https://github.com/evcc-io/evcc/pull/11878 zu tun hat.

Das Log sieht seltsam aus.
evcc scheint das richtige tun zu wollen:

[lp-1  ] DEBUG 2024/03/09 14:05:14 phase scale1p in 27s
[lp-1  ] DEBUG 2024/03/09 14:05:44 phase scale1p in 0s
[lp-1  ] DEBUG 2024/03/09 14:05:44 switched phases: 1p
[lp-1  ] DEBUG 2024/03/09 14:05:44 phase timer inactive
[lp-1  ] DEBUG 2024/03/09 14:05:44 projected site power 1008W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/09 14:05:44 pv disable timer start: 3m0s
[lp-1  ] DEBUG 2024/03/09 14:05:44 pv disable in 3m0s
[lp-1  ] DEBUG 2024/03/09 14:06:14 charger status: B
[lp-1  ] INFO 2024/03/09 14:06:14 stop charging <-

Aber die Easee bzw. die Anbindung führt es nicht richtig aus.
Habe einen Verdacht - schaue gleich mal nach.

GrimmiMeloni, 2024-03-10:
Irgendwie sehen die Messwerte komisch aus.

```
[site  ] DEBUG 2024/03/09 14:06:14 site power: -2831W
...
[lp-1  ] DEBUG 2024/03/09 14:06:18 available power 7068W > 4140W min 3p threshold
```

Die Site sieht 2.8kW verfügbaren Überschuss, und der Loadpoint 7kW. Oder sehe ich das falsch?

MarkusGH, 2024-03-10:
@GrimmiMeloni: Mach Dir keine Gedanken - ich denke ich weiß an was es liegt.

MarkusGH, 2024-03-10:
@GrimmiMeloni, @andig: Ich hatte schon ein schlechtes Gefühl wegen der Umbauten in syncCharger, dachte aber solange sich keiner beschwert wird es schon passen...

MarkusGH, 2024-03-10:
> Die Site sieht 2.8kW verfügbaren Überschuss, und der Loadpoint 7kW. Oder sehe ich das falsch?

@GrimmiMeloni: Unabhängig von der Ursache des Bugs hier ist das tatsächlich seltsam.