evcc-io/evcc #12826: [Feature Request] Get Information of my vehicles even if not connected

kabakakao, 2024-03-09:
**Is your feature request related to a problem? Please describe.**
Not really a Problem of evcc ;)

**Describe the solution you'd like**
It would be nice, if Evcc can get the Information of my vehicle, Even when it is not connected. 

**Describe alternatives you've considered**
I tried using a dedicated program for my FIAT500E to get these Information. Unfortunately I cannot use FiatChamp because it is just usable as Addon which I cannot use. I am running HomeAssistant and EVCC as Docker Containers 

I think,to have Range, ODO and maybe Location Even without connected will be useful also for other users

StefanSchoof, 2024-03-09:
Take a look at https://docs.evcc.io/en/docs/reference/configuration/loadpoints/#poll

kabakakao, 2024-03-09:
Great. That‘s what I was looking for. Sorry 🙈
Haven‘t seen it before

closed 2024-03-09 by kabakakao