evcc-io/evcc #12828: Varta Element

Dosepower, 2024-03-09:
### Describe the bug

habe im log einen error es scheint aber alles zu funktionieren.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: 
  
interval: 10s # control cycle interval  
  
meters:
- type: template  
  template: varta   
  usage: grid    
  host: 192.168.178.10    
  port: 502    
  name: grid1  
- type: template  
  template: kostal-piko-legacy   
  usage: pv    
  host: 192.168.178.32    
  user: pvserver
  password: 
  name: pv2
- type: template
  template: shelly-1pm 
  usage: pv  
  host: 192.168.178.105  
  name: pv3
- type: template  
  template: varta   
  usage: battery    
  host: 192.168.178.10    
  port: 502    
  name: battery4  
    
    
chargers:  
- type: template  
  template: tinkerforge-warp   
  host: 192.168.178.68    
  port: 1883    
  topic: Dosepower    
  user: 
  password: 
  name: wallbox3  
  
vehicles:  
- type: template  
  template: id   
  title: ID 4    
  user: 
  password: 
  vin: 
  capacity: 77    
  name: ev2
  phases: 3
  mode: pv
  mincurrent: 10
  maxcurrent: 16
- name: ev3
  type: template
  template: tesla
  title: Tesla Model Y
  accessToken: 
  refreshToken: 
  phases: 3
  mode: pv
  mincurrent: 10
  maxcurrent: 16
  
  
loadpoints:  
- title: Carport  
  charger: wallbox3  
  vehicle: ev2  
  mode: pv  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 32
  
site:  
  title: Zuhause  
  meters:  
    grid: grid1  
    pv:  
    - pv2
    - pv3
    battery:  
    - battery4 
  residualPower: 300

  
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.39 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.24 # [currency]/kWh
```

### Log details

```text
[site  ] ERROR 2024/03/09 20:17:43 battery 1 power: read failed: read tcp 192.168.178.68:59402->192.168.178.10:502: i/o timeout

kommt alle paar sekunden

erst seit der neusten version
```

### What type of operating system are you running?

HomeAssistant Add-on

andig, 2024-03-10:
Ein i/o timeout is genau das- Gerät antwortet nicht. Warum auch immer...

closed 2024-03-10 by andig

dr-waterstorm, 2024-05-09:
Ich habe ebenfalls eine Varta Element und genau das selbe Problem.
Das ganze scheint an der Batterie selbst zu liegen. Die offizielle Modbus Dokumentation von Varta sagt lustiger weiße auch "Modbus: Request interval <= 1/s"
Und tatsächlich: Ich habe das Modbus als Proxy eingetragen bei evcc und frage per HA config jede Sekunde die aktuellen Werte ab (für mein HA Energy) und siehe da: keine Fehler mehr im Log von evcc.

Möglicher Fix: In der config https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/varta.yaml `scan_interval: 1` bei allen Modbus Sensoren hinzufügen. Das funktioniert in meiner eigenen config von HA wunderbar. Allerdings würde das natürlich der globalen Einstellung von evcc "widersprechen".

andig, 2024-05-09:
Evcc hat kein scan—interval, wohl aber einen `delay` Parameter.

dr-waterstorm, 2024-05-09:
> Evcc hat kein scan—interval, wohl aber einen `delay` Parameter. Gerne PR.

Ah verstehe, danke für das schnelle Feedback. 
Das heißt da müsste man dann die Modbus Implementation anpassen um so einen Parameter zu unterstützen. Ich nehme an das wird dann bisher einfach über den globalen Interval geregelt.
Klingt nicht so einfach wie ich gehofft hatte.

Aktuell ist mein "quick workaround" einfach den Proxy zu nutzen und per HA modbus mit `scan_interval` öfters abzufragen als evcc.

zoomy01, 2025-02-20:
> Aktuell ist mein "quick workaround" einfach den Proxy zu nutzen und per HA modbus mit scan_interval öfters abzufragen als evcc.

sehr guter Workaround, hat mir ebenfalls geholfen und behebt die i/o timeout Fehler! Merkwürdig, dass zu diesem Fehler bei zu wenigen Abfragen kommt. In der Regel ist es ja eher andersherum...