evcc-io/evcc #12837: Grid Meter von Goodwe Hybrid liefert falschen Wert für Gesamteinspeisung

Klaus3d3, 2024-03-10:
### Describe the bug

Hi, wie auf den Fotos zu sehen ist das Komma bei dem Einspeisezähler verrutscht.

[image]

[image]

Aus 1633,18 wird 163.318kWh

### Steps to reproduce

Nothing

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: goodwe-hybrid
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 247
    host: 192.168.0.xxx # Hostname
    port: 502 # Port
```

### Log details

```text
111
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.9

andig, 2024-03-10:
@Klaus3d3 statt epischer Screenshots als Suchspiel wäre ein prägnantes "Energie um Faktor 100 zu gross" Klasse gewesen ;)

andig, 2024-03-10:
Closed in https://github.com/evcc-io/evcc/commit/5fb9da00b3ec5a25accd785a8509a321b8ebf256

closed 2024-03-10 by andig

Klaus3d3, 2024-03-10:
Da ich nicht weiß, an welcher Stelle in den Goodwe eingegriffen wird und überhaupt, warum ausgerechnet dieser Wert hier angezeigt wird, wollte ich den etwas größeren Überblick anbieten.