evcc-io/evcc #12839: EVCC crashes when reading coils

kscholty, 2024-03-10:
### Describe the bug

I have a custom wallbox configured that reads coils to determine the status.
In the past the datatype given there was not relevant and I used to use bool8.
This datatype is no longer supported, so I used uint16 and also tried int16. Both leads to a crash.

It seems the modbus `BoolGetter()` has been reworked in the latest release. It now uses `FloatGetter()` which crashes on the data, which probably is a single byte.

### Steps to reproduce

1. Configure the charger as shown below
2. Start EVCC
3. Connect a Vehicle
...

### Configuration details

```yaml
The charger definition I use is:

chargers:
- type: custom
  name: wallbox_custom
  tos: true
  status:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 100
      type: input
      decode: int16
  enabled:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 400
      type: coil
      decode: int16
  enable:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 400
      type: writecoil
      decode: int16
  maxcurrent:
    source: modbus
    id: 180
    uri: 192.168.100.8:502
    register:
      address: 300
      type: writeholding
      decode: uint16
  phases1p3p:
    source: script
    cmd: /etc/switch-phases.sh ${phases}
    timeout: 15s
```

### Log details

```text
karsten@evcc:~$ evcc -l debug
[main  ] INFO 2024/03/10 11:59:37 evcc 0.124.9
[main  ] INFO 2024/03/10 11:59:37 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/10 11:59:37 starting ui and api at :7070
[db    ] INFO 2024/03/10 11:59:37 using sqlite database: /home/karsten/.evcc/evcc.db
[mqtt  ] INFO 2024/03/10 11:59:37 connecting evcc-1145160712 at tcp://iobroker.home.lan:1885
[mqtt  ] DEBUG 2024/03/10 11:59:37 tcp://iobroker.home.lan:1885 connected
[lp-1  ] DEBUG 2024/03/10 11:59:40 charge total import: 761.844kWh
[lp-2  ] DEBUG 2024/03/10 11:59:40 charge total import: 0.000kWh
[site  ] WARN 2024/03/10 11:59:41 vehicle 'Roller' does not support automatic detection
[site  ] INFO 2024/03/10 11:59:41 site config:
[site  ] INFO 2024/03/10 11:59:41   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/10 11:59:41     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/10 11:59:41     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/10 11:59:41     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/10 11:59:41     battery 2: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/10 11:59:41   vehicles:
[site  ] INFO 2024/03/10 11:59:41     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2024/03/10 11:59:41     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/03/10 11:59:41     vehicle 3: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/03/10 11:59:41 loadpoint 1:
[lp-1  ] INFO 2024/03/10 11:59:41   mode:        pv
[lp-1  ] INFO 2024/03/10 11:59:41   charger:     power ✗ energy ✗ currents ✗ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/10 11:59:41   meters:      charge ✓
[lp-1  ] INFO 2024/03/10 11:59:41     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/03/10 11:59:41 loadpoint 2:
[lp-2  ] INFO 2024/03/10 11:59:41   mode:        pv
[lp-2  ] INFO 2024/03/10 11:59:41   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/03/10 11:59:41   meters:      charge ✓
[lp-2  ] INFO 2024/03/10 11:59:41     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/03/10 11:59:41 phase timer inactive
[lp-1  ] DEBUG 2024/03/10 11:59:41 pv timer inactive
panic: runtime error: index out of range [1] with length 1

goroutine 1 [running]:
encoding/binary.bigEndian.Uint16(...)
        encoding/binary/binary.go:143
github.com/volkszaehler/mbmd/encoding.Int16({0xc00010e7e9?, 0xc0007d7920?, 0xc000c0cde0?})
        github.com/volkszaehler/mbmd@v0.0.0-20231215091549-af16b1f597b9/encoding/number.go:19 +0x2e
github.com/evcc-io/evcc/util/modbus.Register.DecodeFunc.asFloat64[...].func1()
        github.com/evcc-io/evcc/util/modbus/register.go:235 +0x1c
github.com/evcc-io/evcc/provider.(*Modbus).FloatGetter.func1()
        github.com/evcc-io/evcc/provider/modbus.go:116 +0x82
github.com/evcc-io/evcc/provider.(*Modbus).BoolGetter.func1()
        github.com/evcc-io/evcc/provider/modbus.go:158 +0x13
github.com/evcc-io/evcc/charger.(*Charger).Enabled(...)
        github.com/evcc-io/evcc/charger/charger.go:159
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0xc001184c08, 0x0?, 0x52c8320?, 0xc00097d320?)
        github.com/evcc-io/evcc/core/loadpoint.go:639 +0xf35
github.com/evcc-io/evcc/core.(*Site).Prepare(0xc000bbfba0, 0xc000c8aa20, 0xc0016c2480)
        github.com/evcc-io/evcc/core/site.go:877 +0x17b
github.com/evcc-io/evcc/cmd.runRoot(0x522cb80, {0x28ff8d3?, 0x7?, 0x28f299b?})
        github.com/evcc-io/evcc/cmd/root.go:259 +0x124f
github.com/spf13/cobra.(*Command).execute(0x522cb80, {0xc000050640, 0x2, 0x2})
        github.com/spf13/cobra@v1.8.0/command.go:987 +0xab1
github.com/spf13/cobra.(*Command).ExecuteC(0x522cb80)
        github.com/spf13/cobra@v1.8.0/command.go:1115 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.8.0/command.go:1039
github.com/evcc-io/evcc/cmd.Execute()
        github.com/evcc-io/evcc/cmd/root.go:100 +0x1a
main.main()
        github.com/evcc-io/evcc/main.go:44 +0x2d

```
```

### What type of operating system are you running?

Linux

### Version

0.124.9

andig, 2024-03-10:
@kscholty könntest Du Portweiterleitung für info@evcc.io einrichten? Es ist nicht offensichtlich, warum das passiert.

kscholty, 2024-03-10:
Kann ich machen, weiß aber im Moment nicht genau, was du brauchst. SSH Zugriff auf den Host?

andig, 2024-03-10:
Das oder eine einfach Portweiterleitung auf den Modbus

kscholty, 2024-03-12:
Have eine E-Mail mit den Daten geschickt.

andig, 2024-03-13:
@kscholty Du hast ein `coil` mit `int16` Decoding- das geht nicht. Nimm bitte `bool8`, dann gehts.

andig, 2024-03-13:
Bei mir geht das:

```
      register:
        address: 400
        type: coil
        decode: bool8
```

kscholty, 2024-03-13:
Also in 0.124.9 und 09.124.10 geht es nicht, Fehlermeldung, "Invalid register encoding bool8".
bool geht überhaupt nicht, auch sowas wir uint8 oder int8 nicht.

Das ist aber schon länger so. Habe das in den developer Versionen schon beobachtet, aber dort ging bis vor Kurzem int16 noch. Ich hatte das damals so implementiert, dass der Typ intern gesetzt wurde und es deshalb egal war, was man da angab.

andig, 2024-03-13:
Wie gesagt. Bei mir gehts!

andig, 2024-03-13:
Bzw: bei Dir geht **schreiben** nicht, richtig?

kscholty, 2024-03-13:
> Bzw: bei Dir geht **schreiben** nicht, richtig? Im Issue ging es ja erstmal ums Lesen…

Ja, stimmt. Ich hatte in 0.124.9 den Eindruck beides ginge nicht, aber in der aktuellen Version geht das Lesen mit bool8.  Das Problem ist also weg.

Schreiben funktioniert allerdings auch nicht. Kann den Datentyp bool8 da nicht angeben und mit int16 gibt es beim Schreiben auf das Register eine Fehlermeldung.

```
[lp-1  ] DEBUG 2024/03/13 17:02:45 charger status: A
[lp-1  ] INFO 2024/03/13 17:02:45 car disconnected
[lp-1  ] DEBUG 2024/03/13 17:02:45 set session soc limit: 0
[lp-1  ] DEBUG 2024/03/13 17:02:45 set session energy limit: 0
[lp-1  ] ERROR 2024/03/13 17:02:45 charger disable: invalid func code: 5

```

closed 2024-03-15 by andig

kscholty, 2024-03-15:
Funktioniert. Danke :-)