evcc-io/evcc #12843: Huawei SUN2000 with SDongle & Power Sensor falsche Werte

kannbert101, 2024-03-10:
### Describe the bug

Habe keine ähnlichen Foren Einträge etc gefunden!

Es werden seit einiger Zeit die falschen Werte bei der Batterieladung/entladung ausgelesen, sodass bei der Autoladung parallel die Batterie geladen wird.

[image]
[image]

### Steps to reproduce

None

### Configuration details

```yaml
network:
  schema: http
  host: 49686a9f-evcc #evcc.local
  port: 7070

interval: 30s # control cycle interval

sponsortoken: ###############
plant: ###############

log: error
levels:
  site: debug
  lp-1: debug
  lp-2: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters

meters:
  - name: huawei_smartmeter
    type: template
    template: huawei-dongle-powersensor
    usage: grid    
    modbus: tcpip
    id: 1
    host: 192.168.20.222 # Hostname
    port: 502 # Port 

  - name: huawei_battery
    type: template
    template: huawei-dongle-powersensor
    usage: battery    
    modbus: tcpip
    id: 1
    host: ############ # Hostname
    port: 502 # Port

  - name: huawei_pv
    type: template
    template: huawei-dongle-powersensor
    usage: pv    
    modbus: tcpip
    id: 1
    host: ############ # Hostname
    port: 502 # Port 

  - name: fronius_pv
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: ############# # IP-Adresse oder Hostname

[... 163 lines cut ...]
[site  ] DEBUG 2024/03/10 13:27:57 battery power: -127W
[site  ] DEBUG 2024/03/10 13:27:57 grid meter: -93W
[site  ] DEBUG 2024/03/10 13:27:59 grid currents: [-4.79 -3.46 7.92]A
[site  ] DEBUG 2024/03/10 13:28:00 site power: 380W
[lp-1  ] DEBUG 2024/03/10 13:28:00 charge currents: [5.51 0 0]A
[lp-1  ] DEBUG 2024/03/10 13:28:00 detected active phases: 1p
[lp-1  ] DEBUG 2024/03/10 13:28:00 charge total import: 2682.585kWh
[lp-1  ] DEBUG 2024/03/10 13:28:00 charger status: C
[lp-1  ] DEBUG 2024/03/10 13:28:00 soc estimated: 81.00% (vehicle: 81.00%)
[lp-1  ] DEBUG 2024/03/10 13:28:00 vehicle soc: 81%
[lp-1  ] DEBUG 2024/03/10 13:28:00 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/03/10 13:28:00 vehicle range: 277km
[lp-1  ] DEBUG 2024/03/10 13:28:00 pv charge current: 4.35A = 6A + -1.65A (380W @ 1p)
[lp-1  ] DEBUG 2024/03/10 13:28:00 projected site power 380W >= 0W disable threshold
[lp-1  ] DEBUG 2024/03/10 13:28:00 pv disable in 2m29s
[lp-1  ] DEBUG 2024/03/10 13:28:00 pv disable timer remaining: 2m29s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.9

andig, 2024-03-10:
Bitte mal Nightly probieren. Da haben wir eine Änderung gerade zurück gedreht. 

/cc @cmusik

closed 2024-03-10 by andig

cmusik, 2024-03-10:
Das hier ist aber eine Dongle Konfiguration. Da dürfte die Smartlogger Anpassung keine Änderung bringen.

mucki12, 2024-03-10:
@kannbert101 
Ich prüfe das nachher noch einmal, wenn der Speicher wieder etwas zu tun hat. Meine ich konnte das bei meinen Ladevorgängen von gestern/vorgestern nicht beobachten. 
Jetzt gerade hat der Luna nichts zu tun und da kann ich das beschriebene Verhalten nicht nachvollziehen (zumindest davon ausgehend, dass die evcc Werte grundsätzlich und damit auch bei einer Leistung von 0 falsch wären):
[image]
[image]

kannbert101, 2024-03-10:
Kann es an meiner kaskadierten Batterie liegen? Liest evcc da vielleicht nicht alle Werte aus?

mucki12, 2024-03-10:
Bin da wegen fehlender Kaskade nicht so tief im Thema. Was passiert wenn du die Batterie 2x konfigurierst und bei der ersten "storageunit: 1" und bei der zweiten "storageunit: 2" mit gibst?