evcc-io/evcc #12848: Error upgrading to higher version than 0.124.6

Peter1984Git, 2024-03-10:
### Describe the bug

Reading changelogs... Done
(Reading database ... 43966 files and directories currently installed.)
Preparing to unpack .../evcc_0.124.9_armhf.deb ...
Unpacking evcc (0.124.9) over (0.124.6) ...
--------------------------------------------------------------------------------
ERROR: your evcc configuration is not compatible with the new version. Please consider reading the release notes: https://github.com/evcc-io/evcc/releases
checkconfig Output:
[main  ] INFO 2024/03/10 14:34:21 evcc 0.124.9
[main  ] INFO 2024/03/10 14:34:21 using config file: /etc/evcc.yaml
--------------------------------------------------------------------------------
Do you want to keep your old (working) evcc version? [Y/n]: 
y
The old version will be restored. Your evcc configuration stays untouched! Following errors are intended:
dpkg: warning: old evcc package post-removal script subprocess returned error exit status 1
dpkg: trying script from the new package instead ...
dpkg: error processing archive /var/cache/apt/archives/evcc_0.124.9_armhf.deb (--unpack):
 new evcc package post-removal script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/evcc_0.124.9_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

### Steps to reproduce

Upgrading from 0.124.6 to higher version

### Configuration details

```yaml
network:
schema: http
host: evcc.local # .local suffix announces the hostname on MDNS
port: 7070

log: warn
levels:
cache: error

plant: asdf

interval: 30s # control cycle interval

sponsortoken: asdf

telemetry: true

meters:

type: template
template: fronius-gen24
usage: grid
host: 192.168.178.39
port: 502
name: grid1
type: template
template: fronius-gen24
usage: pv
host: 192.168.178.39
port: 502
name: pv2
type: template
template: fronius-gen24
usage: battery
host: 192.168.178.39
port: 502
name: battery3
type: template
template: fronius-solarapi-v1
usage: pv
host: 192.168.178.38
name: pv4
type: template
template: fronius-solarapi-v1
usage: pv
host: 192.168.178.40
name: pvost
chargers:

type: template
template: alfen
id: 1
host: 192.168.178.110
port: 502
modbus: tcpip
name: wallbox6
vehicles:
name: ev5
type: template
template: id
title: ID.4 GTX
user: asdf
password: asdf
vin: asdf
capacity: 77
loadpoints:

title: Carport
charger: wallbox6
#mode: now = volle Leistung minpv = minimale Leistung pv = nur Ueberschuss off = aus
mode: pv
#phases: 0
#mincurrent: 6
#maxcurrent: 16
enable:
threshold: 0
delay: 2m
disable:
threshold: 200
delay: 5m
site:
title: Zuhause
meters:
grid: grid1
pv:
- pv2
- pv4
- pvost
battery:
- battery3
residualPower: 100

mqtt:
broker: 192.168.178.82:1883
topic: evcc
```

### Log details

```text
charging right now... i think the error is in the yaml-file but don´t know whats deprecated.
```

### What type of operating system are you running?

Docker container

### Version

0.124.6

andig, 2024-03-10:
@nicok1 @GrimmiMeloni habt ihr eine Idee was hier los ist? Insbesondere, warum nach

> checkconfig Output:

nix mehr kommt?

nicok1, 2024-03-10:
Ja, ist ein bekannter Fehler. Siehe hier: https://github.com/evcc-io/evcc/discussions/12736#discussioncomment-8708356

In diesem Fall kann der Rollback abgelehnt werden. Liegt am Loglevel.

andig, 2024-03-10:
Ich kann dir nicht folgen. Muss da einfach noch ein `--log error` mit dran?

nicok1, 2024-03-10:
Nein @andig. Blöderweise würde in der ersten Version des Features nach dem Output mit Grep geprüft. Die gesuchte Rückmeldung wird aber mit Loglevel Info oder niedriger nur angezeigt. Es liegt also kein Problem vor (deshalb auch kein Output), aber die Logik in der älteren Version denkt dies dennoch. Der Fehler ist dir aufgefallen und wir haben das in einem Pull Request auf Überprüfung des Returncodes umgestellt. Siehe https://github.com/evcc-io/evcc/pull/12547

Im konkreten Fall kann also die Frage nach Rollback verneint werden. Der Bug ist, dass diese Frage überhaupt gestellt wird.

andig, 2024-03-10:
Ok, da können wir nix machen. In https://github.com/evcc-io/evcc/pull/12849/files#diff-d423837e89a4fc456aaffed43820cd5f26b78eaf251cf4668c46040f6813ef13R29 würde ich (anderes Problem?) trotzdem nochmal sicher stellen wollen, dass *echte* Fehler auch tatsächlich angezeigt werden. Ok?

andig, 2024-03-10:
@Peter1984Git you'll just need to accept the upgrade, sorry for the inconvenience.

closed 2024-03-10 by andig

nicok1, 2024-03-10:
@andig 
Das global konfigurierte Loglevel für diesen Befehl zu überschreiben ist eine gute Idee. Kann ja durchaus sein, dass jemand Fehler unterdrückt hat und sich dann dort wundert. Dauert dann halt wieder + ein installiertes Release, bis das dann bei allen ankommt.

nicok1, 2024-03-10:
> Ok, da können wir nix machen. In https://github.com/evcc-io/evcc/pull/12849/files#diff-d423837e89a4fc456aaffed43820cd5f26b78eaf251cf4668c46040f6813ef13R29 würde ich (anderes Problem?) trotzdem nochmal sicher stellen wollen, dass _echte_ Fehler auch tatsächlich angezeigt werden. Ok?

@andig 
Klingt jetzt vielleicht blöd, aber wir könnten vielleicht etwas für die Leute tun, damit sie nicht in dieses Problem laufen, sofern sie aktuell eine noch fehlerhafte Version haben und dann die neueste Version installieren. Wir könnten die Logausgabe "config valid" im checkconfig Befehl bei gültiger Config auf Loglevel fatal setzen oder die Ausgabe einfach direkt an STDOUT senden. Was würde passieren?

Beim Upgrade würde die neuere Binary mit dieser Anpassung eingesetzt werden. Danach würde der Check (ausgeführt in der alten Version von postremove.sh) mit der neuen Binary mit grep bei gültiger Config in jedem Fall klappen.

andig, 2024-03-10:
Super Idee- in https://github.com/evcc-io/evcc/commit/266e51453ffc2d0d078f7be5d590494e68a24f42 umgesetzt. Klasse!