evcc-io/evcc #12853: Ladeplanung ist zu kleinlich

0x3d13f, 2024-03-10:
### Describe the bug

Ladeplanung mit Ende 07:15 PM eingestellt, Ladung läuft. Ab einer Aktualisierung des Estimators (im Log 17:49:23) wird der Plan in gelb angezeigt. In der Ladeplanung steht: "Zielzeit nicht erreichbar. Voraussichtliches Ende: heute 19:15.", was ja genau der Anforderung entspricht. Offenbar wird bereits ab 1s berechneter Überschreitung der Planzeit die Warnung erzeugt. Ggf. wäre hier ein Threshold von mehr als einer Minute sinnvoll, da die GUI ja nicht mit Sekunden arbeitet.
[image]

### Steps to reproduce

1. Ladung starten
2. Plan für 07:15 PM erstellen, der knapp ausreicht
3. Plan wird während der Ladung um mindestens eine Sekunde überschritten
4. Anzeige aus dem Screenshot verwirrt den Anwender, da die angezeigten Zeiten (Plan und geschätzt) gleich sind, also nichts im engeren Sinne überschritten wird

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: e3dc 
  usage: grid  
  host: e3dc  
  port: 502  
  name: grid1
- type: template
  template: e3dc 
  usage: pv  
  host: e3dc  
  port: 502  
  name: pv2
- type: template
  template: e3dc 
  usage: battery  
  host: e3dc  
  port: 502  
  name: battery3

chargers:
- type: template
  template: keba-modbus 
  id: 255  
  host: 192.168.25.26  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: kia 
  title: EV6  
  icon: car  
[... 122 lines cut ...]
Mar 10 17:50:23 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:23 soc estimated: 68.65% (vehicle: 68.00%)
Mar 10 17:50:23 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:23 vehicle soc: 69%
Mar 10 17:50:23 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:23 vehicle soc limit: 90%
Mar 10 17:50:23 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:23 vehicle range: 296km
Mar 10 17:50:23 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:23 plan: charge 1h24m40s between 2024-03-10 17:50:23 +0100 CET until 2024-03-10 19:15:00 +0100 CET (overruns by 3s, power: 11040W, avg cost: 179.870)
Mar 10 17:50:52 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:52 charge power: 10658W
Mar 10 17:50:52 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:52 charge currents: [15.7 15.6 15.7]A
Mar 10 17:50:52 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:52 detected active phases: 3p
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 charge total import: 5797.896kWh
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 charger status: C
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 soc estimated: 68.75% (vehicle: 68.00%)
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 vehicle soc: 69%
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 vehicle soc limit: 90%
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 vehicle range: 296km
Mar 10 17:50:53 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:50:53 plan: charge 1h24m11s between 2024-03-10 17:50:53 +0100 CET until 2024-03-10 19:15:00 +0100 CET (overruns by 4s, power: 11040W, avg cost: 179.899)
Mar 10 17:51:22 evcc evcc[186965]: [lp-1  ] DEBUG 2024/03/10 17:51:22 charge power: 10677W
```

### What type of operating system are you running?

Linux

### Version

0.124.8

VolkerK62, 2024-03-10:
das gab es vor kurzem schon. https://github.com/evcc-io/evcc/issues/12703
ist im aktuellen Nightly drin

0x3d13f, 2024-03-10:
Stimmt, hatte ich nicht gesehen. Dann mache ich hier zu.

closed 2024-03-10 by 0x3d13f