evcc-io/evcc #12858: Batteriesperre bei Growatt SpH

MaZ181, 2024-03-10:
Hallo zusammen,

mich wollte mal fragen ob es die Funktion Batteriesperre auch für den Growatt SpH Hybrid mit Batterie geben wird?

vielen Dank für die Info

andig, 2024-03-13:
Es braucht dafür eine Schnittstellendoku...

MaZ181, 2024-03-13:
Hallo danke für die Antwort ich konnte folgendes im Internet finden vielleicht geht es mit dem Parameter 
608 
[image]

andig, 2024-03-13:
Es braucht eine vollständige Doku

MaZ181, 2024-03-13:
Das konnte noch finden 

[Uploading New-Modbus.RS485.RTU.Protocal.Latest.Ver.pdf…]()

andig, 2024-03-13:
Dein Post hat keinen Ahnung :(

MaZ181, 2024-03-13:
[New-Modbus.RS485.RTU.Protocal.Latest.Ver.pdf](https://github.com/evcc-io/evcc/files/14587044/New-Modbus.RS485.RTU.Protocal.Latest.Ver.pdf)

MaZ181, 2024-03-13:
Nächster Versuch

andig, 2024-03-13:
Das hat geklappt, ich sehe aber keine passenden Register. Wenn Du welche findest bitte ausprobieren und mit Testergebnis wieder melden!

MaZ181, 2024-03-13:
Ok schade dacht sich kann so unterstützen da ich leider nur Anwender bin. Schade ☹️

MaZ181, 2024-06-07:
Hallo nochmals wir haben einen erfolgreiche Lösung umsetzen können um die Batterie-Sperre über Homeassistent im Growatt Wechselrichter zu realisieren anbei der link ins Forum. 

Wäre Mega wenn man das direkt in EVCC implementieren könnte.

https://www.photovoltaikforum.com/thread/193849-growatt-sph-monitoring-mit-homeassistent/?pageNo=12

Gruß

premultiply, 2024-06-07:
https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf

normal: 1044: 0
hold: 1044: 1
charge: 1044: 1 (uvm.)

> BatFirst bedeutet dass Dein Wechselrichter, wenn er denn Solarstrom produziert, diesen in die Batterie lädt anstatt in Deine Geräte oder ins Netz.
> 
> Um die Batterie aus dem Netz zu laden ist mehr nötig:
> 
> Batfirst, Startzeit setzen, Endzeit Setzen in Bereich 1, 2 oder 3, Bereich 1, 2 oder 3 aktivieren, AC-Laden erlauben.
> D.h. bevor die Batterie aus dem Netz geladen wird musst Du 5 Register beschreiben, und ohne eine 1 bei AC-Laden erlauben wird die Batterie nicht aus dem Netz geladen, selbst wenn man im Batfirst ist.

_

MaZ181, 2024-06-07:
Hier nochmals den direkten link wie das mit Homeassistent gelöst wurde 

https://www.photovoltaikforum.com/thread/193849-growatt-sph-monitoring-mit-homeassistent/?postID=3791609#post3791609

andig, 2024-09-14:
Zwischenzeitlich via https://github.com/evcc-io/evcc/pull/15680 gelöst. Closing.

closed 2024-09-14 by andig

nili14, 2024-11-14:
Hi andig 
 Suche gerade auch nach de Lösung 

habe einen sph hier 

wie hast du es jetzt gelöst 

Komme mit deinem letzten Post nicht klar wo du die Nummer verlinkt hast 

Danke