evcc-io/evcc #12862: Messaging: einige Variablen funktionieren nicht

TheNinth7, 2024-03-11:
### Describe the bug

Habe versucht Messaging mit Pushover einzurichten, und einige der [in der Dokumentation](https://docs.evcc.io/docs/reference/configuration/messaging/) beschriebenen Variablen scheinen nicht zu funktionieren. Nachdem das auch Variablen betrifft die in den [Beispielen in der Dokumentation](https://docs.evcc.io/docs/reference/configuration/messaging/) angeführt sind, vermute ich es handelt sich um einen Bug.

Was nicht funktioniert: 

- .vehicleTitle und .vehicleSoc in connect/disconnect bleiben leer, funktionieren aber in start/stop. 
- .activePhases in start bleibt auch leer. 

### Steps to reproduce

Z. B. Fahrzeug mit der unten angeführten Messaging-Konfiguration anschließen, die oben beschriebenen Werte fehlen dann in der Nachricht.

### Configuration details

```yaml
messaging:
  services:
    - type: pushover
      app: removed
      recipients:
        - removed
  events:
    start:
      title: "Charge of {{.vehicleTitle}} started"
      msg: |
        Wallbox {{.title}} started {{.activePhases}}-phase charging {{.vehicleTitle}} in {{ toString .mode | upper }} mode, with {{round .vehicleSoc 1}}% SoC.
    stop:
      title: "Charge of {{.vehicleTitle}} finished"
      msg: |
        Wallbox {{.title}} finished charging {{.vehicleTitle}}, with {{round .vehicleSoc 1}}% SoC.
    connect:
      title: "{{.vehicleTitle}} connected on wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} connected on wallbox {{.title}} with {{round .vehicleSoc 1}}% SoC.
    disconnect:
      title: "{{.vehicleTitle}} disconnected of wallbox {{.title}}"
      msg: |
        {{.vehicleTitle}} disconnected of wallbox {{.title}} with {{round .vehicleSoc 1}}% SoC.

```
```

### Log details

```text
In den Log Dateien finde ich nichts zu den rausgeschickten Messages.
```

### What type of operating system are you running?

Docker container

### Version

0.124.9

closed 2024-03-31 by andig

andig, 2024-03-31:
> .vehicleTitle und .vehicleSoc in connect/disconnect bleiben leer, funktionieren aber in start/stop.

@TheNinth7 den soc kann es beim Connect noch nicht geben- der muss ja erst gelesen werden. Disconnect sollte eigentlich gehen. Ich vermute das ist eine Folge von https://github.com/evcc-io/evcc/issues/7193 wo zum Zeitpunkt der Nachricht das Fahrzeug schon vollständig abgemeldet ist.

TheNinth7, 2024-04-09:
> @TheNinth7 den soc kann es beim Connect noch nicht geben- der muss ja erst gelesen werden. Disconnect sollte eigentlich gehen. Ich vermute das ist eine Folge von #7193 wo zum Zeitpunkt der Nachricht das Fahrzeug schon vollständig abgemeldet ist.

Ja, verstehe. Aus Benutzersicht wäre es schon interessant das Connect erst dann auszulösen wenn die Daten zu Verfügung stehen, oder vielleicht eine eigene Message für die erfolgte Identifizierung des Fahrzeugs zu haben. Aber das ist dann wohl mehr ein Change Request als eine Issue.

Eventuell wäre es gut die Dokumention anzupassen und die Beispiele dort zu korrigieren. Könnte das gerne machen, bräuchte aber etwas Input dazu welche Variablen wann zur Verfügung stehen.

andig, 2024-04-10:
@TheNinth7 

> bräuchte aber dazu etwas Input dazu welche Variablen wann zur Verfügung stehen.

Welche sollte einfach sein- alles was im Code hinter `publish()` vorkommt. Wann lässt sich so nicht einfach beantworten- da würde ich aktuell auch keinen Aufwand rein stecken wollen.