evcc-io/evcc #12870: Fehlermeldung Aktive Batteriesteuerung SAX Power Akku

Sebreiter, 2024-03-11:
### Describe the bug

Hallo zusammen,
erst einmal Danke für die schnelle Umsetzung.
Ich habe natürlich sofort den Schalter auf "EIN" gestellt. Leider erscheint daraufhin folgende Fehlermeldung.
ERROR 2024/03/06 10:36:47 battery mode: read tcp 192.168.178.20:45390->192.168.178.39:3600: i/o timeout
192.168.178.20=HA
192.168.178.39=SAX
Was will er denn da lesen?
Bei mir läuft das Ganze unter HA. Kann das daran liegen?
Wenn ich den Schalter auf "AUS" stelle, erscheint keine Fehlermeldung.
Viele Grüße, Reiner

### Steps to reproduce

1. Wenn der Schalter "Verhindere Entladung im Schnell-Modus..." auf Ein geschaltet wird, erscheint die Fehlermeldung.
2. Nach dem Ausschalten erscheint sie nicht mehr.
3.
...

### Configuration details

```yaml
Konfiguration:
[main  ] INFO 2024/03/09 14:34:02 evcc 0.124.7
[main  ] INFO 2024/03/09 14:34:02 using config file: /etc/evcc.yaml
[lp-1  ] WARN 2024/03/09 14:34:05 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/09 14:34:05 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/09 14:34:05 deprecated: phases setting is ignored, please remove
[lp-2  ] WARN 2024/03/09 14:34:05 deprecated: minCurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/09 14:34:05 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/09 14:34:05 deprecated: phases setting is ignored, please remove
[lp-3  ] WARN 2024/03/09 14:34:05 deprecated: minCurrent setting is ignored, please remove
[lp-3  ] WARN 2024/03/09 14:34:05 deprecated: maxcurrent setting is ignored, please remove
[lp-3  ] WARN 2024/03/09 14:34:05 deprecated: phases setting is ignored, please remove
[lp-4  ] WARN 2024/03/09 14:34:05 deprecated: minCurrent setting is ignored, please remove
[lp-4  ] WARN 2024/03/09 14:34:05 deprecated: maxcurrent setting is ignored, please remove
[lp-4  ] WARN 2024/03/09 14:34:05 deprecated: phases setting is ignored, please remove
[lp-5  ] WARN 2024/03/09 14:34:05 deprecated: minCurrent setting is ignored, please remove
[lp-5  ] WARN 2024/03/09 14:34:05 deprecated: maxcurrent setting is ignored, please remove
[lp-5  ] WARN 2024/03/09 14:34:05 deprecated: phases setting is ignored, please remove
[lp-1  ] ERROR 2024/03/09 14:34:05 charge total import: write tcp 192.168.178.70:45856->192.168.178.36:502: write: connection reset by peer
[lp-2  ] DEBUG 2024/03/09 14:34:06 charge total import: 11.989kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

sponsortoken: *****

# unique installation id
plant: *****

interval: 10s # control cycle interval

meters:
#- name: grid
#  type: template
[... 389 lines cut ...]
Mar 09 14:32:29 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:29 battery soc: 98%
Mar 09 14:32:29 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:29 battery power: -1W
Mar 09 14:32:29 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:29 grid meter: -2742W
Mar 09 14:32:29 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:29 site power: -2593W
Mar 09 14:32:40 energie evcc[1247]: [site  ] ERROR 2024/03/09 14:32:40 battery mode: read tcp 192.168.178.70:36866->192.168.178.39:3600: i/o timeout
Mar 09 14:32:40 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:40 ----
Mar 09 14:32:40 energie evcc[1247]: [lp-1  ] DEBUG 2024/03/09 14:32:40 charge power: 0W
Mar 09 14:32:40 energie evcc[1247]: [lp-2  ] DEBUG 2024/03/09 14:32:40 charge power: 0W
Mar 09 14:32:40 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:40 pv power: 3118W
Mar 09 14:32:41 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:41 battery soc: 98%
Mar 09 14:32:41 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:41 battery power: 2W
Mar 09 14:32:41 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:41 grid meter: -2788W
Mar 09 14:32:41 energie evcc[1247]: [site  ] DEBUG 2024/03/09 14:32:41 site power: -2636W
Mar 09 14:32:41 energie evcc[1247]: [lp-1  ] DEBUG 2024/03/09 14:32:41 charge currents: [0 0 0]A
Mar 09 14:32:41 energie evcc[1247]: [lp-1  ] DEBUG 2024/03/09 14:32:41 charge total import: 7734.005kWh
Mar 09 14:32:41 energie evcc[1247]: [lp-1  ] DEBUG 2024/03/09 14:32:41 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.124.7

andig, 2024-03-11:
Bitte mal (ohne HA!):

    evcc meter --log error,modbus:trace -b hold

Sebreiter, 2024-03-12:
evcc meter --log error,modbus:trace

[main  ] INFO 2024/03/12 17:36:55 evcc 0.124.9
[main  ] INFO 2024/03/12 17:36:55 using config file: /etc/evcc.yaml
grid
----
[modbus] TRACE 2024/03/12 17:36:56 modbus: send 00 01 00 00 00 06 40 03 00 30 00 01
[modbus] TRACE 2024/03/12 17:36:56 modbus: recv 00 01 00 00 00 05 40 03 02 3f ea
Power: -22W

pv1
---
Power:  6W
Energy: 4506.1kWh

pv2
---
Power:  6W
Energy: 4317.3kWh

battery
-------
[modbus] TRACE 2024/03/12 17:36:56 modbus: send 00 02 00 00 00 06 40 03 00 2f 00 01
[modbus] TRACE 2024/03/12 17:36:56 modbus: recv 00 02 00 00 00 05 40 03 02 41 24
[modbus] TRACE 2024/03/12 17:36:56 modbus: send 00 03 00 00 00 06 40 03 00 2e 00 01
[modbus] TRACE 2024/03/12 17:36:56 modbus: recv 00 03 00 00 00 05 40 03 02 00 59
Power: 292W
Soc:   89%

sd
--
Power:  0W
Energy: 12.0kWh

evcc meter --log error,modbus:trace -b hold

[main  ] INFO 2024/03/12 17:36:10 evcc 0.124.9
[main  ] INFO 2024/03/12 17:36:10 using config file: /etc/evcc.yaml
[modbus] TRACE 2024/03/12 17:36:11 modbus: send 00 01 00 00 00 06 40 06 00 2b 00 00
[main  ] FATAL 2024/03/12 17:36:21 set battery mode: read tcp 192.168.178.70:38080->192.168.178.39:3600: i/o timeout

Gruß, Reiner

andig, 2024-03-13:
@premultiply magst Du mal rein schauen?

abias, 2024-03-26:
Hallo zusammen,

ich habe zwei SAX Homespeicher, die im Verbund laufen, und beide sind an EVCC angebunden. EVCC läuft bei mir als standalone Software. 

Das SAX Template macht(e) genau das was es soll, vielen Dank für die gute Arbeit!

Seit dem EVCC Update von v0.124.0 auf v0.124.10, das ich jüngst gemacht habe, sehe ich aber auch immer mal wieder vergleichbare Fehlermeldungen in der EVCC Weboberfläche:
`battery mode: read tcp 192.168.178.70:41732->192.168.178.30:3600: i/o timeout`

Wenn weitere konkrete Logs oder Tests von einer weiteren Installation hilfreich sind, kann ich diese gerne auf Zuruf liefern.

andig, 2024-03-26:
> sehe ich aber auch immer mal wieder

Der Speicher antwortet nicht. Netzwerkproblem oder Fehler der Speicherfirmware. Anders als oben, denn da scheint es ja nie zu funktionieren. In 124.0 gab es noch kein Battery Control für den Sax?

Sebreiter, 2024-03-26:
Hallo @andig ,
soweit ich weiß, war die Version 0.124.6 die letzte, ohne Battery Control für den SAX Speicher.

Hallo @abias , rein aus Interesse, wie groß ist Deine PV Anlage? Ich habe 7.2 kWp auf dem Dach und einen SAX.

Viele Grüße, Reiner

andig, 2024-03-26:
> soweit ich weiß, war die Version 0.124.6 die letzte, ohne Battery Control für den SAX Speicher.

Was ja dann erklärt warum eine Version on das Feature keinen Fehler mit dem Feature zeigen kann. In beiden Fällen- Eure Speicher antworten nicht. Ich mache hier mal zu- anscheinend ein Thema für SAX...

closed 2024-03-26 by andig

andig, 2024-03-27:
Genau. Keine Antwort innerhalb timeout.

juergen-weber, 2024-06-13:
@abias 
Bist Du bei der Einbindung des Sax in EVCC schon weitergekommen? Wenn das nicht nicht der Fall sein sollte, dann ließ bitte mal diesen 
[Thread](https://community.simon42.com/t/einbindung-speicher-ueber-modbus-tcp/13450/21?page=2)
Es gibt mehrere Firmware Versionen, die und Nutzern nicht bekannt sind. Davon abhängig ist auch die Modbus Implementierung. Zur Zeit wird der Port 502 für Modbus - TCP genutzt und nicht mehr 3600. Will man schreibend auf die Register 43,44 zugreifen, dann muss der Support dies freischalten und man muss selber die Batteriesteuerung für Laden/Entladen übernehmen!!!! Dieses Verhalten soll im Laufe des Jahres geändert werden. Ein genauer Termin ist aber nicht bekannt. 

Jürgen

Butze1978, 2024-09-20:
Hallo zusammen,
gibt es inzwischen ein Skript zur Abschaltung des Sax Speichers, beim Laden mit einem go-e Charger?

Als Programm wir EVCC verwendet, in Verbindung eines Raspberry.

Gruß
Patrick