evcc-io/evcc #12872: Force Vehicle units in Miles (not Km) as part of evcc.yaml ?

IamMattM, 2024-03-11:
How can one set the vehicle to `Miles` for an `ovms` vehicle type ?

The units are correctly set to "Miles" in the OVMS module.

`Soc:           82%
Capacity:      28.0kWh
Charge status: B
Range:         92km
Odometer:      0km
Finish time:   2024-03-10 13:05:00 +0000 GMT
Position:      50.8xxx59,-1.2xxx62
Features:      []
/app # `

The value "92km" in the capture above for command `evcc vehicle` should say "92Miles"

Because my EVCC U.I. has been set to Miles, the actual range is displayed as "57 mi" (92/1.6 approx)

Please help.

IamMattM, 2024-03-11:
Illustration of the issue justifying a new evcc.yaml "unit" parameter:

https://imgur.com/a/cRXH3Nf

IamMattM, 2024-03-11:
The OVMS module can be configured to either Km or Miles but of course if I was to change it top Km so that it becomes compatible with the way evcc would want to receive the data, it will then break the OVMS App as it will then show everything in Km which is not desirable.

I think the case for adding a km/miles setting in evcc.yaml (just like the ovms module has) still stand and is valid a request.

https://imgur.com/a/3z101H0

andig, 2024-03-11:
How do we know how ovms is configured?

IamMattM, 2024-03-11:
> How do we know how ovms is configured?

I don't think the OVMS box configuration can be queried for this info.

This is why I thought that adding a parameter to the evcc.yaml to indicate the unit used by the vehicle integration would cover the issue.

IamMattM, 2024-03-11:
Actually the ovms API seems to suggest the vehicle units can be queried:

https://docs.openvehicles.com/en/latest/protocol_httpapi/requests.html

andig, 2024-03-11:
Which one?

IamMattM, 2024-03-11:
@andig 

https://imgur.com/Mr2oe2L

IamMattM, 2024-03-11:
I imagine that the following from `https://github.com/evcc-io/evcc/blob/master/vehicle/ovms.go`
Could be adapted to pull the `"units"` like it already pulls the `"estimatedrange"`

`type ovmsChargeResponse struct {
	ChargeEtrFull    int64   `json:"charge_etr_full,string"`
	ChargeState      string  `json:"chargestate"`
	ChargePortOpen   int     `json:"cp_dooropen"`
	EstimatedRange   string  `json:"estimatedrange"`
	MessageAgeServer int     `json:"m_msgage_s"`
	Soc              float64 `json:"soc,string"`
}
`

IamMattM, 2024-03-11:
I imagine that the following from : https://github.com/evcc-io/evcc/blob/master/vehicle/ovms.go#L49

could be adapted to also pull the `"units"`

```
type ovmsStatusResponse struct {
	Odometer float64 `json:"odometer,string"`
}
```

andig, 2024-03-12:
Does the `battery` api that returns the remaining range have the same `unit` entry?

IamMattM, 2024-03-12:
@andig 
Just checked the latest workaround but I think there is a minor typo ?

package ovms

```
type StatusResponse struct {
	Unit     string  `json:"unit"`
	Odometer float64 `json:"odometer,string"`
}

type ChargeResponse struct {
	ChargeEtrFull    int64   `json:"charge_etr_full,string"`
	ChargeState      string  `json:"chargestate"`
	ChargePortOpen   int     `json:"cp_dooropen"`
	EstimatedRange   int64   `json:"estimatedrange,string"`
	MessageAgeServer int     `json:"m_msgage_s"`
	Soc              float64 `json:"soc,string"`
}

```

It is  `"units"` with an "s" at the end rather than `"unit"`

https://github.com/evcc-io/evcc/pull/12883/commits/91703aea5866443c8a2c3441cde5862fe669914d

Also, as mentionned above, the same `"units"` could be retrieved as part of the

closed 2024-03-13 by andig

IamMattM, 2024-03-13:
@andig 
Thank you for finding a resolution to the bug so quickly.
I can confirm it is working as expected now.

Ovms: honor units , fixed: 
https://imgur.com/a/TxuV9hO

IamMattM, 2025-12-16:
@andig 

Seems like evcc v0.211.1 is no longer honoring the units in Miles from an OVMS vehicle as per the resolution above.

Not sure when the fix disappeared as I have not used evcc for some time.

andig, 2026-01-02:
Probably another api change