evcc-io/evcc #12873: Auswertung Ladeenergie

Instanz1, 2024-03-11:
### Describe the bug

Ich habe festgestellt das die Auswertung der Ladeenergie nicht mehr funktioniert.

### Steps to reproduce

1. Keine Ahnung, habe immer nur die Updates eingespielt
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 4558c562a212404e35019d8881d69eaa74db07032424b3e6bdf6cf47a08a8c01

interval: 10s # control cycle interval

sponsortoken: habe ich

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.XXX
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.XXX  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.XXX
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.XXX
  name: garage
- type: template
  template: go-e 
  host: 192.168.178.XXX
  name: carport

vehicles:
- type: template
  template: opel 
  title: Opel Corsa  
  user: XXX 
  password: XXX  
  vin: XXXXX  
  capacity: 50  
  name: ev5

loadpoints:
- title: Garage
  charger: garage
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16

- title: Carport
  charger: carport
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2423
  feedin:
    type: fixed
    price: 0.1795
  planner:
    type: gruenstromindex
    zip: XXX

mqtt:
  broker: 192.168.XXX.XXX:1900
  topic: evcc
  clientid: evcc
  user: admin
  password: hab ich
```

### Log details

```text
root@evcc:~# evcc --log debug
[main  ] INFO 2024/03/11 17:13:01 evcc 0.124.9
[main  ] INFO 2024/03/11 17:13:01 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/11 17:13:01 starting ui and api at :7070
[db    ] INFO 2024/03/11 17:13:01 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/11 17:13:01 connecting evcc at tcp://192.168.178.201:1900
[mqtt  ] DEBUG 2024/03/11 17:13:01 tcp://192.168.178.201:1900 connected
[main  ] ERROR 2024/03/11 17:13:03 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"
[lp-1  ] WARN 2024/03/11 17:13:03 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/11 17:13:03 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/11 17:13:03 deprecated: minCurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/11 17:13:03 deprecated: maxcurrent setting is ignored, please remove
[lp-2  ] WARN 2024/03/11 17:13:03 deprecated: phases setting is ignored, please remove
[main  ] ERROR 2024/03/11 17:13:03 failed configuring planner tariff: invalid tariff type: gruenstromindex
[lp-1  ] DEBUG 2024/03/11 17:13:03 charge total import: 3676.234kWh
[lp-2  ] DEBUG 2024/03/11 17:13:03 charge total import: 90.000kWh
[site  ] WARN 2024/03/11 17:13:03 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/03/11 17:13:04 site config:
[site  ] INFO 2024/03/11 17:13:04   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/11 17:13:04     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/11 17:13:04     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/11 17:13:04     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/11 17:13:04   vehicles:
[site  ] INFO 2024/03/11 17:13:04     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/11 17:13:04 loadpoint 1:
[lp-1  ] INFO 2024/03/11 17:13:04   mode:        pv
[lp-1  ] INFO 2024/03/11 17:13:04   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/11 17:13:04   meters:      charge ✓
[lp-1  ] INFO 2024/03/11 17:13:04     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/03/11 17:13:04 loadpoint 2:
[lp-2  ] INFO 2024/03/11 17:13:04   mode:        pv
[lp-2  ] INFO 2024/03/11 17:13:04   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/03/11 17:13:04   meters:      charge ✓
[lp-2  ] INFO 2024/03/11 17:13:04     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/11 17:13:04 phase timer inactive
[lp-1  ] DEBUG 2024/03/11 17:13:04 pv timer inactive
[lp-2  ] DEBUG 2024/03/11 17:13:04 phase timer inactive
[lp-2  ] DEBUG 2024/03/11 17:13:04 pv timer inactive
[main  ] FATAL 2024/03/11 17:13:04 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

0.124.9

Instanz1, 2024-03-11:
Hier einmal angehängte bilder:
[image]
[image]

docolli, 2024-03-12:
Kann es sein, dass du evcc manuell mit einem anderen user-Kontext gestartet hast und nicht als Service? Laut Log nutzt evcc diese Datenbank-Datei `/root/.evcc/evcc.db`. 

Läuft evcc normalerweise als user root bei Dir? Ich denke nicht. Hat micht auch schon mal beim Testen verwirrt, dass auf einmal andere historische Werte (und andere inzwischen in der DB hinterlegte Werte), bis ich auf den Pfad geachtet habe.

Als Service gestartet nutzt evcc bei mir diese Datei `/var/lib/evcc/evcc.db`.

docolli, 2024-03-12:
Noch ein paar Anmerkungen zu deiner config.yaml. Es gab einige breaking changes mit 0.124, darunter ist, dass `mincurrent` und `maxcurrent´ in der yaml veraltet sind und nun übers GUI gesetzt werden.

https://github.com/evcc-io/evcc/releases/tag/0.124.0
https://docs.evcc.io/docs/reference/configuration/loadpoints/#mincurrent

```yaml
loadpoints:
- title: Garage
  charger: garage
  mode: pv
  #phases: 0
  #mincurrent: 6
  #maxcurrent: 16

- title: Carport
  charger: carport
  mode: pv
  #phases: 3
  #mincurrent: 6
  #maxcurrent: 16
```

Dann
[site  ] WARN 2024/03/11 17:13:03 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

```yaml
site:
  title: Mein Zuhause
  residualPower: 100
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery

```

Bei tariffs gibt es keinen planner, sondern co2:
```yaml
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2423
  feedin:
    type: fixed
    price: 0.1795
  planner:
    type: gruenstromindex
    zip: XXX
```

Noch ein Problem: [main  ] ERROR 2024/03/11 17:13:03 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'opel': login failed: oauth2: "server_error" "Internal Server Error"

So sieht das Beispiel für Opel aus:

```yaml
vehicles:
  - name: my_car
    type: template
    template: opel
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: WP0... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 50 # Akkukapazität in kWh (optional) 
```

Schau DIr mal die Beispiel yaml an: https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml

Ich hoffe das hilft Dir weiter. 😃

docolli, 2024-03-14:
```yaml
# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db
```

So hab ich das bei mir umgesetzt.