evcc-io/evcc #12884: vehicle Smart #1 not working

cyberways, 2024-03-12:
### Describe the bug

Good morning,
we use evcc with a BMW i3s for several months now and are very happy with it. Thank you for such great software!
Now we have a Smart #1 that also should be supported by evcc, but does not show the SoC and support charging to a selected SoC. Config. and logs see below.
As I read in the docs smart does not support autoselection of the vehicle connected. Is it normal behaviour that adding the smart to the config also disables auto-detection of the BMW that formerly worked fine, or do we have a bug here?

### Steps to reproduce

see above.

### Configuration details

```yaml
vehicles:
- type: template
  template: bmw
  title: BMW i3s
  user: ***
  password: ***
  vin: ***
  capacity: 40
  icon: car
  cache: 6m
  maxCurrent: 9
  name: bmwi3s

- name: smart
  type: template
  template: smart-hello
  title: smart
  icon: car
  cache: 6m
  capacity: 62
  maxCurrent: 10
  user: ***
  password: ***
```

### Log details

```text
./evcc vehicle smart

smart
-----
.Soc:         json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Capacity:     62.0kWh
Range:        json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Odometer:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Climater:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Position:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
OnIdentified: MaxCurrent:10
Features:     []

and logfile in every output block while evcc is charging the smart:

Mar 09 13:26:07 evcc[23227]: [lp-1  ] ERROR 2024/03/09 13:26:07 vehicle soc: json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
```

### What type of operating system are you running?

Linux

### Version

0.124.9

VolkerK62, 2024-03-12:
try it by adding the `vin` to the vehicle config.

andig, 2024-03-13:
Bitte mal

    evcc vehicle  --log trace smart

cyberways, 2024-03-13:
VIN zur Konfig. hinzugefügt (ohne Erfolg) und log trace gestartet.

Login bei smart erfolgreich, liefert eine Menge JSON mit sehr persönlichen Daten und einen Login-Token.
Beschaffung des Access-Token und Session-Token klappt anschließend auch.
Von Interesse sind wohl die letzten beiden Zeilen des Dialogs. Die Daten kommen alle, und damit sollte eigentlich auch eine automatische Erkennung des ladenden Fahrzeugs möglich sein. Danke!

--
[smart-hello] TRACE 2024/03/13 13:25:27 GET https://api.ecloudeu.com/remote-control/vehicle/status/HESXA*****?latest=true&target=&userId=*****
[smart-hello] TRACE 2024/03/13 13:25:27 {"code":"1000","data":{"result":{"serviceResult":{"error":null,"operationResult":1},"sessionId":"PS048*****"},"vehicleStatus":{"basicVehicleStatus":{"usageMode":"1","engineStatus":"engine_off","position":{"altitude":"","posCanBeTrusted":"","latitude":"","carLocatorStatUploadEn":"false","marsCoordinates":"","longitude":""},"carMode":"0","speed":"0.0","speedValidity":"false","direction":""},"notification":{"notifForEmgyCallStatus":"0"},"eg":{"enableRunning":"false","blocked":{"status":"0"},"panicStatus":"false"},"parkTime":{"status":"1710332486825"},"configuration":{"propulsionType":"4","fuelType":"4","vin":"HESXA*****"},"updateTime":"****","additionalVehicleStatus":{"maintenanceStatus":{"tyreTempWarningPassengerRear":"0","daysToService":"349","engineHrsToService":"500","odometer":"1904.000","brakeFluidLevelStatus":"3","tyreTempDriverRear":"12.000","tyreTempWarningPassenger":"0","tyreTempWarningDriverRear":"0","mainBatteryStatus":{"stateOfCharge":"1","chargeLevel":"94.3","energyLevel":"0","stateOfHealth":"0","powerLevel":"0","voltage":"14.225"},"tyreTempDriver":"12.000","tyreTempPassengerRear":"12.000","tyrePreWarningDriver":"0","distanceToService":"28096","tyrePreWarningPassengerRear":"0","tyreTempWarningDriver":"0","tyreStatusPassengerRear":"258.124","tyreStatusPassenger":"249.886","tyreStatusDriverRear":"255.378","serviceWarningStatus":"0","tyreStatusDriver":"258.124","tyreTempPassenger":"12.000","tyrePreWarningDriverRear":"0","tyrePreWarningPassenger":"0","washerFluidLevelStatus":"1"},"electricVehicleStatus":{"disChargeUAct":"0.0","disChargeSts":"0","wptFineAlignt":"0","chargeLidAcStatus":"2","distanceToEmptyOnBatteryOnly":"269","distanceToEmptyOnBattery100Soc":"324","chargeSts":"0","averPowerConsumption":"-83.8","chargerState":"2","timeToTargetDisCharged":"2047","distanceToEmptyOnBattery20Soc":"65","disChargeConnectStatus":"3","chargeLidDcAcStatus":"1","dcChargeSts":"0","ptReady":"0","chargeLevel":"89","statusOfChargerConnection":"3","dcDcActvd":"1","indPowerConsumption":"1000","dcDcConnectStatus":"0","disChargeIAct":"0.0","dcChargeIAct":"-6.9","chargeUAct":"407.7","bookChargeSts":"0","chargeIAct":"6.000","timeToFullyCharged":"155"},"chargeHvSts":"1","drivingBehaviourStatus":{"gearAutoStatus":"0","gearManualStatus":"0","engineSpeed":"0.000"},"runningStatus":{"ahbc":"0","goodbye":"0","homeSafe":"0","cornrgLi":"0","frntFog":"0","stopLi":"0","tripMeter1":"1050.3","approach":"0","tripMeter2":"26.1","indFuelConsumption":"0","hiBeam":"0","engineCoolantLevelStatus":"3","fuelEnLevel":"0","loBeam":"0","posLiRe":"0","ltgShow":"0","welcome":"0","drl":"0","fuelLevelPct":"0","ahl":"0","fuelEnCns":"0","trunIndrLe":"0","trunIndrRi":"0","afs":"0","dbl":"0","avgSpeed":"**","posLiFrnt":"0","reverseLi":"0","hwl":"0","reFog":"0","flash":"0","allwl":"0","fuelEnCnsFild":"0"},"trailerStatus":{"trailerTurningLampSts":"0","trailerFogLampSts":"0","trailerBreakLampSts":"0","trailerReversingLampSts":"0","trailerPosLampSts":"0"},"climateStatus":{"drvHeatSts":"0","winPosDriver":"0","rrVentDetail":"0","rlVentSts":"0","passVentSts":"0","interiorTemp":"19.300","passVentDetail":"0","sunroofPos":"101","cdsClimateActive":"false","sunroofOpenStatus":"1","rrHeatingDetail":"0","winStatusPassenger":"2","fragActive":false,"winStatusDriver":"2","drvVentSts":"0","winStatusPassengerRear":"2","sunCurtainRearOpenStatus":"1","preClimateActive":false,"rlHeatingDetail":"0","winPosPassengerRear":"0","curtainPos":"0","rlVentDetail":"0","curtainOpenStatus":"1","climateOverHeatProActive":"true","rrVentSts":"0","rrHeatingSts":"0","winPosPassenger":"0","steerWhlHeatingSts":"2","drvVentDetail":"0","winPosDriverRear":"0","exteriorTemp":"11.500","rlHeatingSts":"0","winStatusDriverRear":"2","defrost":"false","drvHeatDetail":"2","passHeatingDetail":"2","airBlowerActive":"false","sunCurtainRearPos":"101","passHeatingSts":"0"},"drivingSafetyStatus":{"doorLockStatusDriverRear":"1","srsCrashStatus":"0","doorOpenStatusPassengerRear":"0","doorPosPassengerRear":"0","doorOpenStatusDriver":"0","seatBeltStatusPassenger":"false","doorPosDriver":"0","seatBeltStatusThPassengerRear":"false","electricParkBrakeStatus":"1","doorLockStatusDriver":"1","seatBeltStatusThDriverRear":"false","tankFlapStatus":"2","seatBeltStatusPassengerRear":"false","doorOpenStatusPassenger":"0","doorPosPassenger":"0","vehicleAlarm":{"alrmTrgSrc":"0"},"doorPosDriverRear":"0","centralLockingStatus":"2","seatBeltStatusDriver":"false","doorLockStatusPassenger":"1","seatBeltStatusMidRear":"false","trunkLockStatus":"1","seatBeltStatusDriverRear":"false","engineHoodOpenStatus":"0","doorOpenStatusDriverRear":"0","doorLockStatusPassengerRear":"1","trunkOpenStatus":"0"},"pollutionStatus":{"interiorPM25":"1","interiorSecondPM25Level":"0","interiorPM25Level":"0","relHumSts":"80","exteriorPM25Level":"0"}},"temStatus":{"swVersion":null,"serialNumber":null,"powerSource":null,"networkAccessStatus [... cut ...]
.Soc:         json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Capacity:     62.0kWh
Range:        json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Odometer:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Climater:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
Position:     json: invalid use of ,string struct tag, trying to unmarshal "" into int: %!s(<nil>)
OnIdentified: MaxCurrent:10
Features:     []

Haleo, 2024-03-13:
Bei mir exakt das selbe Problem. Login funktioniert. evcc vehicle  --log trace smart zeigt bei mir das selbe Verhalten. SOC ist nicht zu sehen.

cyberways, 2024-03-14:
Hallo Haleo und andere Smart #1 Fahrer,
kurz off topic: Folgt dein Smart Ampere-genau den Ladestromvorgaben von evcc/der Wallbox? Meiner tut es nicht. Beim Hochregeln folgt der Smart reproduzierbar nur, wenn der neue Ladestrom mindestens 3A über dem aktuellen liegt, wodurch eine Menge Strom im Netz eingespeist statt im Auto landet. Beim Herunterregeln des Ladestroms bricht regelmäßig die Wallbox die Ladung mit Störung wegen Überstrom ab, weil der Smart nicht dem niedrigeren Ladestrom folgt.
evcc Zykluszeit ist 30s. Mit unserem BMW funktioniert das perfekt.

andig, 2024-03-14:
@marco79cgn willst Du mal einen Blick drauf werfen? Vmtl. hat Smart das bescheuerte "Zahlen als Strings" Format jetzt noch kaputter gemacht, so dass es jetzt manuell dekodiert werden muss. Ich komme aktuell nicht dazu mir das anzuschauen...

Haleo, 2024-03-18:
Ich habe zur Sicherheit übers Wochenende EVCC von Grund auf neu eingerichtet und erhalte wieder den selben Fehler beim Smart. Das Widget von @marco79cgn und so läuft sauber. Die Einrichtung müsste eigentlich von dieser Seite her passen.

andig, 2024-03-18:
> Ich habe zur Sicherheit übers Wochenende EVCC von Grund auf neu eingerichtet 

Das ist nie notwendig. evcc besteht aus einer einzigen Konfigurationsdatei plus Datenbank.

andig, 2024-03-23:
Mangels Testdaten/Feedback machen wir hier zu.

closed 2024-03-23 by andig

Haleo, 2024-03-23:
Schade, wenn ich hätte helfen können gerne melden.

andig, 2024-03-23:
@Haleo dann gerne Zugangsdaten an info@evcc.io zum Testen. Danke!

Haleo, 2024-03-23:
Verflixt und zugenäht. Ich habe ein neues Smart-Konto erstellt und den digitalen Schlüssel diesem freigegeben. Über dieses neue Konto bekomme ich nun den SOC ohne Fehler. Ich habe keine Ahnung wieso. Dementsprechend wäre das mein Lösungsansatz für @cyberways. :-) Für mich ist das Thema damit erledigt. Danke für Eure tolle SW-Lösung und den super Support.

cyberways, 2024-03-25:
Danke für den Tip, das ist ja echt verrückt. Nachdem es dankenswerterweise im Code schon gefixt ist warte ich jetzt mal auf die nächste Release zum Testen und werde berichten.

@Haleo: Funktioniert das Nachregeln des Ladestroms bei deinem Smart korrekt? Siehe mein ausgeblendeter off-topic Post 10 Einträge weiter oben. Habe deshalb Gewährleistung geltend gemacht bei smart bisher ohne Ergebnis bzw. Anerkennung dessen und wüsste gerne, ob das ein Serienproblem ist. Meine Wallbox hat heute schon wieder 3x die Ladung abgebrochen mit Störung bei jeder Wolke, weil der smart nicht richtig den Ladestrom herunterregelt. Vielen Dank!

Haleo, 2024-03-26:
@cyberways Bei mir sieht das sauber aus. Er lädt aber immer 1A unter dem Maximum. Muss ich mal beobachen. Wallbox ist ein go-echarger Homefix v3 bei mir. Jetzt ist ja wieder mehr Sonne da. Melde mich bei Dir. Allerdings hat das bisher immer problemlos funktioniert, inkl. Phasenumschaltung und allem. Gibt ja verschiedene Versionen des OBC. Mein Smart wurde im April 23 gebaut und hat nicht die erste Version.

sffbigmac, 2024-08-14:
@Haleo hat sich da was ergeben? Seit ein paar Tagen sehe ich plötzlich auch das Problem mit meinem smart #1. In der hellosmart app zeigt er mir sen soc allerdings an. evcc mal auf latest geupdated bringt auch keine änderung.

sffbigmac, 2024-08-15:
@andig habe gesehen dass du den PR #13021 zugemacht hattest. Ich sehe wie oben erwähnt aktuell allerdings noch weiterhin genau das Problem. Hier mal ein Auszug aus dem trace was die smart API lieferte (zu dem Zeitpunkt war der smart allerdings nicht an der Wallbox verbunden):

```
"electricVehicleStatus":{
"disChargeUAct":"0.0",
"disChargeSts":"0",
"wptFineAlignt":"0",
"chargeLidAcStatus":"2",
"distanceToEmptyOnBatteryOnly":"199",
"distanceToEmptyOnBattery100Soc":"406",
"chargeSts":"0",
"averPowerConsumption":"-84.4",
"chargerState":"0",
"timeToTargetDisCharged":"2047",
"distanceToEmptyOnBattery20Soc":"81",
"disChargeConnectStatus":"0",
"chargeLidDcAcStatus":"2",
"dcChargeSts":"0",
"ptReady":"0",
"chargeLevel":"52",
"statusOfChargerConnection":"0",
"dcDcActvd":"0",
"indPowerConsumption":"0.0",
"dcDcConnectStatus":"0",
"disChargeIAct":"0.0",
"dcChargeIAct":"0.0",
"chargeUAct":"0.0",
"bookChargeSts":"0",
"chargeIAct":"0.000",
"timeToFullyCharged":"2047"
}
```

Egal ob ich ihn an der WB zum Laden hängen habe oder nicht, das Problem ist das gleiche. EVCC zeigt den SOC nicht mehr an und besagter Error steht im Log.