evcc-io/evcc #12887: MQTT topic evcc/loadpoints/id/climaterActive nicht mehr vorhanden

mdkeil, 2024-03-12:
### Describe the bug

Bis vor kurzem wurde das topic noch gepublished-- kann aber nicht genau nachvollziehen, wann es "entfernt" wurde.. unter `/api/state` gibt es unter den loadpoints `vehicleClimaterActive`. 

Habe via MQTT Explorer geschaut, ob nur das topic geändert wurde, konnte aber nichts finden. Vielleicht wurde das topic ja bewusst entfernt.

### Steps to reproduce

Nothing to report (NTR)

### Configuration details

```yaml
NTR
```

### Log details

```text
NTR
```

### What type of operating system are you running?

Docker container

### Version

0.124.9

docolli, 2024-03-12:
Das wird unter 0.124.9 bei einem Neustart von evcc schon noch gepublished. Eben getestet. Und dann vermutlich nur noch, wenn sich der Wert ändert.
Lass mal dem MQTT Explorer laufen und starte evcc neu.

mdkeil, 2024-03-12:
Das kann ich bestätigen.. witzerweise wird das topic auch alle 20s aktuallisiert.. einziger Unterschied, dass nun ein Auto angeschlossen ist.. nun fehlt aber das topic` evcc/loadpoints/id/planActive` ..irgendwie auf jeden Fall eigenartig das Verhalten.

VolkerK62, 2024-03-12:
> nun fehlt aber das topic evcc/loadpoints/id/planActive

Das ist nach einem Neustart weg und kommt wieder, wenn du mal kurz einen Plan aktivierst.

docolli, 2024-03-12:
>Das kann ich bestätigen.. witzerweise wird das topic nun auch alle 20s aktuallisiert.. einziger Unterschied, dass nun ein Auto angeschlossen ist..

Das ist ganz normal. Nur wenn ein Auto angeschlossen ist, wird auch der Zustand des Autos über das Hersteller-Portal abgefragt. Nur dann hat evcc Infos zum Zustand `vehicleClimaterActive`.

mdkeil, 2024-03-12:
@VolkerK62 ..auch das konnte ich nachvollziehen.. dieses Verhalten ist aber definitiv mal anders gewesen.. habe es halt nur mitbekommen, dass nach einem node-red Neustart einige Werte auf "" standen, wo ich vorher definierte booleans hatte..

VolkerK62, 2024-03-13:
so ging es mir auch ... ich bin bei @docolli , dass es mit #12621  zu tun hat

mdkeil, 2024-03-13:
Das glaube ich persönlich nicht.. dieses Verhalten muss wohl schon länger so sein, nur dass es mir nie aufgefallen war.. ich bin gestern mal bis 0.123.0 zurückgegangen, da war das Verhalten identisch.. Wenn man es weiß, kann man sich ja leicht drauf einstellen, nur des es schwieriger ist, entsprechende topics ohne aktualisierte Doku zu finden, wenn einige Werte erst bei definierten Ereignissen gepublished werden, wobei man dann ja immer noch auf die API ausweichen kann.

docolli, 2024-03-13:
Bis zu diesem Fix hat evcc alle MQTT Topics wohl mit dem Flag "retained" veröffentlicht, sodaß sich der Broker die Topics (und deren letzten Werte) gemerkt hat. Siehe https://www.hivemq.com/blog/mqtt-essentials-part-8-retained-messages/
Vorteil ist, dass der Broker auch noch Werte liefert (aber nicht mehr aktuelle) und die angefragten Topics kennt, wenn evcc nicht mehr aktiv ist.

Seit dem Fix sendet evcc beim Start einen "cleanup"-Befehl `err := m.Handler.Cleanup(m.root+"/#", true)`, mit dem vermutlich ALLE evcc MQTT Topics aus dem Broker gelöscht werden. Dann sendet evcc wieder alle aktuellen MQTT Topics und der Broker hat dann nur noch die aktuellen evcc Topics.
Der Grund ist, dass es einige Umbauten in den MQTT Topics gab (und noch geben wird), ohne das Cleanup verbleiben die im Broker, werden aber nicht mehr aktualisiert. Da der MQTT Topic Umbau schrittweise voran geht, ist nicht klar definiert, wer noch welche Topics im Broker "retained" liegen hat. Da ist ein generelles Löschen beim Start die sicherste Methode aufzuräumen, aber eben mit dem kleinen Nachteil, dass es bei jedem Neustart von evcc etwas dauert, bis der Broker wieder alle Topics aufgesammelt hat. evcc sendet die nur, wenn intern dazu auch Daten vorliegen.

closed 2024-03-13 by andig