evcc-io/evcc #12895: Fahrzeug Lädt direkt mit voller Leistung Porsche Taycan + Porsche Charger Connect

hegi2k, 2024-03-13:
### Describe the bug

Laden startet direkt mit voller leistung beim Einstecken obwohl Anzeige auf Solar steht. Selbst auf Off zu stellen stoppt den Ladevorgang nicht. Muss das Fahrzeug wieder ausstecken
Einen tag vorher hat alles normal funktioniert.
Soc vom Fahrzeug 40 Prozent

### Steps to reproduce

1.
2.
3.
...
Fahrzeug anstecken

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 4087cd7ef3b1a821efefe3b57dc14909bba3021dd7f11f669f96e9524633d54e

interval: 30s # control cycle interval

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.188.43
  name: grid1
- type: template
  template: sma-inverter-speedwire
  usage: pv
  host: 192.168.188.44
  password: 
  name: pv2

chargers:
- type: template
  template: pmcc
  ski: 4a4c-0fa0-288c-fe99-e7c6-1b4a-0e92-47b5-e304-d34d
  ip: 192.168.188.92
  name: wallbox3

loadpoints:
- title: Garage9
  charger: wallbox3
  mode: pv

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          MHcCAQEEIFxO4LBig6/rjCNwj6arKRGUsu3a1a8pY4PcgWuv6eHIoAoGCCqGSM49
          AwEHoUQDQgAE5IbTuB4ZGjN6GxI5CMbpiU5yVXH1lzcbZbxGk1j3KPu17NmuG/E7
          t7KV0r91QATfUt2DxoSn5L1/KfFk5NzYLQ==
[... 95 lines cut ...]
[eebus ] DEBUG 2024/03/13 08:02:26 Send: d:_i:19667_PorscheEVSE-00126095 notify 30 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:27 Send: d:_i:19667_PorscheEVSE-00126095 notify 31 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:27 Send: d:_i:19667_PorscheEVSE-00126095 notify 32 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:28 avahi - error resolving service: Timeout reached
[eebus ] DEBUG 2024/03/13 08:02:28 Send: d:_i:19667_PorscheEVSE-00126095 notify 33 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:29 Send: d:_i:19667_PorscheEVSE-00126095 notify 34 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:30 Send: d:_i:19667_PorscheEVSE-00126095 notify 35 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:31 Send: d:_i:19667_PorscheEVSE-00126095 notify 36 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:31 Send: d:_i:19667_PorscheEVSE-00126095 notify 37 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:32 Send: d:_i:19667_PorscheEVSE-00126095 notify 38 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:33 Send: d:_i:19667_PorscheEVSE-00126095 notify 39 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:34 Send: d:_i:19667_PorscheEVSE-00126095 notify 40 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/03/13 08:02:35 Send: d:_i:19667_PorscheEVSE-00126095 notify 41 DeviceDiagnosisHeartbeatData
^C[eebus ] DEBUG 2024/03/13 08:02:35 mdns: stop announcement
[eebus ] DEBUG 2024/03/13 08:02:35 mdns: stop announcement
[eebus ] DEBUG 2024/03/13 08:02:35 restarting mdns search
```

### What type of operating system are you running?

Linux

### Version

 evcc version 0.124.7

DerAndereAndi, 2024-03-13:
Das Auto ladet IMMER mit voller Leistung bis evcc im Intervall die Daten bekommt und einschreiten kann. Dein Interval ist auf 30s, wenn du es 1s nach dem Intervall einsteckst, dauert es 29s bis geregelt wird. Stell das Interval auf z.b. 10s wenn deine Meter das unterstützen. Dann hast du den Effekt nur für max 10s.

Das ist normales Verhalten so wie evcc arbeitet. Bei EEBUS kann evcc ohne angestecktes Auto die Ladung nicht unterbrechen und damit nur im Intervall einschreiten.

hegi2k, 2024-03-13:
Danke für die Antwort.
Also am Intervall liegt es nicht, da er eine Stunde geladen hat. Komischerweise hat es ja mehrere Tage korrekt funktioniert.
Habe jetzt Direktladen ausgemacht, dann scheint es korrekt zu funktionieren.
Muss ich das, wenn ich an einem HPC Charger lade, wieder einschalten?

Würde es nicht Sinn machen diese Informationen in der Doku zu ergänzen?