evcc-io/evcc #12901: Error connecting Wallbox Copper Business Charger

ddronald, 2024-03-13:
### Describe the bug

Im trying to connect my Wallbox Copper Business Charger via ocpp. The connection is established but i recieve an error:

Value must be maximum 500, but was 551 for feature GetConfiguration 

Config of my wallboxx ocpp settings:
URL: ws://xx.xx.xx.123:8887 (also tried with / after port number)
charge point id: 3300916

### Steps to reproduce

1. First tried with template pulsarplus, but that is not suitable for the Wallbox Copper Business
2. then tried the ocpp template, that gives this error
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: xx.xx.xx.123
  port: 7070

interval: 30s 

meters:
  - name: grid
    type: template
    template: slimmelezer
    usage: grid
    host: xx.xx.xx.120
  - name: pv
    type: template
    template: sma-data-manager
    usage: pv
    modbus: tcpip
    id: 3
    host: xx.xx.xx.122 # Hostname

chargers:                 
  - name: charger                        
    type: template                       
    template: ocpp                       
    timeout: 2m # optional             
                                       
site:                                                     
  - title: Thuis
    meters:                                                                                               
      grid: grid 
      pv: 
        - pv 
      residualPower: 0                                                                                  
      maxGridSupplyWhileBatteryCharging: 0

loadpoints:                                                                                               
  - title: Oprit
    charger: charger
    mode: "off"
```

### Log details

```text
[main  ] INFO 2024/03/13 12:09:55 evcc 0.124.9
[main  ] INFO 2024/03/13 12:09:55 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/13 12:09:55 starting ui and api at :7070
[db    ] INFO 2024/03/13 12:09:55 using sqlite database: /root/.evcc/evcc.db
[ocpp  ] INFO 2024/03/13 12:11:24 charge point connected, registering: 3300916
[ocpp  ] ERROR 2024/03/13 12:11:25 ocpp message (2286910082): PropertyConstraintViolation - Field CallResult.Payload.ConfigurationKey[16].Value must be maximum 500, but was 551 for feature GetConfiguration ([3,"2286910082",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "AllowOfflineTxForUnknownId","value": "true"},{"readonly": false,"key": "AuthorizationCacheEnabled","value": "false"},{"readonly": false,"key": "AuthorizeRemoteTxRequests","value": "true"},{"readonly": false,"key": "BlinkRepeat","value": "10"},{"readonly": false,"key": "ClockAlignedDataInterval","value": "900"},{"readonly": false,"key": "ConnectionTimeOut","value": "120"},{"readonly": false,"key": "ConnectorPhaseRotation","value": "1.Unknown"},{"readonly": true,"key": "ConnectorPhaseRotationMaxLength","value": "1"},{"readonly": true,"key": "GetConfigurationMaxKeys","value": "50"},{"readonly": false,"key": "HeartbeatInterval","value": "60"},{"readonly": false,"key": "LightIntensity","value": "100"},{"readonly": false,"key": "LocalAuthorizeOffline","value": "true"},{"readonly": false,"key": "LocalPreAuthorize","value": "false"},{"readonly": false,"key": "MaxEnergyOnInvalidId","value": "0"},{"readonly": false,"key": "MeterValuesAlignedData","value": "Energy.Active.Import.Register"},{"readonly": true,"key": "MeterValuesAlignedDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValuesSampledData","value": "Current.Export.L1,Current.Export.L2,Current.Export.L3,Current.Import.L1,Current.Import.L2,Current.Import.L3,Current.Offered,Energy.Active.Export.Interval,Energy.Active.Export.Register,Energy.Active.Import.Interval,Energy.Active.Import.Register,Energy.Reactive.Export.Interval,Energy.Reactive.Export.Register,Energy.Reactive.Import.Interval,Energy.Reactive.Import.Register,Frequency,Power.Active.Export,Power.Active.Import,Power.Factor,Power.Offered,Power.Reactive.Export,Power.Reactive.Import,RPM,SoC,Temperature,Voltage.L1-N,Voltage.L2-N,Voltage.L3-N"},{"readonly": true,"key": "MeterValuesSampledDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValueSampleInterval","value": "60"},{"readonly": false,"key": "MinimumStatusDuration","value": "1"},{"readonly": true,"key": "NumberOfConnectors","value": "1"},{"readonly": false,"key": "ResetRetries","value": "0"},{"readonly": false,"key": "StopTransactionOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "StopTransactionOnInvalidId","value": "true"},{"readonly": false,"key": "StopTxnAlignedData","value": ""},{"readonly": true,"key": "StopTxnAlignedDataMaxLength","value": "0"},{"readonly": false,"key": "StopTxnSampledData","value": ""},{"readonly": true,"key": "StopTxnSampledDataMaxLength","value": "0"},{"readonly": true,"key": "SupportedFileTransferProtocols","value": ""},{"readonly": true,"key": "SupportedFeatureProfiles","value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"},{"readonly": true,"key": "SupportedFeatureProfilesMaxLength","value": "6"},{"readonly": false,"key": "TransactionMessageAttempts","value": "3"},{"readonly": false,"key": "TransactionMessageRetryInterval","value": "60"},{"readonly": false,"key": "UnlockConnectorOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "WebSocketPingInterval","value": "60"},{"readonly": false,"key": "LocalAuthListEnabled","value": "true"},{"readonly": true,"key": "LocalAuthListMaxLength","value": "5000000"},{"readonly": true,"key": "SendLocalListMaxLength","value": "450"},{"readonly": true,"key": "ReserveConnectorZeroSupported","value": "false"},{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "1"},{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"},{"readonly": true,"key": "ChargingScheduleMaxPeriods","value": "100"},{"readonly": true,"key": "ConnectorSwitch3to1PhaseSupported","value": "false"},{"readonly": true,"key": "MaxChargingProfilesInstalled","value": "1"},{"readonly": false,"key": "WebSocketUrl","value": "ws://xx.xx.xx.123:8887/3300916"},{"readonly": false,"key": "chargingALimitConn1","value": "12"},{"readonly": false,"key": "AuthEnabled","value": "true"},{"readonly": false,"key": "AuthEnabledOffline","value": "true"},{"readonly": false,"key": "AuthDisabledIdTag","value": "NoAuthorization"},{"readonly": false,"key": "minSoC","value": "9"},{"readonly": false,"key": "maxSoC","value": "97"},{"readonly": true,"key": "SSID","value": ""},{"readonly": true,"key": "WiFiSignalStrength","value": "0"},{"readonly": true,"key": "SerialNumber","value": "517092"},{"readonly": true,"key": "PUK","value": "36758811"},{"readonly": true,"key": "WebSocketUrl","value": "ws://xx.xx.xx.123:8887"},{"readonly": true,"key": "WebSocketUser","value": "3300916"},{"readonly": true,"key": "WebSocketPassword","value": ""} ]}])
```

### What type of operating system are you running?

Docker container

### Version

0.124.9

andig, 2024-03-13:
Depends on https://github.com/lorenzodonini/ocpp-go/pull/258

andig, 2024-03-14:
Until upstream is fixed you'll need to use 

    getconfiguration: false

Seems the charger is violating the spec...

closed 2024-03-14 by andig