evcc-io/evcc #12903: UI: Make loadpoints configurable

andig, 2024-03-13:
Loadpoints are still setup in yaml in first step, but become partly configurable in config ui:

- basic lp settings move to config ui
- main ui still contains 1p3p (for supporting chargers) + smartcost (depending on tariff)
- similarly, vehicle settings move to config ui

basic settings are:

	Title_          string `mapstructure:"title"`
	ConfiguredPhases_ int           `mapstructure:"phases"`
	MinCurrent_       float64       `mapstructure:"minCurrent"`
	MaxCurrent_       float64       `mapstructure:"maxCurrent"`

step 2:

	Priority_       int    `mapstructure:"priority"`
	Soc             SocConfig
	Enable, Disable ThresholdConfig

Matthias3678, 2024-04-08:
How can I undo settings that I have made via the UI?

My car can only charge on 2-phase. I have an Elli without a measuring device. Now it counts incorrectly and charging only starts late.

[image]

naltatis, 2024-04-09:
This is off-topic here. But the solution is to specify the phases at your vehicle: https://docs.evcc.io/docs/reference/configuration/vehicles/#phases

closed 2025-01-31 by naltatis