evcc-io/evcc #12904: Wärmepumpe Ladestop täglich simulieren

motze92, 2024-03-13:
**Is your feature request related to a problem? Please describe.**
Ich hab meine IDM Wärmepumpe per Modbus integriert. Die WP macht die Steuerung aber autonom und EVCC teilt ihr nur mit, wie viel Überschuss vorhanden ist und lese die aktuelle Leistung der WP aus. Daher ist der "Ladevorgang" ewig aktiv. Ich würde aber gerne sehen wie viel meine WP Täglich verbraucht hat.

**Describe the solution you'd like**
Option um den "Ladevorgang" täglich um 00:00 stoppen und neu starten.

VolkerK62, 2024-03-13:
Weiß nicht ob das in deinem Fall auch funktioniert. 
Ich setzte bei meinen Shelly um Mitternacht den Modus auf "aus" und dann wieder auch "schnell". Dadurch wird die Ladesession beendet und ich habe für jeden Tag ein Ladelog.
Das funktioniert aber nur, weil das Schalten auf "aus" gleichbedeutend mit "Fahrzeug disconnect" ist.

naltatis, 2024-03-13:
Ja, der aktuelle Workaround ist das manuelle Ein-/Ausschalten des Ladepunkts. Das funktioniert bei Switchsockets. Mittelfristig wollen wir hier aber eine zeitbasierte Auswertung pro Ladepunkt anbieten. Die Darstellung über die Ladevorgänge ist ja für Nicht-Fahrzeuge eher suboptimal.

closed 2024-03-13 by naltatis