evcc-io/evcc #12915: Tariffs: ocotpusenergy - 30 Minutes interval - U.I using initial hourly cost / no averaging -

IamMattM, 2024-03-13:
I would like to clarify how a smart grid tariff like the one from` octopusenergy` `AGILE-23-12-06` , which changes every 30minutes,  is handled by EVCC ?

I can see that the U.I only presents the cost of the Kwh in a full hourly slot, with the cost for this hour slot based on the cost at the beginning of the hour only (no averaging with the second half of the hour).

Is evcc correctly handling tariffs that change 30mins ?

`evcc tariff` API request command , for `AGILE-23-12-06`correcly shows those half-hourly costs changing but I am not certain as to whether evcc works in the backgground "honouring" those intra-hourly varying costs ?

https://imgur.com/a/yxKRL3v

naltatis, 2024-03-13:
The UI currently assumes that prices only change every hour. Cheap grid charging, cost calculations and afaik also the planning algorithm are not limited to fixed time intervals.

What would your expectation for the UI be? Half hour slots? Avg. values? Is it always 30m intervals at octopus?

IamMattM, 2024-03-13:
@naltatis 

Thanks for confirming that the grid charging + cost calculations + planning algorithm are not bound by a fixed 1 hour time interval. That's great news.

With regards to Octopus energy, I can confirm that at least in the U.K. , the shortest tariff interval is indeed a minimum of 30mins.

I have come across this post, that implies other energy supplier smart tariffs might be also worked on a very short 15minutes intervals:
From : [](https://github.com/evcc-io/evcc/issues/12698) , "VARIO_PLUS -> this is the tariff who can change every 15 minutes."

In terms of U.I. , I would imagine that if possible, it would be fantastic to actually derive the interval duration from the API request (So if there are any records for change of tariff at time  `xx:xx:15` or `xx:xx:30` beyond the `xx:xx:00` , it is then known that the intervals to be displayed in the U.I would want to match...?

For example , from octopus I get two days of 30minutes intervals:
```
2024-03-12 21:00:00    2024-03-12 21:30:00    0.138
2024-03-12 21:30:00    2024-03-12 22:00:00    0.125
2024-03-12 22:00:00    2024-03-12 22:30:00    0.138
2024-03-12 22:30:00    2024-03-12 23:00:00    0.119
2024-03-12 23:00:00    2024-03-12 23:30:00    0.131
2024-03-12 23:30:00    2024-03-13 00:00:00    0.101
2024-03-13 00:00:00    2024-03-13 00:30:00    0.101
2024-03-13 00:30:00    2024-03-13 01:00:00    0.115
2024-03-13 01:00:00    2024-03-13 01:30:00    0.106
2024-03-13 01:30:00    2024-03-13 02:00:00    0.084
2024-03-13 02:00:00    2024-03-13 02:30:00    0.092
2024-03-13 02:30:00    2024-03-13 03:00:00    0.088
2024-03-13 03:00:00    2024-03-13 03:30:00    0.091
2024-03-13 03:30:00    2024-03-13 04:00:00    0.086
2024-03-13 04:00:00    2024-03-13 04:30:00    0.108
2024-03-13 04:30:00    2024-03-13 05:00:00    0.108
2024-03-13 05:00:00    2024-03-13 05:30:00    0.123
2024-03-13 05:30:00    2024-03-13 06:00:00    0.139
2024-03-13 06:00:00    2024-03-13 06:30:00    0.154
2024-03-13 06:30:00    2024-03-13 07:00:00    0.178
2024-03-13 07:00:00    2024-03-13 07:30:00    0.151
2024-03-13 07:30:00    2024-03-13 08:00:00    0.179
2024-03-13 08:00:00    2024-03-13 08:30:00    0.154
2024-03-13 08:30:00    2024-03-13 09:00:00    0.155
2024-03-13 09:00:00    2024-03-13 09:30:00    0.146
2024-03-13 09:30:00    2024-03-13 10:00:00    0.148
2024-03-13 10:00:00    2024-03-13 10:30:00    0.146
2024-03-13 10:30:00    2024-03-13 11:00:00    0.143
2024-03-13 11:00:00    2024-03-13 11:30:00    0.137
2024-03-13 11:30:00    2024-03-13 12:00:00    0.132
2024-03-13 12:00:00    2024-03-13 12:30:00    0.126
2024-03-13 12:30:00    2024-03-13 13:00:00    0.130
2024-03-13 13:00:00    2024-03-13 13:30:00    0.132
2024-03-13 13:30:00    2024-03-13 14:00:00    0.127
2024-03-13 14:00:00    2024-03-13 14:30:00    0.127
2024-03-13 14:30:00    2024-03-13 15:00:00    0.125
2024-03-13 15:00:00    2024-03-13 15:30:00    0.115
2024-03-13 15:30:00    2024-03-13 16:00:00    0.124
2024-03-13 16:00:00    2024-03-13 16:30:00    0.272
2024-03-13 16:30:00    2024-03-13 17:00:00    0.271
2024-03-13 17:00:00    2024-03-13 17:30:00    0.272
2024-03-13 17:30:00    2024-03-13 18:00:00    0.292
2024-03-13 18:00:00    2024-03-13 18:30:00    0.312
2024-03-13 18:30:00    2024-03-13 19:00:00    0.316
2024-03-13 19:00:00    2024-03-13 19:30:00    0.170
2024-03-13 19:30:00    2024-03-13 20:00:00    0.151
2024-03-13 20:00:00    2024-03-13 20:30:00    0.149
2024-03-13 20:30:00    2024-03-13 21:00:00    0.128
2024-03-13 21:00:00    2024-03-13 21:30:00    0.146
2024-03-13 21:30:00    2024-03-13 22:00:00    0.134
2024-03-13 22:00:00    2024-03-13 22:30:00    0.121
2024-03-13 22:30:00    2024-03-13 23:00:00    0.109
2024-03-13 23:00:00    2024-03-13 23:30:00    0.131
2024-03-13 23:30:00    2024-03-14 00:00:00    0.115
2024-03-14 00:00:00    2024-03-14 00:30:00    0.102
2024-03-14 00:30:00    2024-03-14 01:00:00    0.122
2024-03-14 01:00:00    2024-03-14 01:30:00    0.121
[... 19 lines cut ...]
2024-03-14 11:00:00    2024-03-14 11:30:00    0.143
2024-03-14 11:30:00    2024-03-14 12:00:00    0.134
2024-03-14 12:00:00    2024-03-14 12:30:00    0.139
2024-03-14 12:30:00    2024-03-14 13:00:00    0.136
2024-03-14 13:00:00    2024-03-14 13:30:00    0.126
2024-03-14 13:30:00    2024-03-14 14:00:00    0.122
2024-03-14 14:00:00    2024-03-14 14:30:00    0.119
2024-03-14 14:30:00    2024-03-14 15:00:00    0.119
2024-03-14 15:00:00    2024-03-14 15:30:00    0.116
2024-03-14 15:30:00    2024-03-14 16:00:00    0.122
2024-03-14 16:00:00    2024-03-14 16:30:00    0.258
2024-03-14 16:30:00    2024-03-14 17:00:00    0.280
2024-03-14 17:00:00    2024-03-14 17:30:00    0.285
2024-03-14 17:30:00    2024-03-14 18:00:00    0.302
2024-03-14 18:00:00    2024-03-14 18:30:00    0.298
2024-03-14 18:30:00    2024-03-14 19:00:00    0.298
2024-03-14 19:00:00    2024-03-14 19:30:00    0.154
2024-03-14 19:30:00    2024-03-14 20:00:00    0.143
2024-03-14 20:00:00    2024-03-14 20:30:00    0.138
2024-03-14 20:30:00    2024-03-14 21:00:00    0.126
2024-03-14 21:00:00    2024-03-14 21:30:00    0.141
2024-03-14 21:30:00    2024-03-14 22:00:00    0.130
2024-03-14 22:00:00    2024-03-14 22:30:00    0.116
2024-03-14 22:30:00    2024-03-14 23:00:00    0.060
```

naltatis, 2024-03-13:
> In terms of U.I. , I would imagine that if possible, it would be fantastic to actually derive the interval duration from the API request

Yes, this sounds like a good compromise where users with hourly prices (every tariff in Germany I know of) don't get unnecessary detail and users of finer tariffs (30m, 15m) get more bars depending on the API response. 

This is not a trivial change. But since the underlying logic should work as expected, right now, I'd see this as a lower priority feature. But we should definitely do this.

naltatis, 2024-03-14:
> Planner accepts full hours only, too.

🙄

premultiply, 2024-03-14:
Perspektivisch müssten wir wohl auf die 15 Minuten Basistakt des Strommarkts kommen.

joriws, 2025-03-12:
**Nordpool goes to 15min periods in June 2025.**

ref. [(https://www.nordpoolgroup.com/en/trading/transition-to-15-minute-market-time-unit-mtu/)]

tantive, 2025-04-26:
Tibber will also move to 15 minute slots
[https://tibber.com/de/magazine/inside-tibber/viertelstuendliche-strompreise](https://tibber.com/de/magazine/inside-tibber/viertelstuendliche-strompreise)

RTTTC, 2025-09-17:
So, are we ready for 15 minute time slots ?
If I understand correctly, EU moves from 1h to 15 minute intervals on 2025-09-30