evcc-io/evcc #12921: Victron battery: active battery control sets minSOC after charging

Robman1, 2024-03-13:
### Describe the bug

The acitve battery control works fine during the charge process. However after the planned charge process I would like to have the minSOC setting as it was set before the process. Actually I realize minSOC is set to a value close to the actual SOC when charging begins. I already use "minsoc: -1" (which is by the way not documented in the victron battery device manual on evcc.io) and thought this would set minsoc to the value as it was set before controlled charging.

My yaml entry for the victron battery device

  - name: battery 
    type: template
    template: victron-energy 
    usage: battery  
    host: 192.168.0.149  
    port: 502  
    capacity: 6.5
    minsoc: -1

### Steps to reproduce

1.set planned charging
2.let do the planned charging
3.after charging the minsoc is set to the value the battery had before the charging process (or close to)
...

### Configuration details

```yaml
network:
  schema: http
  port: 7070
  host: docker

log: debug

interval: 3s

telemetry: true

sponsortoken: ###

meters:
  - type: template
    template: victron-energy 
    usage: grid  
    host: 192.168.0.149  
    port: 502  
    name: grid
    
  - name: pv
    type: template
    template: sma-inverter
    usage: pv
    host: 192.168.0.150 # IP-Adresse oder Hostname
    password: ###

  - name: battery 
    type: template
    template: victron-energy 
    usage: battery  
    host: 192.168.0.149  
    port: 502  
    capacity: 6.5
    minsoc: -1

chargers:
  - type: template
    template: go-e-v3 
    host: 192.168.0.162  
    name: Walli

loadpoints:
  - title: Garage
    charger: Walli
    mode: pv
    phases: 1
    mincurrent: 6
    maxcurrent: 32
    #resetOnDisconnect: true
    vehicle: Dacia 

site:
  title: Zuhause
  meters:
    grid: grid
    pv: pv
    battery: battery

vehicles:
  - name: Dacia
    type: template
    template: dacia
    title: Dacia
    user: ###
    password: ###
    capacity: 26.8
    icon: car
    phases: 1
    mode: pv
    mincurrent: 6
    maxcurrent: 32
    #targetSoc: 90
   
   

  - name: vehicle_5
    type: template
    template: offline
    title: Wärmepumpe
    icon: waterheater

tariffs:
  currency: EUR # (default EUR)
  grid:
    # dynamic grid price
    type: tibber
    token: ###

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.1288 # [currency]/kWh

  co2:
    type: grünstromindex
    zip: "###"

# mqtt message broker
mqtt:
  broker: 192.168.0.163:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: ###
  password: ###
```

### Log details

```text
/app # evcc --log debug
[main  ] INFO 2024/03/13 21:49:12 evcc 0.124.9
[main  ] INFO 2024/03/13 21:49:12 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/13 21:49:12 starting ui and api at :7070
[db    ] INFO 2024/03/13 21:49:12 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/03/13 21:49:12 connecting evcc at tcp://192.168.0.163:1883
[mqtt  ] DEBUG 2024/03/13 21:49:12 tcp://192.168.0.163:1883 connected
[mqtt  ] ERROR 2024/03/13 21:49:12 tcp://192.168.0.163:1883 connection lost: read tcp 192.168.0.163:49678->192.168.0.163:1883: read: connection reset by peer
[mqtt  ] DEBUG 2024/03/13 21:49:12 tcp://192.168.0.163:1883 connected
[mqtt  ] ERROR 2024/03/13 21:49:13 tcp://192.168.0.163:1883 connection lost: EOF
[lp-1  ] WARN 2024/03/13 21:49:14 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/13 21:49:14 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/03/13 21:49:14 deprecated: phases setting is ignored, please remove
[mqtt  ] DEBUG 2024/03/13 21:49:14 tcp://192.168.0.163:1883 connected
[lp-1  ] DEBUG 2024/03/13 21:49:14 charge total import: 624.397kWh
[site  ] WARN 2024/03/13 21:49:14 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/03/13 21:49:16 vehicle 'Wärmepumpe' does not support automatic detection
[site  ] INFO 2024/03/13 21:49:16 site config:
[site  ] INFO 2024/03/13 21:49:16   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/13 21:49:16     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/13 21:49:16     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/03/13 21:49:16     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/03/13 21:49:16   vehicles:
[site  ] INFO 2024/03/13 21:49:16     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/03/13 21:49:16     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/03/13 21:49:16 loadpoint 1:
[lp-1  ] INFO 2024/03/13 21:49:16   mode:        pv
[lp-1  ] INFO 2024/03/13 21:49:16   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/03/13 21:49:16   meters:      charge ✓
[lp-1  ] INFO 2024/03/13 21:49:16     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/03/13 21:49:16 set buffer soc: 100
[site  ] DEBUG 2024/03/13 21:49:16 set priority soc: 100
[site  ] DEBUG 2024/03/13 21:49:16 set battery discharge control: true
[lp-1  ] DEBUG 2024/03/13 21:49:16 set smart cost limit: 0.15
[lp-1  ] DEBUG 2024/03/13 21:49:16 phase timer inactive
[lp-1  ] DEBUG 2024/03/13 21:49:16 pv timer inactive
[lp-1  ] INFO 2024/03/13 21:49:16 vehicle updated: unknown -> Dacia
[lp-1  ] DEBUG 2024/03/13 21:49:16 set charge mode: pv
[site  ] WARN 2024/03/13 21:49:16 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[main  ] FATAL 2024/03/13 21:49:16 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
/app #
```

### What type of operating system are you running?

Docker container

### Version

0.124.9

andig, 2024-03-13:
> would like to have the minSOC setting as it was set before the process

Robman1, 2024-03-13:
OK - thank you for the quick and clear answer