evcc-io/evcc #12925: read failed: modbus: response data size '2' does not match request quantity '2'

thomas-hutterer-tik, 2024-03-14:
### Describe the bug

data reading from modbus fails with unclear error message response 2 and requested 2 are obviously the same

### Steps to reproduce

1. install configuration
2. start evcc
3. watch the log
...

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "5s"
sponsortoken: "..."
log: "info"
meters:
- name: "THT-StW SmartPi "
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
  currents:
  - source: "mqtt"
    topic: "wa-pi/grid/p1"
  - source: "mqtt"
    topic: "wa-pi/grid/p2"
  - source: "mqtt"
    topic: "wa-pi/grid/p3"
- name: "Fronius Symo"
  type: "template"
  template: "fronius-solarapi-v1"
  usage: "pv"
  host: "192.168.1.8"
- name: "Sofar"
  type: "template"
  template: "sofarsolar-g3"
  usage: "pv"
  host: "192.168.1.4"
  port: 8899
  id: 1
  modbus: "tcpip"
- name: "Sofar Speicher"
  type: "template"
  template: "sofarsolar-g3"
  usage: "battery"
  host: "192.168.1.4"
  port: 8899
  id: 1
  modbus: "tcpip"
chargers:
- name: "Keba-1"
  type: "template"
  template: "keba"
  host: "192.168.1.17"
  id: 1
site:
  title: "St Wolfgang"
  meters:
    grid: "THT-StW SmartPi "
    pv:
    - "Fronius Symo"
    - "Sofar"
    battery:
    - "Sofar Speicher"
  residualPower: 100
loadpoints:
- title: "Keba-1"
  charger: "Keba-1"
  phases: 1
  mode: "minpv"
mqtt:
  broker: "127.0.0.1"
  topic: "evcc"
  user: "..."
  password: "..."
```

### Log details

```text
pi@raspberrypi:~ $ sudo systemctl restart evcc
pi@raspberrypi:~ $ journalctl -fu evcc
-- Journal begins at Mon 2023-08-14 06:59:00 CEST. --
Mar 14 09:12:22 raspberrypi evcc[18734]: [main  ] INFO 2024/03/14 09:12:22 using config file: /etc/evcc.yaml
Mar 14 09:12:22 raspberrypi evcc[18734]: [main  ] INFO 2024/03/14 09:12:22 starting ui and api at :7070
Mar 14 09:12:22 raspberrypi evcc[18734]: [db    ] INFO 2024/03/14 09:12:22 using sqlite database: /var/lib/evcc/evcc.db
Mar 14 09:12:22 raspberrypi evcc[18734]: [mqtt  ] INFO 2024/03/14 09:12:22 connecting evcc-614296402 at tcp://127.0.0.1:1883
Mar 14 09:12:22 raspberrypi evcc[18734]: [mqtt  ] DEBUG 2024/03/14 09:12:22 tcp://127.0.0.1:1883 connected
Mar 14 09:12:22 raspberrypi evcc[18734]: [main  ] WARN 2024/03/14 09:12:22 create meter 1: name must not contain special characters or spaces: THT-StW SmartPi
Mar 14 09:12:22 raspberrypi evcc[18734]: [main  ] WARN 2024/03/14 09:12:22 create meter 2: name must not contain special characters or spaces: Fronius Symo
Mar 14 09:12:22 raspberrypi evcc[18734]: [main  ] WARN 2024/03/14 09:12:22 create meter 4: name must not contain special characters or spaces: Sofar Speicher
Mar 14 09:12:22 raspberrypi evcc[18734]: [lp-1  ] WARN 2024/03/14 09:12:22 deprecated: phases setting is ignored, please remove
Mar 14 09:12:22 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:22 charge total import: 2638.216kWh
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23 site config:
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23   meters:      grid ✓ pv ✓ battery ✓
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23     grid:      power ✓ energy ✗ currents ✓
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23     pv 1:      power ✓ energy ✗ currents ✗
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23     pv 2:      power ✓ energy ✓ currents ✗
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] INFO 2024/03/14 09:12:23     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:23 loadpoint 1:
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:23   mode:        minpv
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:23   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:23   meters:      charge ✓
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:23     charge:    power ✓ energy ✓ currents ✓
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:23 phase timer inactive
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:23 pv timer inactive
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] WARN 2024/03/14 09:12:23 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
Mar 14 09:12:23 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:23 ----
Mar 14 09:12:23 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:23 charge power: 0W
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 pv power: 3179W
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 battery soc: 43%
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 battery power: 90W
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 grid meter: -1W
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 grid currents: [-1.4 -4.45 5.87]A
Mar 14 09:12:24 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:24 site power: 189W
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:24 charge currents: [0 0 0]A
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:24 charge total import: 2638.216kWh
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:24 charger status: A
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] INFO 2024/03/14 09:12:24 car disconnected
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:24 set session soc limit: 0
Mar 14 09:12:24 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:24 set session energy limit: 0
Mar 14 09:12:28 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:28 ----
Mar 14 09:12:28 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:28 charge power: 0W
Mar 14 09:12:29 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:29 pv power: 3177W
Mar 14 09:12:30 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:30 battery soc: 43%
Mar 14 09:12:30 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:30 battery power: -250W
Mar 14 09:12:30 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:30 grid meter: -780W
Mar 14 09:12:30 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:30 grid currents: [-4.47 -4.59 5.73]A
Mar 14 09:12:30 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:30 site power: -930W
Mar 14 09:12:30 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:30 charge currents: [0 0 0]A
Mar 14 09:12:30 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:30 charge total import: 2638.216kWh
Mar 14 09:12:30 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:30 charger status: A
Mar 14 09:12:33 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:33 ----
Mar 14 09:12:33 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:33 charge power: 0W
Mar 14 09:12:35 raspberrypi evcc[18734]: [site  ] ERROR 2024/03/14 09:12:35 pv 2 energy: read failed: modbus: response data size '2' does not match request quantity '2'
Mar 14 09:12:35 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:35 pv power: 501178W
Mar 14 09:12:36 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:36 battery soc: 43%
Mar 14 09:12:36 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:36 battery power: -250W
Mar 14 09:12:36 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:36 grid meter: -442W
Mar 14 09:12:36 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:36 grid currents: [-3.97 -4.11 6.21]A
Mar 14 09:12:36 raspberrypi evcc[18734]: [site  ] DEBUG 2024/03/14 09:12:36 site power: -592W
Mar 14 09:12:36 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:36 charge currents: [0 0 0]A
Mar 14 09:12:36 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:36 charge total import: 2638.216kWh
Mar 14 09:12:36 raspberrypi evcc[18734]: [lp-1  ] DEBUG 2024/03/14 09:12:36 charger status: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.10

andig, 2024-03-14:
Bitte trace log des Fehlers

andig, 2024-03-14:
@premultiply der Modbus Fehler hat nix mit dem Intervall zu tun. Es braucht aber trotzdem ein passendes Log.

andig, 2024-03-14:
@thomas-hutterer-tik ich sehe da keinen Fehler?

thomas-hutterer-tik, 2024-03-14:
Hier mit dem beschriebenen Fehler, ziemlich am Ende

pi@raspberrypi:~ $ journalctl -fu evcc
-- Journal begins at Mon 2023-08-14 06:59:00 CEST. --
Mar 14 17:03:29 raspberrypi evcc[33499]: [modbus] TRACE 2024/03/14 17:03:29 modbus: send 00 43 00 00 00 06 01 03 06 06 00 01
Mar 14 17:03:29 raspberrypi evcc[33499]: [cache ] TRACE 2024/03/14 17:03:29 pvPower: 404
Mar 14 17:03:29 raspberrypi evcc[33499]: [cache ] TRACE 2024/03/14 17:03:29 pvEnergy: 48.300000000000004
Mar 14 17:03:29 raspberrypi evcc[33499]: [cache ] TRACE 2024/03/14 17:03:29 pv: [{4 0} {400 48.300000000000004}]
Mar 14 17:03:29 raspberrypi evcc[33499]: [modbus] TRACE 2024/03/14 17:03:29 modbus: recv 00 43 00 00 00 05 01 03 02 00 00
Mar 14 17:03:29 raspberrypi evcc[33499]: [modbus] TRACE 2024/03/14 17:03:29 modbus: send 00 44 00 00 00 06 01 03 06 9a 00 02
Mar 14 17:03:29 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:29 recv wa-pi/grid: '-348.02002'
Mar 14 17:03:29 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:29 recv wa-pi/grid/p1: '-0.3953337'
Mar 14 17:03:29 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:29 recv wa-pi/grid/p2: '-0.50894123'
Mar 14 17:03:29 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:29 recv wa-pi/grid/p3: '-0.5637211'
Mar 14 17:03:30 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:30 recv wa-pi/grid: '-342.94'
Mar 14 17:03:30 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:30 recv wa-pi/grid/p1: '-0.38742507'
Mar 14 17:03:30 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:30 recv wa-pi/grid/p2: '-0.4940192'
Mar 14 17:03:30 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:30 recv wa-pi/grid/p3: '-0.5655973'
Mar 14 17:03:31 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:31 recv wa-pi/grid: '-345.95'
Mar 14 17:03:31 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:31 recv wa-pi/grid/p1: '-0.39707828'
Mar 14 17:03:31 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:31 recv wa-pi/grid/p2: '-0.50289434'
Mar 14 17:03:31 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:31 recv wa-pi/grid/p3: '-0.559128'
Mar 14 17:03:32 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:32 recv wa-pi/grid: '-343.43'
Mar 14 17:03:32 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:32 recv wa-pi/grid/p1: '-0.38896397'
Mar 14 17:03:32 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:32 recv wa-pi/grid/p2: '-0.49796247'
Mar 14 17:03:32 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:32 recv wa-pi/grid/p3: '-0.5612646'
Mar 14 17:03:33 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:33 recv wa-pi/grid: '-344.38'
Mar 14 17:03:33 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:33 recv wa-pi/grid/p1: '-0.39248672'
Mar 14 17:03:33 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:33 recv wa-pi/grid/p2: '-0.4984888'
Mar 14 17:03:33 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:33 recv wa-pi/grid/p3: '-0.5616119'
Mar 14 17:03:34 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:34 recv wa-pi/grid: '-344.06'
Mar 14 17:03:34 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:34 recv wa-pi/grid/p1: '-0.39232513'
Mar 14 17:03:34 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:34 recv wa-pi/grid/p2: '-0.49796247'
Mar 14 17:03:34 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:34 recv wa-pi/grid/p3: '-0.5608352'
Mar 14 17:03:35 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:35 recv wa-pi/grid: '-344.96'
Mar 14 17:03:35 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:35 recv wa-pi/grid/p1: '-0.3941829'
Mar 14 17:03:35 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:35 recv wa-pi/grid/p2: '-0.4964655'
Mar 14 17:03:35 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:35 recv wa-pi/grid/p3: '-0.5643811'
Mar 14 17:03:36 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:36 recv wa-pi/grid: '-343.93'
Mar 14 17:03:37 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:37 recv wa-pi/grid/p1: '-0.3904'
Mar 14 17:03:37 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:37 recv wa-pi/grid/p2: '-0.4989948'
Mar 14 17:03:37 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:37 recv wa-pi/grid/p3: '-0.56191957'
Mar 14 17:03:38 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:38 recv wa-pi/grid: '-345.72'
Mar 14 17:03:38 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:38 recv wa-pi/grid/p1: '-0.3979895'
Mar 14 17:03:38 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:38 recv wa-pi/grid/p2: '-0.49924785'
Mar 14 17:03:38 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:38 recv wa-pi/grid/p3: '-0.5612749'
Mar 14 17:03:39 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:39 recv wa-pi/grid: '-345.41'
Mar 14 17:03:39 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:39 recv wa-pi/grid/p1: '-0.39297146'
Mar 14 17:03:39 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:39 recv wa-pi/grid/p2: '-0.4997117'
Mar 14 17:03:39 raspberrypi evcc[33499]: [mqtt  ] TRACE 2024/03/14 17:03:39 recv wa-pi/grid/p3: '-0.5653243'
Mar 14 17:03:39 raspberrypi evcc[33499]: [site  ] ERROR 2024/03/14 17:03:39 battery 1 energy: read failed: read tcp 192.168.1.7:34618->192.168.1.4:8899: i/o timeout
Mar 14 17:03:39 raspberrypi evcc[33499]: [modbus] TRACE 2024/03/14 17:03:39 modbus: send 00 45 00 00 00 06 01 03 06 08 00 01
[... cut ...]
Mar 14 17:03:59 raspberrypi evcc[33499]: [modbus] TRACE 2024/03/14 17:03:59 modbus: recv 00 47 00 00 00 05 01 03 02 12 7b
Mar 14 17:03:59 raspberrypi evcc[33499]: [site  ] ERROR 2024/03/14 17:03:59 pv 2 energy: read failed: modbus: response data size '2' does not match request quantity '2'
Mar 14 17:03:59 raspberrypi evcc[33499]: [site  ] DEBUG 2024/03/14 17:03:59 pv power: 623116W
Mar 14 17:03:59 raspberrypi evcc[33499]: [cache ] TRACE 2024/03/14 17:03:59 pvPower: 623116
Mar 14 17:03:59 raspberrypi evcc[33499]: [cache ] TRACE 2024/03/14 17:03:59 pvEnergy: 0

premultiply, 2024-03-14:
Ich sehe da pausenlose Modbus-Kommunikation. bzw. eigentlich ständig nur
`modbus: send 00 46 00 00 00 06 01 03 05 c4 00 01` ohne Antwort.

closed 2024-03-14 by andig