evcc-io/evcc #12926: Abbildung Start- und Ziel-Soc am Ladevorgang

fL4sH3r, 2024-03-14:
Besteht die Möglichkeit die Starttemperatur {vehicleSoc} und die Zieltemperatur {limitSoc} im Ladevorgang mit entsprechender Einheit abzubilden bzw. für alle Loadpoints ? Dann wüsste man wie viel Energy für das Delta aufgenommen wurde. 
Vorallem weil die Start- und ZielSoc ja variieren können.

Ist nur so eine Idee.

[image]

[image]

andig, 2024-03-14:
Die Energie bekommst du doch schon vom Zähler?!

naltatis, 2024-03-14:
Duplikat zu https://github.com/evcc-io/evcc/issues/6144

closed 2024-03-14 by naltatis

fL4sH3r, 2024-03-14:
Alles klar, danke für die Info.

fL4sH3r, 2024-03-14:
> Die Energie bekommst du doch schon vom Zähler?!

Ja das stimmt aber grade im Bereich Warmwasser hat man keinen Bezug von der Energiemenge zum Temperaturunterschied.
Bsp. 1500 Wh für 10°C  oder 1500 Wh für 30°C daher wäre das dort schon Sinnvoll. ;-)

naltatis, 2024-03-14:
Bei Wasser hast du ja, im Gegensatz zu angekabelten Fahrzeugen, ohnehin auch noch das Problem, dass du parallele Entnahme hast. Hier brauchts langfristig sicher noch ne bessere Art der Auswertung.