evcc-io/evcc #12927: Webinterface not starting up if Tibber Pulse is not working

machristoph1, 2024-03-14:
### Describe the bug

I recently experienced the problem that I could no longer reach the evcc Webinterface (connection timed out as well as connection refused messages) without any problems being reported in the logfiles. 
Just found by coincidence that my Tibber Pulse had run out of batteries. After replacing the batteries and restarting the evcc-service, the UI could be reached again. 
Not sure whether this is intended behavior. For sure, evcc cannot work as it is supposed to do in a setting like mine without a working Tibber pulse, but I would nevertheless have expected an error message instead of the interface not showing up. 

### Steps to reproduce

1. Take batteries out of Tibber Pulse
2. Try to reach the evcc Webinterface
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: localhost # .local suffix announces the hostname on MDNS
  port: 7070

log: debug 

interval: 60s # control cycle interval

sponsortoken: XXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: tibber-pulse 
  usage: grid  
  token: xxx
  homeid: xxx
  name: tibber_garage

chargers:
- type: template
  template: go-e-v3
  host: 192.168.1.y
  name: goe_abc

vehicles:
- type:  custom
  name: GLE350de
  title: GLE
  icon: car  
  #capacity: 27 #wäre der richtige Wert, da evcc aber Bedarf überschätzt, korrigiere ich den nach unten
  capacity: 24.0  
  phases: 2
  soc:
     source: http
     uri: http://192.168.1.x:8093/v1/state/mercedesme.0.xxx.state.soc.displayValue/plain?extraPlain=true
     method: GET
     timeout: 20s
  range: 
     source: http
     uri: http://192.168.1.x:8093/v1/state/mercedesme.0.xxx.state.rangeelectric.displayValue/plain?extraPlain=true
     method: GET
     timeout: 20s
  onIdentify:
    mode: pv
    priority: 1  

loadpoints:
- title: GO-E Rott Garage
  charger: goe_abc
  vehicle: GLE350de
  mode: pv

tariffs:
  grid:
    type: tibber
    token: xxx
    homeid: xxx

site:
- title: RottMaurer
  meters:
    grid: tibber_garage

influx: 
   url: http://localhost:8086
   database: evcc
   user: vvv
   password: ppp

# push messages
messaging:
  events:
    connect: # vehicle connect event
      title: "Car connected"
      msg: "${vehicleName} Car connected, mode:${mode}"
    guest:
      title: "Guest connected"
      msg: "${vehicleName} Guest connected, mode:${mode}"
  services:
    - type: script
      cmdline: /usr/local/bin/evcc_schedule.sh # ev. anpassen
      timeout: 50s
```

### Log details

```text
Output from journalctl

Mar 13 22:06:10 raspicm evcc[2474]: [main ] INFO 2024/03/13 22:06:10 evcc 0.124.10
Mar 13 22:06:10 raspicm evcc[2474]: [main ] INFO 2024/03/13 22:06:10 using config file: /etc/evcc.yaml
Mar 13 22:06:10 raspicm evcc[2474]: [main ] INFO 2024/03/13 22:06:10 starting ui and api at :7070
Mar 13 22:06:10 raspicm evcc[2474]: [db ] INFO 2024/03/13 22:06:10 using sqlite database: /var/lib/evcc/evcc.db
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-14:
Bitte mal trace log, ich glaube aber das Problem hatten wir schonmal. Suchfunktion?

machristoph1, 2024-03-14:
Ich habe es nicht gefunden beim Suchen (was nicht heißt, dass es das nicht gab). 
Es ist aktuell behoben, da der Pulse bei mir in einem Außenkasten ist, würde ich die Batterien ungerne wieder ausbauen, weil das eine größere Aktion ist.
Ich habe aber gestern, als das Problem bestand, den Service mal mit level trace gestartet. Hilft es, wenn ich das aus dem Log rauskopiere?

machristoph1, 2024-03-14:
Meintest Du den? https://github.com/evcc-io/evcc/issues/10407#issue-1954162247 
Da mein Pulse leider Batterien frisst, wird das in ein paar Wochen wieder auftreten, wenn ich weiß, was ich dann machen soll, kann ich gerne entsprechende Logs erstellen.

andig, 2024-03-14:
https://github.com/evcc-io/evcc/issues/10205
https://github.com/evcc-io/evcc/issues/10407

andig, 2024-03-14:
Ohne Netzzähler gehts nicht- dann wäre evcc ohne Funktion

andig, 2024-03-14:
Wenns reproduzierbar ist können wir schauen woher der Hänger kommt. Irgendwo fehlt wohl ein Timeout um evcc abzubrechen.

closed 2024-03-14 by andig

machristoph1, 2024-03-14:
Dass evcc dann ohne Funktion ist, ist mir klar. Wenn es mit einer Fehlermeldung ausstiege, wäre ja völlig ok. 

Und ja, scheint mir reproduzierbar. Geht ja wohl auch einfach über das Ziehen der Bridge. Ich provoziere das am Wochenende mal mit evcc --log trace und schicke den Output.

andig, 2024-03-14:
Gerne. Wenn reproduzierbar wäre Portweiterleitung für info@evcc.io gut

machristoph1, 2024-03-14:
Ist irgendwo beschrieben, was ich dafür tun muss? Kann ich sicher machen, aber natürlich nur für sehr begrenzte Zeit, will nach außen Ports nicht beliebig lange aufmachen

machristoph1, 2024-03-14:
Bzw. bekommen wir da eine möglichst sichere Lösung über VPN oder SSH oder so?

andig, 2024-03-14:
As you like. Entweder Portweiterleitung oder SSH für 

    ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIGi8rUJarKZuWnZNOnhtcXZCrPskbF7HOcig6nkjWlU2

Bitte Mail an info@evcc.io

machristoph1, 2024-03-14:
Ok. Ich schaue mir das am Wochenende an. Was braucht der Benutzer, für den ich den SSH-Key als Authentication akzeptiere, denn für Rechte?

malud, 2024-03-15:
Macht die Port-Weiterleitung Sinn wenn die Tibber Daten via API bezogen werden? Zumal er ggf. den Stecker zieht - dann ist der Pulse/Bridge ohnehin nicht erreichbar.

Wenn ein Fehler von der API zurück kommt sieht das wie folgt aus:

```graphql
subscription{
  liveMeasurement(homeId:"96a14971-525a-4420-aae9-e5aedaa129ff") {
    timestamp
    power
    maxPower
  }
}
```

```json
{
  "errors": [
    "Home with id \"96a14971-525a-4420-aae9-e5aedaa129ff\" does not have matching owner or grid contract owner d708818c-37f1-4e4c-a132-8f3785e3b46f"
  ]
}
#or
{
  "errors": [
    "Invalid homeId 'xxxxx-525a-4420-aae9-e5aedaa129ff'"
  ]
}
```

Auch möglich: `"Your subscription data will appear here after server publication"` (ggf offline case)

oder schlicht ein entsprechender StatusCode.

machristoph1, 2024-03-15:
[evcc.log](https://github.com/evcc-io/evcc/files/14613235/evcc.log)
Hier ist ein Auszug aus dem Systemlog nach dem Start von evcc im Trace-Modus
Das Muster der letzten Meldungen wiederholt sich dann jede Minute

andig, 2024-03-15:
@malud @machristoph1 hier gehts darum, dass evcc gar nicht erst startet. Den Fehler möchte ich reproduzieren/abstellen.