evcc-io/evcc #12931: After 0.124.10 Update on HA Bad Gateway 502

BaxPlay360, 2024-03-14:
### Describe the bug

After updating to 0.124.10 in home assistant, I can no longer get EVCC to run. I get a “Bad Gateway 502” error message. But when I load the DEMO YAML into the config EVCC works. Only when I load my config, which has always worked, do I get this message.

### Steps to reproduce

1. Update from 0.124.9 to 0.124.10

### Configuration details

```yaml
network:
schema: http
host: evcc.local # .local suffix announces the hostname on MDNS
port: 7070

log: debug
levels:
cache: error

interval: 10s # control cycle interval

telemetry: false

meters:

type: template
template: tibber-pulse
usage: grid
token: xxx
homeid: xxx
name: grid1
type: template
template: huawei-sun2000-rs485
id: 1
host: 192.168.178.194
port: 502
usage: pv
modbus: tcpip
storageunit: 0
name: pv2
type: template
template: huawei-sun2000-rs485
id: 2
host: 192.168.178.194
port: 502
usage: pv
modbus: tcpip
storageunit: 0
name: pv4
type: template
template: huawei-sun2000-rs485
id: 1
host: 192.168.178.194
port: 502
usage: battery
modbus: tcpip
storageunit: 1
capacity: 15
name: battery3
chargers:

type: template
template: go-e-v3
host: 192.168.178.29
name: wallbox4
loadpoints:

title: Garage
charger: wallbox4
mode: pv
phases: 3
mincurrent: 6
maxcurrent: 16
site:
title: Zu Hause
meters:
grid: grid1
pv:
- pv2
battery:
- battery3

tariffs:
currency: EUR # (default EUR)
grid:
type: tibber
token: "xxx"
feedin:
type: fixed
price: 0.086 # [currency]/kWh

mqtt:
broker: mqtt.eclipseprojects.io:1883
topic: evcc
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/03/14 11:42:51 evcc 0.124.10
[main  ] INFO 2024/03/14 11:42:51 using config file: /config/evcc.yaml
[main  ] INFO 2024/03/14 11:42:51 starting ui and api at :7070
[db    ] INFO 2024/03/14 11:42:51 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/03/14 11:42:51 connecting evcc-940402670 at tcp://mqtt.eclipseprojects.io:1883
[mqtt  ] DEBUG 2024/03/14 11:42:51 tcp://mqtt.eclipseprojects.io:1883 connected
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.10

andig, 2024-03-14:
Bitte ohne HA probieren

closed 2024-03-14 by andig

andig, 2024-03-14:
Wäre super wenn evcc dann abbrechen würde, dafür bräuchte ich aber Remotezugang