evcc-io/evcc #12935: Introduce new add-hoc mode for PV + Battery

alexonbandit, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
I was wondering if a new charge mode "PV + Battery" was already considered? Could not find anything that refers to my use case which is as follows:
- It is a sunny day, 11am and I know I am going to use my EV in say 3 hours.
- PV production is for example at app. 6kWh, the average house consumption is 1kWh, house battery e.g. 70% and evcc is set to mode "PV". 
So currently, I see a tendency from evcc to stay rather below 5kWh for charging the car than above 5kWh which is perfectly ok under normal circumstances. So over time the house batteries SoC goes up while the car is charging. 
This however leads to the situation, that the car is not fully charged but the house battery is almost full when I take the car for a ride. After that, the situation is, the PV production is still high and the house battery can only load a few minutes before it fully loaded and the excess power goes into the grid which is not perfect. 

**Describe the solution you'd like**
Introduce an add-hoc "PV+Battery" mode for which evcc considers the house battery for charging to the level were the battery has priority (aka prioritySoc) over the car. Since evcc may not know the max. discharge power the house battery delivers this might also be configurable. 

**Describe alternatives you've considered**
Maybe "residualPower" may be used to achieve the desire behavior. However not really handy to use.

VolkerK62, 2024-03-14:
have a lock here, it is one of many ideas   https://github.com/evcc-io/evcc/discussions/3530

alexonbandit, 2024-03-14:
Thanks for sharing the link! Sorry for any inconvenience.

closed 2024-03-14 by alexonbandit

andig, 2024-03-14:
Apart from what @VolkerK62 says lets look at what happens:

> So currently, I see a tendency from evcc to stay rather below 5kWh for charging the car than above 5kWh which is perfectly ok under normal circumstances.

So why would it do that? Strategy is "no grid consumption" (we're thinking about a winter mode "no grid feed-in", too). If your vehicle (maybe Tesla) can only be controlled at 1A steps and the charger is 3p you will loose up to 3p x 230V = 690W that go into the battery. You can play with residual power to (over)compensate that by setting to say -690W.

premultiply, 2024-03-14:
Maybe https://github.com/evcc-io/evcc/discussions/12190 does match better for this case.
This is more what is labeld "Assisted PV" in https://github.com/evcc-io/evcc/discussions/3255

alexonbandit, 2024-03-15:
Hi @andig I totally agree that the "no grid consumption strategy" is in 99% the desired system behavior!  

> So why would it do that? Strategy is "no grid consumption" (we're thinking about a winter mode "no grid feed-in", too). If your vehicle (maybe Tesla) can only be controlled at 1A steps and the [...]

Although winter mode, as I understood it, would also be very nice to have, I belief it does not match my use case depicted. It would approach the sweet spot in the control logic just from the other side, right? If so, battery discharge might be very low. 

Your assumption is correct, I am using a TWC3 in combination with a TESLA Model3. So playing with the residual power is not real convenient to use on the fly since my idea was to use it only as an add-hoc charging session only. If I would set residual power to say -3000W, I should get the desire behavior that the battery is also providing energy for the EV charging. But after the battery is drained, it would consume energy from the grid, right?

I thing #12190 indeed pretty much covers my use case as @premultiply assumes. 
Maybe this is much more complex than I though it was. I was thinking if the new mode is selected (aka PV + Bat or Assisted PV) the control logic check if battery SoC is above PrioritySoC and if true, adds the max discharge power to the current PV production until PrioritySoC is reached. When prioritySoC is reached evcc might go back to regular PV mode or to make it more complex introduces some sort of hysteresis to prevent the control logic to jitter. 

Unfortunately my programming skills are asymptotically at 0. Again thank you very much for your patience and big sorry that I did not find the associated post before raising this feature request! You guys are doing a fantastic job and evcc is really awesome!

premultiply, 2024-03-15:
AUnfortunately, there is no "maximum discharge power" for a battery. This is a dynamic thing that depends on many factors like soc, temperature, age, cell balancing, power limits and self-consumption of the inverter, ...
Therefore, this point can only be approached in a similar way to excess charging, where an attempt is made to minimize the feed-in power using control technology.
You must always be aware that the battery has its own independent control system that cannot be directly influenced by evcc. It always tries to equalize the power balance with the grid to 0 within its performance range and nothing else.

In order to approximate the maximum power output of the battery, a (small) power draw from the grid must therefore be constantly produced, which the battery control then increases over several interations up to its externally unknown, instantaneous power limit.

As clear as the theory is in principle, this still requires a few modifications to evcc and further considerations, as it is particularly important to clarify how, for example, several charging points with different or opposing modes are handled in detail.