evcc-io/evcc #12947: Fehler bei Ladeplanung (1-phasig/3-phasig)

Aheko, 2024-03-14:
### Describe the bug

Habe heute ein Update von v0.124.6 auf v0.124.10 gemacht. Seitdem wird bei der Berechnung der Startzeit für Ladeplanung nicht mehr berücksichtigt, ob für die Wallbox 1-phasig oder 3-phasig eingestellt ist. Es bleibt immer bei der Zeit für 3-phasiges Laden.
Bei v0.124.6 funktionierte das noch.

### Steps to reproduce

1. Einstellung im Ladepunkt 1-phasig über **WEB-Oberfläche evcc**
2. Starten Ladeplanung 
3. Berechnung der Startzeit geht trotzdem von 3-phasig aus
...

### Configuration details

```yaml
vehicles:
- type: template
  template: audi 
  title: Audi A3  
  user: info@andreaskoerner.de  
  password: zzz3BY4b7Mh7  
  vin: WAUZZZGY5PA019271  
  capacity: 13  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  meter: charge6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
```

### Log details

```text
[main  ] INFO 2024/03/14 19:35:01 evcc 0.124.10
[main  ] INFO 2024/03/14 19:35:01 using config file: /etc/evcc.yaml
[main  ] INFO 2024/03/14 19:35:01 starting ui and api at :7070
[db    ] INFO 2024/03/14 19:35:01 using sqlite database: /home/user/.evcc/evcc.db
[eebus ] INFO 2024/03/14 19:35:01 Local SKI:  47921b52254921f374219f89dad5cba801fdd7d3
[eebus ] DEBUG 2024/03/14 19:35:01 mdns: announce
[eebus ] DEBUG 2024/03/14 19:35:01 mdns: using avahi
[eebus ] DEBUG 2024/03/14 19:35:01 starting websocket server on :4712
[eebus ] DEBUG 2024/03/14 19:35:01 mdns: start search
[eebus ] DEBUG 2024/03/14 19:35:01 ski: 54555f4f2db270b53c8cd841dcf473e7d9552f6a name: Elli-Wallbox-2209A24BSD brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2209A24BSD register: false host: wallbox-2209A24BSD.local port: 4712 addresses: [192.168.178.78]
[eebus ] DEBUG 2024/03/14 19:35:01 delaying connection to 54555f4f2db270b53c8cd841dcf473e7d9552f6a by 2.737s to minimize double connection probability
[eebus ] DEBUG 2024/03/14 19:35:04 trying to connect to 54555f4f2db270b53c8cd841dcf473e7d9552f6a at 192.168.178.78
[eebus ] DEBUG 2024/03/14 19:35:04 initiating connection to 54555f4f2db270b53c8cd841dcf473e7d9552f6a at 192.168.178.78:4712
[eebus ] DEBUG 2024/03/14 19:35:04 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/14 19:35:05 Recv: d:_i:47859_Elli-Wallbox-2209A24BSD: to NodeManagement read 3708680 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD reply 2 3708680 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD result 3 3708680 ResultData 0
[eebus ] DEBUG 2024/03/14 19:35:05 Recv: d:_i:47859_Elli-Wallbox-2209A24BSD: to NodeManagement reply 3708682 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/03/14 19:35:05 ev connected
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD read 6 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 5 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD read 7 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 8 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD read 9 NodeManagementUseCaseData
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 10 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 11 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 12 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 13 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 14 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD call 15 NodeManagementBindingRequestCall
[eebus ] DEBUG 2024/03/14 19:35:05 Send: d:_i:47859_Elli-Wallbox-2209A24BSD read 16 DeviceConfigurationKeyValueDescriptionListData
[... 83 lines cut ...]
[lp-1  ] ERROR 2024/03/14 19:35:16 charger: timeout
[eebus ] DEBUG 2024/03/14 19:35:16 avahi - error resolving service: Timeout reached
[site  ] DEBUG 2024/03/14 19:35:26 ----
[lp-1  ] DEBUG 2024/03/14 19:35:26 charge power: 1182W
[site  ] DEBUG 2024/03/14 19:35:26 pv power: 0W
[site  ] DEBUG 2024/03/14 19:35:26 battery soc: 51%
[site  ] DEBUG 2024/03/14 19:35:26 battery power: 3499W
[site  ] DEBUG 2024/03/14 19:35:26 grid meter: -1853W
[site  ] DEBUG 2024/03/14 19:35:26 site power: 1746W
[lp-1  ] DEBUG 2024/03/14 19:35:26 charge voltages: [230 228 231]V
[lp-1  ] DEBUG 2024/03/14 19:35:26 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/14 19:35:26 charge currents: [5.31 0.036 0.031]A
[lp-1  ] DEBUG 2024/03/14 19:35:26 charge total import: 993.506kWh
[lp-1  ] ERROR 2024/03/14 19:35:26 charger: timeout
^C[eebus ] DEBUG 2024/03/14 19:35:29 mdns: stop announcement
[eebus ] DEBUG 2024/03/14 19:35:29 mdns: stop announcement
```

### What type of operating system are you running?

Linux

### Version

v0.124.10

andig, 2024-03-14:
Leider kst fon deinem Fehler im Logfile nichts zu sehen?

VolkerK62, 2024-03-14:
```
[lp-1  ] DEBUG 2024/03/14 19:35:06 charge voltages: [230 227 230]V
[lp-1  ] DEBUG 2024/03/14 19:35:06 detected connected phases: 3p
```
deshalb wird mit 3p gerechnet.
Wenn das Auto nur 1p kann, dann `phases: 1` beim vehicle definieren

andig, 2024-03-14:
…und unbedingt das Passwort vom Auto ändern und dieses PW NIEMALS irgendwo sonst verwenden! Das ist verbrannt!

closed 2024-03-14 by andig