evcc-io/evcc #12960: Wrong Loads Calc

kohbi, 2024-03-15:
### Describe the bug

The calculation of the consumed kwh in Montly Dashboard is wrong. 

look at Screeshot 

[image]

### Steps to reproduce

1.update evcc to last version
2. switch to other month, same issue
3.
...

### Configuration details

```yaml
[main  ] INFO 2024/03/15 10:38:55 evcc 0.124.10
[main  ] INFO 2024/03/15 10:38:55 using config file: /home/pi/evcc.yaml

Konfiguration (/home/pi/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

tariffs:
  grid:
    # either
    type: tibber
    token: *****
    homeid: "ab267ab3-e032-4107-a185-1c8b056de57e" # optional if multiple homes associated to account

#tariffs:
 # currency: EUR
 # grid:
 #   type: fixed
 #   price: 0.39
 # feedin:
 #   type: fixed
 #   price: 0.085

meters:
- type: template
  template: senec-home 
  usage: grid  
  host: 192.168.178.138  
  name: grid1
  schema: https
- type: template
  template: senec-home 
  usage: pv  
  host: 192.168.178.138  
  name: pv2
  schema: https
- type: template
  template: senec-home 
  usage: battery  
  host: 192.168.178.138  
  name: battery3
  schema: https
chargers:
- type: template
  template: daheimladen-mb 
  host: 192.168.178.54  
  port: 502  
  name: wallbox5

vehicles:
- type: template
  template: renault 
  title: ZOE  
  user: *****
  password: *****
  vin: *****
  capacity: 50  
  name: ev4

- name: my_car
  type: template
  template: mg2mqtt
  title: MG5-MQTT # Wird in der BenutzeroberflÃ¤che angezeigt (Optional)
  user: *****
  vin: *****
  capacity: 61 # AkkukapazitÃ¤t in kWh (Optional) 
loadpoints:
- title: WALLBOX
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 8
  maxcurrent: 16
#  resetOnDisconnect: true  
 # disable:
      #delay: 1m
      #threshold: 200
  soc:
    poll:
      mode: charging
      interval: 5m
    estimate: true

site:
  title: HOME
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50
#  prioritySoC: 30
 # bufferSoc: 50
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

Version: `0.124.10`
```

### Log details

```text
[main  ] INFO 2024/03/15 10:40:44 evcc 0.124.10
[main  ] INFO 2024/03/15 10:40:44 using config file: /home/pi/evcc.yaml
[main  ] INFO 2024/03/15 10:40:44 starting ui and api at :7070
[db    ] INFO 2024/03/15 10:40:44 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2024/03/15 10:40:44 connecting evcc-1382159621 at tcp://localhost:                                                                                                                                                                                                1883
[mqtt  ] DEBUG 2024/03/15 10:40:44 tcp://localhost:1883 connected
[lp-1  ] WARN 2024/03/15 10:40:46 deprecated: minCurrent setting is ignored, ple                                                                                                                                                                                                ase remove
[lp-1  ] WARN 2024/03/15 10:40:46 deprecated: maxcurrent setting is ignored, ple                                                                                                                                                                                                ase remove
[lp-1  ] WARN 2024/03/15 10:40:46 deprecated: phases setting is ignored, please                                                                                                                                                                                                 remove
[lp-1  ] DEBUG 2024/03/15 10:40:46 charge total import: 5172.800kWh
[site  ] INFO 2024/03/15 10:40:54 site config:
[site  ] INFO 2024/03/15 10:40:54   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/03/15 10:40:54     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/15 10:40:54     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/03/15 10:40:54     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/03/15 10:40:54   vehicles:
[site  ] INFO 2024/03/15 10:40:54     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/03/15 10:40:54     vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/03/15 10:40:54 loadpoint 1:
[lp-1  ] INFO 2024/03/15 10:40:54   mode:        pv
[lp-1  ] INFO 2024/03/15 10:40:54   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/03/15 10:40:54   meters:      charge ✓
[lp-1  ] INFO 2024/03/15 10:40:54     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/03/15 10:40:54 phase timer inactive
[lp-1  ] DEBUG 2024/03/15 10:40:54 pv timer inactive
[site  ] WARN 2024/03/15 10:40:54 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/03/15 10:40:54 ----
[lp-1  ] DEBUG 2024/03/15 10:40:54 charge power: 0W
[site  ] DEBUG 2024/03/15 10:40:55 pv power: 1743W
[site  ] DEBUG 2024/03/15 10:40:55 battery soc: 47%
[site  ] DEBUG 2024/03/15 10:40:55 battery power: -744W
[site  ] DEBUG 2024/03/15 10:40:55 grid meter: -16W
[site  ] DEBUG 2024/03/15 10:40:55 site power: -660W
[lp-1  ] DEBUG 2024/03/15 10:40:56 charge voltages: [235 235 234]V
[lp-1  ] DEBUG 2024/03/15 10:40:56 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/15 10:40:56 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/15 10:40:56 charge total import: 5172.800kWh
[lp-1  ] DEBUG 2024/03/15 10:40:56 charger status: A
[lp-1  ] INFO 2024/03/15 10:40:56 car disconnected
[lp-1  ] DEBUG 2024/03/15 10:40:56 set session soc limit: 0
[lp-1  ] DEBUG 2024/03/15 10:40:56 set session energy limit: 0
[site  ] DEBUG 2024/03/15 10:41:04 ----
[lp-1  ] DEBUG 2024/03/15 10:41:04 charge power: 0W
[site  ] DEBUG 2024/03/15 10:41:04 pv power: 2041W
[site  ] DEBUG 2024/03/15 10:41:04 battery soc: 47%
[site  ] DEBUG 2024/03/15 10:41:04 battery power: -1078W
[site  ] DEBUG 2024/03/15 10:41:04 grid meter: 14W
[site  ] DEBUG 2024/03/15 10:41:04 site power: -965W
[lp-1  ] DEBUG 2024/03/15 10:41:05 charge voltages: [236 235 234]V
[lp-1  ] DEBUG 2024/03/15 10:41:05 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/15 10:41:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/15 10:41:05 charge total import: 5172.800kWh
[lp-1  ] DEBUG 2024/03/15 10:41:05 charger status: A
[site  ] DEBUG 2024/03/15 10:41:14 ----
[lp-1  ] DEBUG 2024/03/15 10:41:14 charge power: 0W
[site  ] DEBUG 2024/03/15 10:41:14 pv power: 2651W
[site  ] DEBUG 2024/03/15 10:41:14 battery soc: 47%
[site  ] DEBUG 2024/03/15 10:41:14 battery power: -935W
[site  ] DEBUG 2024/03/15 10:41:14 grid meter: -718W
[site  ] DEBUG 2024/03/15 10:41:14 site power: -1552W
[lp-1  ] DEBUG 2024/03/15 10:41:15 charge voltages: [236 236 234]V
[lp-1  ] DEBUG 2024/03/15 10:41:15 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/15 10:41:15 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/15 10:41:15 charge total import: 5172.800kWh
[lp-1  ] DEBUG 2024/03/15 10:41:15 charger status: A
[site  ] DEBUG 2024/03/15 10:41:24 ----
[lp-1  ] DEBUG 2024/03/15 10:41:24 charge power: 0W
[site  ] DEBUG 2024/03/15 10:41:24 pv power: 2479W
[site  ] DEBUG 2024/03/15 10:41:24 battery soc: 47%
[site  ] DEBUG 2024/03/15 10:41:24 battery power: -935W
[site  ] DEBUG 2024/03/15 10:41:24 grid meter: -544W
[site  ] DEBUG 2024/03/15 10:41:24 site power: -1379W
[lp-1  ] DEBUG 2024/03/15 10:41:25 charge voltages: [237 236 235]V
[lp-1  ] DEBUG 2024/03/15 10:41:25 detected connected phases: 3p
[lp-1  ] DEBUG 2024/03/15 10:41:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/03/15 10:41:25 charge total import: 5172.800kWh
[lp-1  ] DEBUG 2024/03/15 10:41:25 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.124.10

andig, 2024-03-15:
Funny. It's working for me. Could you share a log with `db: trace`?

StefanSchoof, 2024-03-15:
Und wenn ich 45,7 + 29,1 + 31,8 + 38,7 + 42,2 rechne komme ich schon auf 187,5 also mehr als die 151,15 du hast.

kohbi, 2024-03-15:
oh sorry... evcc rechnet doch richtig....
daheimladen rechnet falsch ... 
habe mich verleiten lassen =)

closed 2024-03-15 by kohbi