evcc-io/evcc #12961: Mobile UI: Aktiven Ladepunkt an erste Stelle setzen

Naiki92, 2024-03-15:
**Describe the solution you'd like**
Es wäre praktisch in der mobilen UI den aktuell aktiven Ladepunkt immer an erster Stelle zu haben, wenn nur einer aktiv ist. Aktuell muss immer nach links geswiped werden, was bei mehreren Wallboxen manchmal etwas nervig sein kann.

**Additional context**
Nice to have und würde den Waf erheblich erhöhen ;-)

Aktueller Stand:
[image]

andig, 2024-03-15:
Mehrfach diskutiert- wir möchten konsistente Reihenfolge beibehalten.

@naltatis könnte man aktive Ladepunkte im Flipper evtl. farblich anders markieren?

Klausmd5, 2024-03-15:
Wie wärs mit einer Option in den Settings? Wobei des vermutlich aktuell zu viel Aufwand für den Nutzen ist.

naltatis, 2024-03-15:
Ja, in der Tat ein Thema, was wir schon oft diskutiert haben. Was wir ja bereits umgesetzt haben ist, dass oben im Energiediagramm jeder aktive Ladevorgang seine eigene "Klammer" mit dem jeweiligen Fahrzeug/Ladepunkt Icon bekommt. Damit bekommst du einen Überblick, was gerade aktiv ist (vorausgesetzt die Fahrzeuge haben unterscheidbare Icons - Auswahl ist aktuell noch eingeschränkt).

Automatisches Umsortieren der UI (auch via Opt-In Setting) bringt uns (und auch dem Nutzer) in der Tat deutlich mehr Probleme, als es in der Praxis lösen würde. Klickflächen während einer Interaktion automatisch austauschen ist ein UX-No-Go. Was wir als "leichtgewichtige Variante" mal umsetzen könnten, ist eine Einstellung, die beim Aufruf von evcc einmalig(!) den aktiven Ladevorgang anspringt.

Es gab auch schon den Wunsch, eines Direkteinsprungs auf einen bestimmten Ladepunkt. Konkretes Szenario: Haushalt mit zwei Personen, die jeweils ein Fahrzeug und Ladepunkt haben. Person 1 möchte immer bei Ladepunkt 1 starten, Person zwei bei Ladepunkt 2.

Hier könnte ein Pro-Device (Browser) Setting ausreichen:

Ladepunkt-Anzeige: ["Ladepunkt 1", "Ladepunkt 2", ... "Ladepunkt X", "Erster aktiver Ladepunkt", "Zuletzt angesehener Ladepunkt"]

Offene Frage ist hier nur, wie wir "Erster Aufruf" definieren: Ist Web(-App) schließen und wieder öffnen schon ein Neustart? Browserfenster kurz in den Hintergrund setzen und wieder nach vorne holen?

naltatis, 2024-03-15:
Farbliche Markierung ist an der Stelle etwas schwierig, da wir dann an der Stelle über Farbe nicht nur "ausgewählt"/"nicht ausgewählt" sondern auch "lädt"/"lädt nicht" kommunizieren. Also Schwarz/Grau/Grün/Hellgrün. Ich hab das https://github.com/evcc-io/evcc/pull/12973 jetzt mal über Form (⚡) gelöst.

Naiki92, 2024-03-15:
> Ich hab das https://github.com/evcc-io/evcc/pull/12973 jetzt mal über Form (⚡) gelöst. 

Sehr cool 👌 Kommt der Sache schon mal nen Schritt näher.

PR #12973 (2024-03-15): Mobile UI: Active loadpoint indicator
fixes #12961

- shows lighting instead of dot for load points that are charging.
- distinguish between connected/not connected

⚡= charging
● = connected
○ = not connected

all connected, 2nd charging & selected
[image]

all connected, 2nd charging, 1st selected
[image]

not connected, connected & selected, not connected, connected
[image]

not connected, charging & selected, not connected, connected
[image]
closed 2024-03-15 by andig